kizniche/Mycodo #725: No data on Live screen

zsole2, 2019-12-28:
v8.2.1
Pi Zero

While setting up a new system, I run into the following:
- added simply a CPU temperature input, activated
- no data is shown on Live screen
- log level raised to debug
- demon log shows that the readings are carried out, and stored to influxdb:
```
2019-12-28 11:20:39,105 - DEBUG - mycodo.inputs.rpi_cpu_gpu_temperature_de9d27e7 - Acquiring Measurements...
2019-12-28 11:20:39,110 - DEBUG - mycodo.inputs.rpi_cpu_gpu_temperature_de9d27e7 - CPU Temperature: 32.552
2019-12-28 11:20:39,170 - DEBUG - mycodo.inputs.rpi_cpu_gpu_temperature_de9d27e7 - GPU Temperature: 32.6
2019-12-28 11:20:39,315 - DEBUG - mycodo.controllers.controller_input_de9d27e7 - Adding measurements to InfluxDB with ID de9d27e7-6ef2-49c7-aa6b-5fc4ccfe3d53: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 32.552, 'timestamp_utc': None}, 1: {'measurement': 'temperature', 'unit': 'C', 'value': 32.6, 'timestamp_utc': None}}
```

- web log is full with the following repeated:
```
Dec 28 11:17:00 chz-1 gunicorn[4367]: 2019-12-28 11:17:00,872 URL for 'last_data' raised and error: database not found: mycodo_db
Dec 28 11:17:00 chz-1 gunicorn[4367]: Traceback (most recent call last):
Dec 28 11:17:00 chz-1 gunicorn[4367]:   File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_general.py", line 305, in last_data
Dec 28 11:17:00 chz-1 gunicorn[4367]:     raw_data = dbcon.query(query_str).raw
Dec 28 11:17:00 chz-1 gunicorn[4367]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/influxdb/client.py", line 461, in query
Dec 28 11:17:00 chz-1 gunicorn[4367]:     in data.get('results', [])
Dec 28 11:17:00 chz-1 gunicorn[4367]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/influxdb/client.py", line 460, in <listcomp>
Dec 28 11:17:00 chz-1 gunicorn[4367]:     for result
Dec 28 11:17:00 chz-1 gunicorn[4367]:   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/influxdb/resultset.py", line 25, in __init__
Dec 28 11:17:00 chz-1 gunicorn[4367]:     raise InfluxDBClientError(self.error)
Dec 28 11:17:00 chz-1 gunicorn[4367]: influxdb.exceptions.InfluxDBClientError: database not found: mycodo_db
```

kizniche, 2019-12-28:
Have you reviewed the install/setup log?

zsole2, 2019-12-28:
Install appeared OK, finished successfully. Just now double-checked, the word 'error' only appears within liberror

zsole2, 2019-12-29:
Did fresh installs three times on two different SD cards, with the same results...

zsole2, 2019-12-29:
Installed v8.1.0, and it works correctly...

leoneo, 2019-12-29:
Hi 
I tested all the latest versions of Raspbian, Mycodo only works on the Lite version.

kizniche, 2020-01-08:
@leoneo Can you attach your setup.log so I can see if there are any errors from missing dependencies or something else?

leoneo, 2020-01-09:
Sorry I can not have the setup.log, mycodo is running well with Lite version. I remember that somes problems come with the python version and permission I think.

kizniche, 2020-01-13:
Install issue moved to #732

closed 2020-01-13 by kizniche