kizniche/Mycodo #726: Reorder Actions in Functions

leoneo, 2019-12-29:
Hi,

Can you please add the function to reorder Actions in Functions? 

And maybe the possibility to process two or more Actions in the same time. Because I need to ramp two PWM output in same time, actually Mycodo ramp the first PWM according the duration and execute the next action ramp PWM. 

I have solve my problem by duplicat the function and ramp the second PWM in the same period, but I thing it is note well.

Thank you
Leo

kizniche, 2020-01-08:
I like these ideas:

1. Add ability to reorder action sequence.

2. Add ability to execute action and not wait until it finishes to execute the next action. This, when enabled for an action, will execute the action as a thread rather than waiting for the action to finish before executing the next action.

The first feature is merely a convenience to not have to delete many actions to reorder the sequence, but the second feature makes the action system much more powerful and easier to configure complex action sequences.

leoneo, 2020-01-09:
Yes that's it.

Perhaps it can be done by putting numbers in front of the actions to define the order of execution, if several actions have the same order number then they are executed at the same time.