kizniche/Mycodo #728: Custom Output Status

mrplow, 2020-01-06:
**Is your feature request related to a problem? Please describe.**
Yes but only a minor problem. Mostly to do with using Linux Command Outputs.
An Output's Active/Inactive status is based off of Mycodo being the sole controller of that device.
Depending on one's usage that may not always be the case. People like flipping switches what can I say.

**Describe the solution you'd like**
Perhaps an option to input a linux command, in my case a script which would curl a response from the device, which returns a boolean value. This value would be applied against the device's status.

Thanks!

kizniche, 2020-01-08:
I could add two features:

1. Currently if the output is on, it will not attempt to turn it on when instructed (returning an error "already on"). I could add a checkbox that would enable sending the on command even if it's listed as already being on.

2. I could add a command field that would be executed and, if a 1 is returned set the output to on, and if a 0 is returned, set the output to off. Another field could be added "status execution duration (seconds)" that will define how often to execute the output status command.

mrplow, 2020-01-09:
> I could add two features:
>
> 1. Currently if the output is on, it will not attempt to turn it on when instructed (returning an error "already on"). I could add a checkbox that would enable sending the [...]

Perfect!

kizniche, 2020-04-11:
I just added a "Force Output" option for Command/Python/Wireless Outputs. I started to build in the functionality for a status command, but stopped because it started to become very complicated and decided to put it off for another day or to abandon that idea.

kizniche, 2020-05-12:
You'll be happy to hear I've begun refactoring the whole Output Controller to allow custom output modules to be used (similar to custom Inputs and Controllers that can be created and imported). This means you will soon be able to create your own Output module that can incorporate any mechanism for detecting the state of the output you desire. I'm a little more than half way finished with this feature, and it will be available in the 8.5.0 release. When the release is made, we can use this issue to develop and test a new module with your desired effect.

mrplow, 2020-05-12:
That's great @kizniche, awesome job.

Looking forward to 8.5.0 (but no rush)

kizniche, 2020-07-23:
The custom output module feature has been fully-implemented. In each output module located at ~/Mycodo/mycodo/outputs/ there is a function is_on() which will be queried before turning an output on or off and when a user is on the Output page of the web interface (to determine whether to show the output as on or off). You should now be able to easily create a new output module (merely copy the most similar one and modify the is_on() function, and set "output_name_unique" to a unique name for the output), and import it on the Configuration -> Output page.

ebo, 2020-12-17:
I was just reading through the issues and saw this.  From your description it looks like the feature is already added.  Has it been documented elsewhere?  Might be time to close this one.

closed 2024-03-20 by mrplow