kizniche/Mycodo #729: Dashboard does not load on start

zsole2, 2020-01-06:
V8.2.1
Buster Lite
Pi Zero

When I select the landing page in configuaration as Dashboard, I receive the following on startup:

```
Error 500: Internal Server Error
Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2446, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1951, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restplus/api.py", line 584, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1820, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1949, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1935, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_general.py", line 75, in home
    return redirect(url_for('routes_page.page_dashboard'))
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/helpers.py", line 370, in url_for
    return appctx.app.handle_url_build_error(error, endpoint, values)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2215, in handle_url_build_error
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/helpers.py", line 358, in url_for
    endpoint, values, method=method, force_external=external
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/werkzeug/routing.py", line 2020, in build
    raise BuildError(endpoint, values, method, self)
werkzeug.routing.BuildError: Could not build url for endpoint 'routes_page.page_dashboard'. Did you forget to specify values ['dashboard_id']?
```
Apparently the addition of multiple dashboards caused this issue.

kizniche, 2020-01-06:
Duplicate of #727. This is included in the next release, which I'll get to shortly... I'm currently returning from travel for work.

zsole2, 2020-01-06:
Ahh, I see now... Closing...

closed 2020-01-06 by zsole2