kizniche/Mycodo #730: Raspbian desktop using gpio expander tool

tomkm7p, 2020-01-08:
Hi,

Does anyone have any expereince running mycodo through a desktop install of raspbian and using the gpio pins of a Pi Zero through Gpio Expander tool ?

https://www.raspberrypi.org/blog/gpio-expander/

I would like to leverage the processing power of a few old laptops I got and after reading about the desktop raspbian os, felt like this is a viable solution.

tomkm7p, 2020-01-08:
The tool supports python and scratch, not sure what other programming languages mycodo is written on

kizniche, 2020-01-08:
@zsole2 runs Mycodo on a Pi Zero. It's not incredibly resource-heavy. If you're already using the zero for the GPIO, why not just give running everything on the Pi?

To answer your question, right now running Mycodo on a PC is very experimental. Docker has just been integrated into the latest versions, but I wouldn't say it's at a state where I would want to start tackling bugs/issues with it.

However,dont let that stop you from experimenting if you already have a PC and Pi Zero you want to play around with. Just understand that you're in uncharted territory and you are likely to encounter issues.

kizniche, 2020-05-14:
I've been working on a new feature for the 8.5.0 release, which adds support for user-created output modules that allow custom-built output modules to be imported and used By Mycodo, much the same way custom Inputs and controllers operate. All information about an output is now contained within a single file that can be developed, which should make development and testing of new outputs very quick. What this means for this issue, is the GPIO expander is more accessible as a potential new output. I don't have a Pi Zero myself, but I'm willing to help develop an output that can make use of [GPIO Zero](https://github.com/gpiozero/gpiozero) and [Remote GPIO](https://gpiozero.readthedocs.io/en/stable/remote_gpio.html). If a functional output can be developed, we can move to integrating it into the built-in set for Mycodo. Please let me know if this is of any interest to you and if you're willing to test.

kizniche, 2020-05-14:
In my latest progress update, I threw together a Remote GPIO output module. It's currently untested, but goes to show how the new system promote rapid development, as I was able to build it in just a few minutes. You can find it [here](https://github.com/kizniche/Mycodo/blob/1b6c0643b7b325096e243cbf9f62bedcf59ba050/mycodo/outputs/examples/output_remote_gpio.py). If you want to test, you'll need to [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch), then import it on the ```Configure -> Outputs``` page.

kizniche, 2020-05-15:
FYI, I moved the custom Remote GPIO outputs (two of them now, one for on/off, another for PWM) to https://github.com/kizniche/Mycodo-custom while they're untested.

closed 2020-08-11 by kizniche