kizniche/Mycodo #731: Error 32 with SHT31 smart gadget

BirdBare, 2020-01-11:
### Describe the problem/bug
Mycodo is having problems connecting to the SHT31 smart gadget. This can happen at startup and after running for over 24 hours.

Error -> 2020-01-11 16:28:34,622 - ERROR - mycodo.inputs.sht31_smart_gadget_31aed3bd - InputModule.get_measurement() method raised IOError: [Errno 32] Broken pipe

### Versions:

 - Mycodo Version: [Latest]
 - Raspberry Pi Version: [3B+]
 - Raspbian OS Version: [Buster Lite]

### Reproducibility

It happens on and off. Sometimes at startup, sometimes not for a day max.

Thanks for fixing the last issue!

kizniche, 2020-01-13:
I've had the same issue. I haven't been able to figure out what causes it to replicate it. I also haven't been able to figure out how to reliably "reset" Bluetooth to get it to work. Just to clarify your issue, you no longer acquire measurements after this error?

kizniche, 2020-01-13:
I need to get a new battery for mine if I'm going to be able to test. Does deactivating and reactivation the input restore functionality?

BirdBare, 2020-01-14:
I have tried deactivating the input, rebooting the pi, turning BT on and off on device, and rebooting the device. From my testing, only rebooting the pi and device works.

I'll retry deactivating the input tonight for you. If you need anymore information I am your guinea pig.

BirdBare, 2020-01-14:
I missed your above comment. Yes, I am unable to acquire measurements. I too am looking for the root cause. If I find anything I'll send it your way. 

More information, 
I have 4 of the SHT31 BT sensors. Each with a 60 second period. Could the measurements be overlapping causing confusion for the BT controller?

kizniche, 2020-01-14:
If there are multiple SG inputs, a lock file is used so only one can be accessed at a single time (limitation of Bluetooth). The lock file is essentially for using Bluetooth. Once one input relinquished the lock file, another input can claim it, until all inputs acquire their measurements.

I would like to find a way to reset the integrated Bluetooth module if such an error occurs so rebooting isn't necessary. I'm sure USB Bluetooth modules can be reset, for example by resetting the USB hub power, but integrating a fix in the input module for the integrated RPi BT is preferred.

kizniche, 2020-01-14:
I do recall now I had to reset the device as well sometimes. This doesn't bode well for adding a solution that can be fully-realized through software alone. It would be nice to find the actual cause of the issue, whether it's the SG or the RPi BT that's causing it. Perhaps it's the "download history" feature that's causing the issue. Have you set the input to not download history and only pull the latest measurement and observed if the issue occurs?

BirdBare, 2020-01-14:
I think you are using the bluepy library right? I found this issue pertaining to a broken pipe: https://github.com/ChristianKuehnel/btlewrap/issues/14
They show the issue has been fixed. But it may lead us to a solution.

I experience the issue without download history checked. It honestly seems to happen at random in my limited experience.

BirdBare, 2020-01-14:
I may have misspoke. I thought I had unchecked download history but it was checked. I have unchecked it now and reset my devices. I will keep you updated if the error occurs.

kizniche, 2020-01-14:
bluepy 1.3.0 should be the version installed, either when the module is installed or during an upgrade, so that bug shouldn't be an issue. A broken pipe can be caused by a number of issues. After testing without downloading data, I would test with just one SG input activated, if you are able.

wigglesdetroit, 2020-01-15:
I have two of these sensors. One has been rock solid and one gave me so many of the same problems you guys are having I switched . Everything else is the same I believe as far as version number of software and pi version.  Can I do anything to help troubleshoot? Awesome software by the way!

kizniche, 2020-01-15:
Interesting. Can you check the version number of you SG firmware? This info can be found in the daemon log after activating the input.

wigglesdetroit, 2020-01-16:
They both appear to have the same firmware version. 1.3 

INFO - mycodo.inputs.sht31_smart_gadget_c3761248 - Sensirion AG, SmartGadget, SN: 00000000, Name: Smart Humigadget, Firmware: 1.3, Hardware: 1, Software: 0.1, Log Interval: 600.0 sec

INFO - mycodo.inputs.sht31_smart_gadget_7f8f3dff - Sensirion AG, SmartGadget, SN: 00000000, Name: Smart Humigadget, Firmware: 1.3, Hardware: 1, Software: 0.1, Log Interval: 600.0 sec

Side note, I had to SSH in to look at the log. In the software it gives me a message "File empty: /var/log/mycodo/mycodo.log" I found the log at "/var/log/mycodo/mycodo.log.1"

kizniche, 2020-01-16:
That log quirk is a known issue, and results from the use of logrotate, which splits the log when it gets to a certain size. I still haven't found a good way to have Mycodo work well with logrotate, but I'll add it as it's own issue to remind me to try again to fix it.

As for the SG, I'll have to take a look at it again when I get a new battery. Thanks for the sensor info. If you come across any clues as to what the issue may be, please let me know.

BirdBare, 2020-02-09:
I've been testing the devices. Unchecking download history definitely helped. Now I only need to reset the RPI every couple days.

One thing I noticed, it always happens after 2 or 3 days consistently. It is definitely something on the RPI side. That the only update I have for now.

On a slightly related note. Can I interface directly with the daemon to set and read inputs and outputs? I have to reset the RPI every few days so I need to reset some outputs. It would be nice to instead save the state on shutdown and reload the states on startup using the daemon.

Thanks for this amazing software!

BirdBare, 2020-04-05:
I read a previous issue with the SHT32 and you mentioned implementing a locking mechanism for using multiple devices. Was that implemented? I think that may be the issue.

kizniche, 2020-04-05:
Locking has been in use since the Input was introduced to Mycodo. The issue is likely the integrated Bluetooth of the Pi. Do you have a USB Bluetooth dongle to test with?

BirdBare, 2020-04-05:
I don't but I will get one. Should I plug into the usb and select hci[1]?

Thanks!

kizniche, 2020-04-05:
Yes, it should be that easy, as long as the Bluetooth dongle is supported in Linux.

BirdBare, 2020-04-29:
Buying a bluetooth dongle fixed the issue. I can make it about a week before I lose connections. I have added a save states function to my Mycodo install so I can restart the RPI each night when the lights are off. The dongle and restarting has completely fixed my problems.

Thanks Kyle!

closed 2020-04-29 by BirdBare

kizniche, 2020-04-29:
Great. Thanks for reporting your findings. Hopefully the Pi 5 improves their Bluetooth.