kizniche/Mycodo #732: Test Mycodo installation on Raspbian Desktop

kizniche, 2020-01-13:
From #725, Mycodo may have issues installing on Raspbian Desktop. Need to review setup.log to determine the issue.

eap904, 2020-02-12:
I just did a fresh install with the Desktop, but did not yet try and add any inputs, the problem I had was not being able to access via VNC viewer, which is currently the only / best way to access my machines remotely

kizniche, 2020-02-13:
VNC is easily enabled through raspi-config

eap904, 2020-02-14:
Yes, I can do that, but I'm unable to login to the VNC cloud service via command line

samsixtysix, 2020-05-06:
Installed fine for me; no errors in install log and I can get data on the live screen; but, I have a problem renaming Outputs/Inputs.

If I try to change the default name of any Output it behaves as if I have tried to reorder the outputs (top one gives an error, lower ones give 'success reorder output' ).

Inputs toggle activate/deactivate when I try to rename them.

In both cases the name does not change from the default.

Functions are the same but Methods can be named.

I will try and see if a fresh install helps.

kizniche, 2020-05-06:
Never heard of this issue and not really sure how it would even be possible. What version Mycodo and what browser and browser version?

samsixtysix, 2020-05-06:
Mycodo 8.4.0
RPI3b Latest Raspbian
Tried accessing from a laptop running Win10Pro, Brave, Chrome then IE
Also desktop running Win7Pro Chrome
All the same result.

Since it's a new install I will just redo it and let you know if the problem is still there; unless there is anything you want me to check first before wiping the SD card?

samsixtysix, 2020-05-06:
Ok, now working - think it was user error as I was expecting it to save the name change when I entered it in the name field but it requires expanding the item and clicking save (which I wasn't doing because I have not decided on pin assignment yet and it won't save without an assigned pin).

Sorry for wasting your time.  The good news is it did install fine on Raspbian desktop.

kizniche, 2020-05-07:
No problem. There's probably room for improving help text or implementing a tutorial of sorts for new users. It's hard for me to think about what it's like to experience the Mycodo interface for the first time, so I appreciate the feedback. I'm also glad to hear it installs fine on the desktop version of Raspbian. Thanks.

samsixtysix, 2020-05-07:
Well just from a newbie usability viewpoint: 

Could you make the name field editable only when the item is expanded?
or
Make the name field non editable all the time, but have a second editable name field when expanded?

It's a minor point but if I can edit the field I expect it to retain the changes without having to go on to expand the item, configure the pin then click save.  Probably for most users it's a non issue but I wanted to name all my inputs and outputs before going through and fully configuring each one.

If it had simply ignored the edits then I probably would have figured it out sooner but when I entered the new name it confusingly tried to reorder the outputs/activate the input.

Please take this as hopefully constructive feedback and not criticism!

kizniche, 2020-05-09:
Just restyled the Output page and among the changes, I used your suggestion to move the Name field.

closed 2020-06-01 by kizniche