kizniche/Mycodo #733: Paho MQTT upgrade break previous disable funtionality

jlb-qi, 2020-01-13:
### Describe the problem/bug

In previous versions of paho mqtt package, setting the keep_alive variable to 0  was an acceptable input. With a recent upgrade, this causes Mycodo to be unable to connect to the mosquitto broker.

### Versions:

 - Mycodo Version: 8.2.2
 - Raspberry Pi Version: 4
 - Raspbian OS Version: Buster

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Go to Setup > Data
2. Setup/Create an MQTT connection OR activate an existing one
3. Activate the input with debug logging on
4. See error in the logs

### Expected behavior

A connection with moquitto broker is establish when correct input fields are complete.

### Additional context

I've updated the code for the mqtt_paho.py file here. https://gist.github.com/jlb-qi/61826ff817ff969df6b7d21a13502bd2

I have tested it with the latest release version of Mycodo 8.2.2

kizniche, 2020-01-13:
What is the specific fix?

jlb-qi, 2020-01-15:
I just changed the default value from 0 to 60. Previously 0 would have meant the loop would have been kept alive indefinitely. I had a quick look to see if there was any form validation function that would prevent 0 from being input but I couldn't find anything within the code base. Happy to be proven wrong though.

kizniche, 2020-01-16:
Made keepalive default to 60, added check to prevent saving a keepalive value <= 0. This will be incorporated in the next release.

closed 2020-01-27 by kizniche