kizniche/Mycodo #735: Issue with Custom Input with Custom Unit

not5, 2020-01-22:
Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue.

### Describe the problem/bug

Trying at add a custom input that has a custom input.

### Versions:

 - Mycodo Version: 8.2.2
 - Raspberry Pi Version: 3B+
 - Raspbian OS Version: Buster Desktop

### Reproducibility

I've created a custom input (attached), but getting an error when going to the input page saying that the custom input ("resistance") was not found in the dictionary, although I've created it.

[bme680.py.txt](https://github.com/kizniche/Mycodo/files/4097375/bme680.py.txt)
You'll have to convert the file back to python (from text). Github doesn't allow .py extension

kizniche, 2020-01-22:
Isn't the Gas sensor measurement IAQ, not resistance? See pages 8 and 9 of https://www.bosch-sensortec.com/media/boschsensortec/downloads/datasheets/bst-bme680-ds001.pdf

You also need to add the unit. In this module's case, that would be "Ohms", though I would stick to the singular "Ohm" as that's the current convention with units.

not5, 2020-01-22:
I did create the custom unit. ID: "resistance"; Unit Name: "resistance; Unit: "Ohms". You're correct though that the convention should be singular. I can make that change.

You're right that the measurement is IAQ, but that's only if you use Bosch's Software Environmental Cluster. The python examples that I found just pulled a resistance value from the sensor. I was planning on using the history of the resistance value to trend the indoor air quality... high resistance = Bad; middle resistance = OK; low resistance = Good; etc.

kizniche, 2020-01-22:
Can you provide the exact error?

not5, 2020-01-22:
`Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2446, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1951, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restplus/api.py", line 584, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1820, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1949, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1935, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 1979, in page_data
    devices_1wire_w1thermsensor=devices_1wire_w1thermsensor)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 140, in render_template
    ctx.app,
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 120, in _render
    rv = template.render(context)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/asyncsupport.py", line 76, in render
    return original_render(self, *args, **kwargs)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 1008, in render
    return self.environment.handle_exception(exc_info, True)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 780, in handle_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/_compat.py", line 37, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/data.html", line 3, in top-level template code
    {% set help_page = ["data-acquisition", dict_translation['data']['title']] %}
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 1005, in render
    return concat(self.root_render_func(self.new_context(vars)))
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/data.html", line 22, in root
    <input type="hidden" name="form-name" value="addSensor">
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/layout.html", line 335, in root
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/data.html", line 70, in block_body
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/data_options/input.html", line 269, in root
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/data_options/input_options/TEMPLATE_OPTIONS_Measurements_Convert.html", line 52, in root
    <div>
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 411, in getitem
    return obj[argument]
jinja2.exceptions.UndefinedError: 'collections.OrderedDict object' has no attribute 'resistance'`

kizniche, 2020-01-22:
In addition to crating a unit, did you create the measurement?

kizniche, 2020-01-22:
I need to add a check when importing an Input Module to return an error if any units/measurements are missing and for the user to first add them.

Did creating a resistance measurement fix the issue?

kizniche, 2020-01-22:
Also, once you test the module to work, I can add it as a built-in Input.

not5, 2020-01-23:
I added the measurement "resistance" and assigned it units "Ohms" (I will fix the actual names later). Still same error.

kizniche, 2020-01-23:
It's a bug and shouldn't happen, so adding the ability to delete a specific input from a page other than the input config page is unnecessary. I'll test in the next day or two and see what's wrong. You can always export your settings database before testing a new input module, then if anything goes wrong, just import your saved database and it's back to a working state.

kizniche, 2020-01-23:
The custom input import is a very new feature, so treat it as experimental until all the bugs are ironed out.

kizniche, 2020-01-23:
Well, all of this is experimental, really. It's a rough development road when I'm pretty much the only developer.

not5, 2020-01-23:
You're so right- really appreciate all the work you're doing on this project. The work product definitely speaks for itself.

No rush on my end- just trying to navigate the process to add additional sensors to mycodo. The additional logic check for "does this measurement/unit exist?" when adding a custom input seems to be a good idea though.

As it stands, mycodo is super versatile and bug free from a daily driver stand point. I can envision numerous uses for the backend (and frontend) that you've developed.

kizniche, 2020-01-23:
In reviewing the issue, notice that there is a "safe name" created for the unit and measurement names you enter. These are what need to be used in Input Modules. So, the issue came down to you using a capital O in Ohms. See my screenshot, below. Using lowercase letters, there is no issue. Though, preventing a user from importing a module that includes units/measurements that don't exist will still need to be added, but there's nothing currently preventing the system from working.

[image]

not5, 2020-01-23:
Thanks for the resolution! I'll make that change and report back if there's anything, but for now I'm closing this and marking as resolved.

kizniche, 2020-01-23:
Just added a check for the units/measurements of the Input being imported to exist.

[image]

Also added checks that a unit/measurement can't be deleted or modified (name changed) if an Input depends on it.

kizniche, 2020-01-23:
Also, please leave the issue open next time if there's a code fix that needs to still be implemented. If I wasn't working on this and put it off for later, I might forget if the issue wasn't still open.

kizniche, 2020-01-23:
Also just added the ability to create measurement/unit IDs with uppercase letters (by adding the input field "ID") to the measurement/unit creation/mod sections. Now you could create a unit with the ID "Ohms" now if you wanted to.

kizniche, 2020-01-23:
I'm reopening this because I'd like to know if the input module you created works. If so, I'll add it to the built-in set. Let me know when you get around to testing.

kizniche, 2020-01-29:
Have you had a chance to test the input?

not5, 2020-01-29:
I don't have it working quite yet... still debugging. I'm getting an error on deactivating saying there's no stop() function. Any ideas?

kizniche, 2020-01-29:
What is the exact error?

kizniche, 2020-01-29:
Please provide log lines from activation until deactivation, including the activation line.

kizniche, 2020-01-30:
In reviewing the bme680 library, I found a few things wrong with your input module:

You are using an old version of the library (greater detail, below):

```python
    'dependencies_module': [
        ('pip-pypi', 'bme680', 'bme280'),
    ],
```

Should be:

```python
    'dependencies_module': [
        ('pip-pypi', 'bme680', 'pimoroni-bme280'),
        ('pip-pypi', 'smbus2', 'smbus2')
    ],
```

Initializing the library:

```python
            import bme680

            self.i2c_address = int(str(input_dev.i2c_location), 16)
            self.i2c_bus = input_dev.i2c_bus
            self.sensor = bme680(
                address=self.i2c_address,
                busnum=self.i2c_bus)
```

Should be ([ref](https://github.com/pimoroni/bme680-python/blob/3a48112445a012b4ec38105b750d27f9591cdc75/library/bme680/__init__.py#L30)):

```python
            import bme680
            from smbus2 import SMBus

            self.i2c_address = int(str(input_dev.i2c_location), 16)
            self.i2c_bus = input_dev.i2c_bus
            self.sensor = bme680.BME680(
                i2c_addr=self.i2c_address,
                i2c_device=SMBus(self.i2c_bus))
```

Here is the edited module: [bme680.py-2.txt](https://github.com/kizniche/Mycodo/files/4134971/bme680.py-2.txt)

You will first have to uninstall the current bme680 library from your Python environment, as this appears to be a non-maintained version ([ref, check latest release date](https://pypi.org/project/bme680/)) of the pimoroni library ([ref, check latest release date](https://pypi.org/project/pimoroni-bme280/)). Both of these PyPI libraries are built from the code on the github repository ([ref](https://github.com/pimoroni/bme680-python)), only one is up-to-date.

Here are the steps I would take to get the new library and input module installed:

1. Deactivate any BME680 inputs.
2. Uninstall the bme680 python module with ```~/Mycodo/env/bin/pip uninstall bme680```
3. Delete any BME680 Inputs.
4. Delete the imported BME680 input module.
5. Import the new BME680 input module.
6. Add a BME680 input, which will prompt you to install the pimoroni-bme280 library before being able to add the input.
7. Add the BME680 input module.

not5, 2020-02-01:
Kyle,

Thanks for taking a closer look at this. Most of the code changes from the above are spot on; however, the module that you lined to 'pimoroni-bme280' isn't for the same sensor, the bme**6**80.

I actually already caught most of what you pointed out after posting the initial issue and getting the measurement/units all sorted out.

So the error that I'm getting when deactivating is 

> Error: Could not deactivate Input controller with ID 6daff1f5-0911-473e-a79d-2cbd43ef96fa: 'NoneType' object has no attribute 'stop_input'

Which log should I be pulling to paste in here? The daemon log doesn't have any entries related to this input module.

kizniche, 2020-02-01:
Darn, sorry for that simple number oversight.

That error is the result of something in your input module causing an exception that is completely killing the input module thread, which is why there's no input to be deactivated. First, are you running your Input in debug mode? Also, do you have try/except blocks around your code to test for exceptions? I would suggest a t/e block with logging lines (e.g. self.logger.excpetion(1)). Also, that is only the deactivation line, which doesn't include the activation line and any log lines between, that I requested.

not5, 2020-02-01:
I got it working! There was a line that didn't have a "self." prefix to the sensor that was causing an error.

All the outputs are as expected, but I need to sample the gas_resistance for while to make sure it's giving me the kind of information I want.
The github repo for the sensor had an IAQ example code that used an average function, so I may have a companion "math" input with this.

Quick note: A logic check that I think needs to be in mycodo is if you try to use an I2C address that is already in use, there should be an error raised. I accidentally activated it on 0x76, and I have another sensor at this address.. it killed the other sensor and wouldn't allow me to re-enable it. I had to reboot.

[bme680.py.txt](https://github.com/kizniche/Mycodo/files/4142599/bme680.py.txt)

kizniche, 2020-02-01:
Great to hear!

I'd recommend integrating things like smoothing/averaging into the input with the use of the custom fields (text inputs, number inputs, dropdowns, check boxes) to enable/disable/configure input options. This way all users can benefit from functions users would commonly want to use with a particular input and preclude the use of another controller like a Math controller.

There are certain cases where multiple inputs may be reading from the same sensor, so that kind of check baked-in to Mycodo would break that functionality. I'd rather leave it up to the user to create a custom input if they want to make any restrictions like that.

not5, 2020-02-02:
So, it's not smoothing or averaging, I was wrong.

The MAX reading of data is perceived as the "best" IAQ and then you would compare measurements to this max to get a relative IAQ. Ideally, you'd have "fresh air" to set the max.

There is some compensation that needs to be done to account for humidity, which is a log function. I need to further research this though.

Here's a pretty good thread discussing: https://forums.pimoroni.com/t/bme680-observed-gas-ohms-readings/6608/15

kizniche, 2020-02-03:
I just pushed a commit that adds support for BME680. I also added the ability to select sensor options from the Input config (see screenshot, below). If you want to test importing the new [bme680.py](https://github.com/kizniche/Mycodo/blob/92029c08c20e779124f6e4f990027819e99e5b30/mycodo/inputs/bme680.py), let me know if it works.

[image]

kizniche, 2020-02-03:
Found a bug in the input, fixing and committing new file. Stand by.

kizniche, 2020-02-03:
Fixed a bug in the BME60 Input. Here's the latest [bme680.py](https://github.com/kizniche/Mycodo/blob/96654842c8cb48c7ea7dddc626366225805f0aad/mycodo/inputs/bme680.py) to test.

not5, 2020-02-04:
Working very nicely. Loving the flexibility with the custom options.

not5, 2020-02-09:
Did you change something additional in the base_controller.py / controller_conditional.py file?

I have an error loading the input now:

`FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/Mycodo/mycodo/user_python_code/conditional_e6cbd3fd-f61e-4e2e-9718-adc8515f1391.py'
2020-02-09 09:51:29,628 - ERROR - mycodo.controllers.controller_conditional_e6cbd3fd - Exception executing check_conditionals() on attempt 1 of 3. Waiting 10 seconds and trying again.
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 126, in attempt_execute
    func()
  File "/var/mycodo-root/mycodo/controllers/controller_conditional.py", line 168, in check_conditionals
    with open(self.file_run, 'r') as file:
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/Mycodo/mycodo/user_python_code/conditional_e6cbd3fd-f61e-4e2e-9718-adc8515f1391.py'
2020-02-09 09:51:34,374 - ERROR - mycodo.controllers.controller_input_e95d2eb3 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 427, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-02-09 09:51:39,678 - ERROR - mycodo.controllers.controller_conditional_e6cbd3fd - Exception executing check_conditionals() on attempt 2 of 3. Waiting 10 seconds and trying again.
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 126, in attempt_execute
    func()
  File "/var/mycodo-root/mycodo/controllers/controller_conditional.py", line 168, in check_conditionals
    with open(self.file_run, 'r') as file:
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/Mycodo/mycodo/user_python_code/conditional_e6cbd3fd-f61e-4e2e-9718-adc8515f1391.py'
2020-02-09 09:51:43,036 - ERROR - mycodo.controllers.controller_input_91de9bf0 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 427, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-02-09 09:51:49,303 - ERROR - mycodo.controllers.controller_input_e95d2eb3 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 427, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-02-09 09:51:49,727 - ERROR - mycodo.controllers.controller_conditional_e6cbd3fd - Exception executing check_conditionals() on attempt 3 of 3.
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 126, in attempt_execute
    func()
  File "/var/mycodo-root/mycodo/controllers/controller_conditional.py", line 168, in check_conditionals
    with open(self.file_run, 'r') as file:`

kizniche, 2020-02-09:
When a Conditional is saved, it creates a file with the conditional code (with additional code that makes it a complete module). This is the file that is executed when it's activated. There are only a few reasons why this file won't exist anymore:

1. It wasn't successfully moved during an upgrade.
2. It was deleted.
3. A restore was performed and it wasn't properly restored.
4.?

All you have to do to recreate this file is save your conditional.

not5, 2020-02-09:
All good now. Thanks!

not5, 2020-02-11:
Sorry to reopen this, for the BME680 input, can you add one last custom option: set_temp_offset value. I noticed that my temperature is higher than it should be, which after some research needs to be compensated for with a temperature offset. I could do this outside of the input with a math input, but since bme680.py has the offset built in, it makes sense to add a custom option to the input itself.

kizniche, 2020-02-11:
Done

closed 2020-02-11 by kizniche