kizniche/Mycodo #736: jinja2.exceptions.UndefinedError: 'each_id_and_measure' is undefined when creating a

dylandn, 2020-01-27:
Mycodo 8.2.2
If I add a Guage widget and set the measurement to any value from a PID controller, the dashboard crashes with this error. Adding a Gauge widget and setting measurement to a sensor input works fine, there's something breaky with PId controller .

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2446, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1951, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restplus/api.py", line 584, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1820, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1949, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1935, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 804, in page_dashboard
    y_axes=y_axes)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 140, in render_template
    ctx.app,
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/templating.py", line 120, in _render
    rv = template.render(context)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/asyncsupport.py", line 76, in render
    return original_render(self, *args, **kwargs)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 1008, in render
    return self.environment.handle_exception(exc_info, True)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 780, in handle_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/_compat.py", line 37, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard.html", line 3, in top-level template code
    {% set help_page = ["dashboard", dict_translation['dashboard']['title']] %}
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 1005, in render
    return concat(self.root_render_func(self.new_context(vars)))
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard.html", line 22, in root
    <script type="text/javascript" src="/static/js/modules/solid-gauge.js"></script>
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/layout.html", line 335, in root
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard.html", line 298, in block_body
    toastr['error'](btn_val.split("/")[0] + ": " + data);
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/pages/dashboard_options/display_gauge_angular.html", line 98, in root
    color: '{{colors_gauge_angular[each_widget.unique_id][n]['hex']}}'
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/jinja2/environment.py", line 430, in getattr
    return getattr(obj, attribute)
jinja2.exceptions.UndefinedError: 'each_id_and_measure' is undefined

kizniche, 2020-01-27:
Thanks for the bug reports. I pushed fixes for both of them. Can you [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) and test that they are both working? Since your dashboard(s) are in a messed up state, first delete all dashboards from the diagnostic page, then go to the default dashboard that is created after the dashboards are deleted, and try adding a gauge.

dylandn, 2020-01-27:
All fixed up, thank you!

closed 2020-01-27 by kizniche