kizniche/Mycodo #737: Can't add new Dashboard

dylandn, 2020-01-27:
Mycodo 8.2.2, 

Related to my previously created issue, in my quest to diagnose what's wrong with my dashboards, I went to  Configure->Diagnostics->Delete All Dashboard Elements, and now when I attempt to add a new dashboard:

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2446, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1951, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restplus/api.py", line 584, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1820, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1949, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1935, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 543, in page_dashboard_add
    dashboard_id = utils_dashboard.dashboard_add()
  File "/home/pi/Mycodo/mycodo/mycodo_flask/utils/utils_dashboard.py", line 42, in dashboard_add
    new_dash.name = '{} {}'.format(TRANSLATIONS['dashboard']['title'], last_dashboard.id + 1)
AttributeError: 'NoneType' object has no attribute 'id'

kizniche, 2020-01-27:
I forgot to update the diagnostics functions when I added support for multiple dashboards, so this likely will break your config. I'll fix this, and you and upgrade to master, redelete your dashboards from the config menu, and start with an empty dashboard.

dylandn, 2020-01-27:
Updated to master, works fine. Thank you!

closed 2020-01-27 by dylandn