kizniche/Mycodo #739: Integration with Home Assistant and OpenHab

cohe4ko, 2020-02-04:
What scenario should be used for such integration?
My instance of MyCodo has many sensors connected, how can I use the data from sensors in Home Assistant?
I thought that I can use MQTT broker. Is it the correct way?

kizniche, 2020-02-04:
I'm sure there are a number of ways. Perhaps the simplest would be the API: https://github.com/kizniche/Mycodo/blob/master/mycodo-manual.rst#rest-api

cohe4ko, 2020-02-04:
I tried to find how to get the last measure of sensor input, but I can't find anything suitable.
Could you, please, give me an example of a curl request on how to get the last measurement of a particular sensor.

kizniche, 2020-02-04:
In Issue #714 you are missing the unit, but I see you have the unit name in this URL. However, you should be using the unit ID rather than the name. In this case, it should be "C" for Celsius. You can go to Configure -> Measurements to see all the unit IDs.

closed 2020-02-04 by cohe4ko

lalebarde, 2023-02-25:
I am investigating on my side for MQTT.