kizniche/Mycodo #740: Error with BMP180

cohe4ko, 2020-02-04:
after upgrade to the current version 8.2.3 BMP180 is throwing errors in log  

### Versions:

 - Mycodo Version: 8.2.3
 - Raspberry Pi Version: 3B+

### Logs

2020-02-04 14:14:31,060 - ERROR - mycodo.controllers.controller_input_f50efe72 - initialize_variables() Exception: [Errno 121] Remote I/O error
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 74, in run
    self.initialize_variables()
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 338, in initialize_variables
    self.measure_input = input_loaded.InputModule(self.input_dev)
  File "/home/pi/Mycodo/mycodo/inputs/bmp180.py", line 63, in __init__
    self.bmp = BMP085.BMP085(busnum=self.i2c_bus)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Adafruit_BMP/BMP085.py", line 69, in __init__
    self._load_calibration()
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Adafruit_BMP/BMP085.py", line 72, in _load_calibration
    self.cal_AC1 = self._device.readS16BE(BMP085_CAL_AC1)   # INT16
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Adafruit_GPIO/I2C.py", line 200, in readS16BE
    return self.readS16(register, little_endian=False)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Adafruit_GPIO/I2C.py", line 177, in readS16
    result = self.readU16(register, little_endian)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Adafruit_GPIO/I2C.py", line 164, in readU16
    result = self._bus.read_word_data(self._address,register) & 0xFFFF
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Adafruit_PureIO/smbus.py", line 187, in read_word_data
    ioctl(self._device.fileno(), I2C_RDWR, request)
OSError: [Errno 121] Remote I/O error
2020-02-04 14:14:31,065 - INFO - mycodo.controllers.controller_input_f50efe72 - Activated in 282.2 ms
2020-02-04 14:14:31,066 - ERROR - mycodo.controllers.controller_input_f50efe72 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 427, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-02-04 14:14:46,000 - ERROR - mycodo.controllers.controller_input_f50efe72 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 427, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'

not5, 2020-02-09:
Try enabling and disabling the input as well as rebooting the backend. This happened to me after an upgrade once. It had to do with loss of i2c connection.

kizniche, 2020-02-09:
Nothing in the traceback indicates what the issue could be. Since nothing about Mycodo changed with regard to the BMP180, I would suspect you may have a hardware/software issue outside Mycodo. I'd recommend following @not5's recommendation of rebooting. If that doesn't fix it, perhaps try flashing a new test SD card and installing Mycodo again to see if the problem persists. I'd also remove all other I2C sensors and test your BMP180 in isolation on the Pi, or potentially on another device like an Arduino to determine if it's another sensor interfering with the BMP180 or an issue with the BMP180 itself.

closed 2020-02-26 by kizniche