kizniche/Mycodo #741: Api error

cohe4ko, 2020-02-04:
### Describe the problem/bug
Server error un such request: 
https://192.168.1.11/api/measurements/last/76305962-e363-4deb-a38c-78c12cd24b28/0c3ede1b-0999-4597-b57c-289233885197/0/10

### Versions:

 - Mycodo Version: 8.2.3
 - Raspberry Pi Version: 3B+

### Reproducibility
{
  "error": "Traceback (most recent call last):\n  File \"/home/pi/Mycodo/mycodo/mycodo_flask/api/measurement.py\", line 259, in get\n    unique_id, unit, channel, duration_sec=past_seconds)\n  File \"/home/pi/Mycodo/mycodo/utils/influx.py\", line 440, in read_influxdb_single\n    last_measurement = client.query(query).raw\n  File \"/home/pi/Mycodo/env/lib/python3.7/site-packages/influxdb/client.py\", line 450, in query\n    expected_response_code=expected_response_code\n  File \"/home/pi/Mycodo/env/lib/python3.7/site-packages/influxdb/client.py\", line 302, in request\n    raise InfluxDBClientError(response.content, response.status_code)\ninfluxdb.exceptions.InfluxDBClientError: 400: {\"error\":\"error parsing query: found 0c3ede1b, expected identifier at line 1, char 25\"}\n\n",
  "message": "An exception occurred"
}

kizniche, 2020-02-04:
You're missing the unit. I would recommend going to https://your.pi.IP.address/api/ and finding the endpoint you want to test, then clicking "Try it out" and entering the required information, and clicking execute.

kizniche, 2020-02-04:
I see your URL also has two unique IDs, when it should only have one.

cohe4ko, 2020-02-04:
I see your URL also has two unique IDs, when it should only have one. -> yes, I know this, I just show that there is no values check.

kizniche, 2020-02-04:
Here is the swagger API documentation and testing interface. It's a great tool to testing and learning about the API.

[image]

kizniche, 2020-02-04:
The API is relatively new, so sanity checks aren't fully covered. I'll add a check for the unit's existence and return an informative error if it doesn't exist.

kizniche, 2020-02-04:
Were you able to get the endpoint working?

cohe4ko, 2020-02-04:
Yes, this was my request 
curl -X GET "https://192.168.1.11/api/measurements/last/2df2d928-d7c1-4f28-9b32-158069952d42/C/0/100" -H "accept: application/vnd.mycodo.v1+json" -H "X-API-KEY: 8BSKgc3BiPaAh8Ws8I5fssVWCAvaabAkbA0YvjwRt3J0A1NVz9sguyP82F0uHT9D67HyKmlaGabJ4wZGEesUdGvFRMzo/lHu7OlZpQyTJGtcU3/VsWDMGnQi1r84Tgs71tSzqYMkSQmZfnLLew="

closed 2020-02-09 by kizniche