kizniche/Mycodo #743: PID unable to read measurement

not5, 2020-02-09:
###Describe the problem/bug

PID controller is unable to measure input. On the dashboard, the PID controller is reading max age exceeded, despite the measurement being less than max age.

### Versions:

 - Mycodo Version: 8.2.4
 - Raspberry Pi Version: 3B+
 - Raspbian OS Version: Buster Desktop

### Log Output

> 2020-02-09 16:23:29,682 - ERROR - mycodo.controllers.controller_pid_3eab43a0 - Exception while reading measurement from the influxdb database: No application found. Either work inside a view function or push an application context. See http://flask-sqlalchemy.pocoo.org/contexts/.
> Traceback (most recent call last):
>   File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/util/_collections.py", line 1030, in __call__
>     return self.registry[key]
> KeyError: 1815082080
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/var/mycodo-root/mycodo/controllers/controller_pid.py", line 644, in get_last_measurement
>     duration_sec=int(self.max_measure_age))
>   File "/var/mycodo-root/mycodo/utils/influx.py", line 476, in read_last_influxdb
>     measure=measure, duration_sec=duration_sec, value='LAST')
>   File "/var/mycodo-root/mycodo/utils/influx.py", line 428, in read_influxdb_single
>     test_unit_channel(unit, channel)
>   File "/var/mycodo-root/mycodo/utils/influx.py", line 717, in test_unit_channel
>     if unit not in add_custom_units(Unit.query.all()):
>   File "/var/mycodo-root/env/lib/python3.7/site-packages/flask_sqlalchemy/__init__.py", line 519, in __get__
>     return type.query_class(mapper, session=self.sa.session())
>   File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/orm/scoping.py", line 78, in __call__
>     return self.registry()
>   File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/util/_collections.py", line 1032, in __call__
>     return self.registry.setdefault(key, self.createfunc())
>   File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/orm/session.py", line 3234, in __call__
>     return self.class_(**local_kw)
>   File "/var/mycodo-root/env/lib/python3.7/site-packages/flask_sqlalchemy/__init__.py", line 136, in __init__
>     self.app = app = db.get_app()
>   File "/var/mycodo-root/env/lib/python3.7/site-packages/flask_sqlalchemy/__init__.py", line 982, in get_app
>     'No application found. Either work inside a view function or push'
> RuntimeError: No application found. Either work inside a view function or push an application context. See http://flask-sqlalchemy.pocoo.org/contexts/.
> 2020-02-09 16:23:29,684 - DEBUG - mycodo.controllers.controller_pid_3eab43a0 - Last measurement unsuccessful. Turning outputs off.

kizniche, 2020-02-09:
This issue is the result of 4e9c3460b6f4b65de921c18ada1529f29ada49fa

I'll revert the change and make a new bugfix release.

kizniche, 2020-02-09:
v8.2.5 should fix this issue. Thanks for the bug report.

closed 2020-02-09 by kizniche