kizniche/Mycodo #744: PID Dashboard Setpoint Incorrect

not5, 2020-02-10:
### Describe the problem/bug

The PID dashboard controller is showing the Setpoint as setpoint+1 for some reason.

Not critical, and I understand that the dashboard control is supplemental to everything.

kizniche, 2020-02-21:
I can't replicate this issue. Can you take screenshots of both your PID settings and the PID dashboard widget?

not5, 2020-02-22:
Sure thing, although I'm not sure what the screenshot tells you other than validating my issue.
[image]
[image]

kizniche, 2020-02-22:
Your settings indicate you have a 1 degree band set, which you didn't indicate indicate in the issue. This is the +1.

not5, 2020-02-22:
So the Band is added to the setpoint on the PID dashboard controller. Got it. Thanks!

closed 2020-02-22 by not5

kizniche, 2020-02-22:
Yes. Notice if you display the setpoint on your graph it will also be this value (at present). The setpoint will change based on the band and what stage the PID is at, either set to the higher or lower band (you can also display the setpoint with the upper and lower band on the graph). In theory this should work, but I haven't used the band feature in a while. If you think it's behaving improperly, let me know.

kizniche, 2020-02-22:
> I'm not sure why the setpoint spans previous.

When the PID runs every period, it will calculate the new P, I, and D values along with the other values. Regardless of whether there's an actual output value that will meet the thresholds to turn on an output, the PID values will be saved to the influx database (setpoint, control variable, etc.). The widgets merely query this database and pull whatever data is there. So, the setpoint is saved every period the PID is active.

kizniche, 2020-03-05:
I actually got the setpoint for PIDs with bands to display correctly on dashboard widgets. Now, if the band is in effect, the widget setpoint will change to that value, and once the measurement passes that min or max, it will change to the actual [unbanded] setpoint until the measurement passes a band. This is the behavior is should have had all along. Now that I've been using PIDs with bands, it started bothering me I couldn't immediately see what band was in effect. This will be out in the next release.

not5, 2020-03-05:
Very slick implementation. Thanks!