kizniche/Mycodo #746: ow-shell should not be a dependency for 1wire devices

drgrumpy, 2020-02-14:
- Mycodo Version: 8.25
 - Raspberry Pi Version: Pi Zero W
 - Raspbian OS Version: Buster Lite

Having been running my set up for a couple of years (with version 6.45) without issue, I am just building a new system, naturally I just installed the latest version 8.25....
BUT, previously accessing the 1 wire bus was not dependent on owfs or ow-shell, and I cannot see why it should now be needed, nor can I identify in the changelog when this dependency was added... clearly I have been away too long and I missed something ? 
Especially on a pi-zero I don't want to be installing stuff that is not necessary, ow-shell is just adding unnecessary bloat.

So to avoid installing it I added manually installed w1thermsensor, then  
I commented out the:
` ('apt', 'ow-shell', 'ow-shell')`
line in ds18b20.py and restarted the daemon, I can then select my ds18b20s without any need for ow-shell.

kizniche, 2020-02-14:
Check out https://github.com/kizniche/Mycodo/issues/582

The DS18B20 has a library option to select either w1thermsensor or ow-shell. I should have created two different input modules rather than using one where the library is selected. It's a bit late to do that now without potentially breaking users' systems, so what I would recommend if you really don't want ow-shell is to copy the DS18B20 input module from the inputs directory, change the name to something unique, then import it as a custom input.

drgrumpy, 2020-02-15:
Yes I saw #582. And if I understand that post correctly, seems to be a sledgehammer to crack a nut....  a shame I didn't pick it up...

I do not think owfs was needed there, nor is w1-gpio kernel module, i.e. do not add/comment out:
`# dtoverlay=w1-gpio`
in config.txt,  this is only needed if the pi is controlling the 1-wire bus, and may interfere with ds2482 in my experience)

That (sheepwalk) setup uses a ds2482-100 connected to i2c, this is what I have been using for some years and is better, more reliable way to control long networks, there is already a kernel module built into the kernel:
`sudo modprobe ds2482`
to make permanent add to /etc/modules

- Use i2detect to find out address:
`i2detect - y 1 `
(default is at address 18 I think)
- Let the system know where it is:
`sudo sh -c "echo ds2482 0x18 > /sys/bus/i2c/devices/i2c-1/new_device"`
 - To ensure loaded at boot add to /etc/rc.local and make it executable:
`echo ds2482 0x18 > /sys/bus/i2c/devices/i2c-1/new_device`

Then any devices connected to the ds2482 just show up on the w1 bus
`ls /sys/bus/w1/devices/`

kizniche, 2020-02-15:
Actually, I realized I can do an update that splits the inputs into two (each using each library). Then update the user's database based on the selected library to use the appropriate input.

I didn't know enough about what the issue was, and there's usually no harm in adding the use of an additional library/method to access a sensor. But I really should have made 2 inputs. I believe back then I didn't have the library listed in the input selection title, so I wasn't using multiple libraries for the same sensor. Now there are several sensors with multiple libraries that can be chosen to access it, each with its own separate module.

kizniche, 2020-02-17:
I just pushed a commit that splits the inputs, one for each library.

kizniche, 2020-02-26:
Closed with release of 8.3.0

closed 2020-02-26 by kizniche