kizniche/Mycodo #747: Error setting DS18b20 resolution

drgrumpy, 2020-02-14:
- Mycodo Version: 8.25
 - Raspberry Pi Version: Zero W
 - Raspbian OS Version: Buster Lite

Clean install (with one mod to mycodo see #746)  first ds18b20 sensor added, attempt to set sensor resolution on configuration page:

Error: Error while setting resolution of sensor with ID 28-000006daa134: 'NoneType' object has no attribute 'device'

I can successfully set the resolution directly from the terminal by writing to the device

kizniche, 2020-02-17:
I just pushed a potential fix for this. I haven't tested it, thought.

kizniche, 2020-02-17:
In the meantime, you can set the resolution from the Input settings. This resolution setting from the calibration page merely makes the resolution setting persistent, whereas the Input settings is not persistent and is set every time the input is activated.

drgrumpy, 2020-02-20:
Great thanks for sorting

closed 2020-02-20 by drgrumpy