kizniche/Mycodo #750: TSL2561 - no data

drgrumpy, 2020-02-21:
Version 8.25

Connect TSL2561 to i2c
Add new tsl2561 input
Activate
Result -> no data

Log:
2020-02-21 18:03:47,618 - ERROR - mycodo.inputs.tsl2561_0b63ec45 - Error: unsupported operand type(s) for >>: 'float' and 'int'
Traceback (most recent call last):
  File "/home/drgrumpy/Mycodo/mycodo/inputs/tsl2561.py", line 85, in get_measurement
    self.value_get(0), self.value_get(1)))
  File "/var/mycodo-root/env/lib/python3.7/site-packages/tsl2561/tsl2561.py", line 255, in _calculate_lux
    channel0 = (broadband * chScale) >> TSL2561_LUX_CHSCALE
TypeError: unsupported operand type(s) for >>: 'float' and 'int'

So looks like the issue lies with the tsl2561.py library file

A quick looking at the library on github: the latest version is 3.4 with change "make fully typed"
My guess is we need to make sure that we use the latest version ?

How do I install into the appropriate place for mycodo?
Can I just copy into /var/mycodo-root/env/lib/python3.7/site-packages/tsl2561/ ?

kizniche, 2020-02-21:
Thanks for the bug report. I just pushed a potential fix. Can you test the latest [tsl2561.py](https://github.com/kizniche/Mycodo/blob/af704bb38731b600c4ec90b56b983c25917a2004/mycodo/inputs/tsl2561.py)?

drgrumpy, 2020-02-21:
I just tried.... by simply replacing the old tsl261.py file with the new version.....
and restarted the backend daemon
(not sure if this is the best way... been a while)

now get into a big mess, and mycodoflask/gunicorn does not seem to start/fails, so I can't access via web:

> $ tail -n 30 /var/log/mycodo/mycodo.log
>     self.initialize_variables()
>   File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 248, in initialize_variables
>     self.dict_inputs = parse_input_information()
>   File "/var/mycodo-root/mycodo/utils/inputs.py", line 100, in parse_input_information
>     input_custom = load_module_from_file(full_path, 'inputs')
>   File "/var/mycodo-root/mycodo/utils/modules.py", line 17, in load_module_from_file
>     module_custom = importlib.util.module_from_spec(spec)
>   File "<frozen importlib._bootstrap>", line 580, in module_from_spec
> AttributeError: 'NoneType' object has no attribute 'loader'
> 2020-02-21 20:30:14,804 - INFO - mycodo.controllers.controller_input_0b63ec45 - Activated in 314.1 ms
> 2020-02-21 20:30:14,807 - ERROR - mycodo.controllers.controller_input_0b63ec45 - loop() Error
> Traceback (most recent call last):
>   File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 88, in run
>     self.loop()
>   File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 154, in loop
>     if (now > self.next_measurement and
> TypeError: '>' not supported between instances of 'float' and 'NoneType'
> 2020-02-21 20:30:14,810 - INFO - mycodo.daemon - All activated Input controllers started
> 2020-02-21 20:30:14,817 - INFO - mycodo.daemon - All activated Math controllers started
> 2020-02-21 20:30:14,814 - ERROR - mycodo.controllers.controller_input_0b63ec45 - Run Error
> Traceback (most recent call last):
>   File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 94, in run
>     time.sleep(self.sample_rate)
> TypeError: an integer is required (got type NoneType)
> 2020-02-21 20:30:14,820 - INFO - mycodo.daemon - All activated PID controllers started
> 2020-02-21 20:30:14,827 - INFO - mycodo.daemon - All activated LCD controllers started
> 2020-02-21 20:30:14,824 - INFO - mycodo.controllers.controller_input_0b63ec45 - Deactivated in 0.1 ms
> 2020-02-21 20:30:14,829 - INFO - mycodo.daemon - All activated Custom controllers started
> 2020-02-21 20:30:15,334 - INFO - mycodo.daemon - Mycodo daemon started in 13.922 seconds
> 2020-02-21 20:30:15,350 - INFO - mycodo.daemon - 38.46 MB RAM in use

And:
sudo systemctl status mycodoflask
● mycodoflask.service - gunicorn daemon for Mycodo flask web user interface
   Loaded: loaded (/home/drgrumpy/Mycodo/install/mycodoflask.service; enabled; vendor preset: 
   Active: failed (Result: exit-code) since Fri 2020-02-21 20:25:21 GMT; 20min ago
  Process: 445 ExecStart=/var/mycodo-root/env/bin/gunicorn --workers 1 --worker-class gthread 
 Main PID: 445 (code=exited, status=3)

Feb 21 20:25:17 phspi06 gunicorn[445]:     input_custom = load_module_from_file(full_path, 'in
Feb 21 20:25:17 phspi06 gunicorn[445]:   File "/home/drgrumpy/Mycodo/mycodo/utils/modules.py",
Feb 21 20:25:17 phspi06 gunicorn[445]:     module_custom = importlib.util.module_from_spec(spe
Feb 21 20:25:17 phspi06 gunicorn[445]:   File "<frozen importlib._bootstrap>", line 580, in mo
Feb 21 20:25:17 phspi06 gunicorn[445]: AttributeError: 'NoneType' object has no attribute 'loa
Feb 21 20:25:17 phspi06 gunicorn[445]: [2020-02-21 20:25:17 +0000] [555] [INFO] Worker exiting
Feb 21 20:25:21 phspi06 gunicorn[445]: [2020-02-21 20:25:21 +0000] [445] [INFO] Shutting down:
Feb 21 20:25:21 phspi06 gunicorn[445]: [2020-02-21 20:25:21 +0000] [445] [INFO] Reason: Worker
Feb 21 20:25:21 phspi06 systemd[1]: mycodoflask.service: Main process exited, code=exited, sta
Feb 21 20:25:21 phspi06 systemd[1]: mycodoflask.service: Failed with result 'exit-code

drgrumpy, 2020-02-21:
Reverting to the original tsl2516.py still won't start....
I guess the issue is now that I should have disabled the tsl2561 sensor first...
How can do that from command line?

kizniche, 2020-02-21:
You can stop the daemon with ```sudo service mycodo stop```, then overwrite the file with ```cp -rf ~/tsl2561.py ~/Mycodo/mycodo/inputs/tsl2561.py```, then start the daemon with ```sudo service mycodo start```

kizniche, 2020-02-21:
I have this sensor, so I'll test with it later.

drgrumpy, 2020-02-21:
So I should do:
mv ~/Mycodo/databases/mycodo.db ~/Mycodo/databases/mycodo.db.backup
sudo systemctl start mycodoflask

drgrumpy, 2020-02-21:
Tried that and still broken...

drgrumpy, 2020-02-21:
> You can stop the daemon with `sudo service mycodo stop`
> 
> then overwrite the file with `cp -rf ~/tsl2561.py ~/Mycodo/mycodo/inputs/tsl2561.py`
> 
> then start the daemon with `sudo service mycodo start`
That's what I tried to start with....

kizniche, 2020-02-21:
> So I should do:
> mv ~/Mycodo/databases/mycodo.db ~/Mycodo/databases/mycodo.db.backup
> sudo systemctl start mycodoflask

The frontend (flask) has no impact on the backend (daemon) that actually conducts the measurements, so stopping/starting the frontend will have no effect on anything. Also, messing the settings database (mycodo.db) will also have no effect, as it merely stores settings.

kizniche, 2020-02-21:
> I have this sensor, so I'll test with it later.

Actually, I just checked and I don't have that sensor here (it's out on a farm), so I won't be able to test myself.

drgrumpy, 2020-02-21:
That is what I originally thought... I guess I was clutching at straws...

kizniche, 2020-02-21:
I won't be able to figure out what your issue is unless you can provide the excerpt of the daemon log that starts at the input activation, through to the end of the last error for the input.

drgrumpy, 2020-02-21:
So  it seems I have perhaps another separate issue: mycodoflask fails to start:
Several reboots and same thing:
Feb 21 22:34:19 phspi06 gunicorn[441]: [2020-02-21 22:34:19 +0000] [548] [ERROR] Exception in worker process
Feb 21 22:34:19 phspi06 gunicorn[441]: Traceback (most recent call last):
Feb 21 22:34:19 phspi06 gunicorn[441]:   File "/home/drgrumpy/Mycodo/env/lib/python3.7/site-packages/gunicorn/arbiter.py",
Feb 21 22:34:19 phspi06 gunicorn[441]:     worker.init_process()
Feb 21 22:34:19 phspi06 gunicorn[441]:   File "/home/drgrumpy/Mycodo/env/lib/python3.7/site-packages/gunicorn/workers/gthr
Feb 21 22:34:19 phspi06 gunicorn[441]:     super().init_process()
Feb 21 22:34:19 phspi06 gunicorn[441]:   File "/home/drgrumpy/Mycodo/env/lib/python3.7/site-packages/gunicorn/workers/base
Feb 21 22:34:19 phspi06 gunicorn[441]:     self.load_wsgi()
Feb 21 22:34:19 phspi06 gunicorn[441]:   File "/home/drgrumpy/Mycodo/env/lib/python3.7/site-packages/gunicorn/workers/base
Feb 21 22:34:19 phspi06 gunicorn[441]:     self.wsgi = self.app.wsgi()
Feb 21 22:34:19 phspi06 gunicorn[441]:   File "/home/drgrumpy/Mycodo/env/lib/python3.7/site-packages/gunicorn/app/base.py"
Feb 21 22:34:19 phspi06 gunicorn[441]:     self.callable = self.load()
Feb 21 22:34:19 phspi06 gunicorn[441]:   File "/home/drgrumpy/Mycodo/env/lib/python3.7/site-packages/gunicorn/app/wsgiapp.
Feb 21 22:34:19 phspi06 gunicorn[441]:     return self.load_wsgiapp()
Feb 21 22:34:19 phspi06 gunicorn[441]:   File "/home/drgrumpy/Mycodo/env/lib/python3.7/site-packages/gunicorn/app/wsgiapp.
Feb 21 22:34:19 phspi06 gunicorn[441]:     return util.import_app(self.app_uri)
Feb 21 22:34:19 phspi06 gunicorn[441]:   File "/home/drgrumpy/Mycodo/env/lib/python3.7/site-packages/gunicorn/util.py", li
Feb 21 22:34:19 phspi06 gunicorn[441]:     mod = importlib.import_module(module)
Feb 21 22:34:19 phspi06 gunicorn[441]:   File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
Feb 21 22:34:19 phspi06 gunicorn[441]:     return _bootstrap._gcd_import(name[level:], package, level)
Feb 21 22:34:19 phspi06 gunicorn[441]:   File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
Feb 21 22:34:19 phspi06 gunicorn[441]:   File "<frozen importlib._bootstrap>", line 983, in _find_and_load
Feb 21 22:34:19 phspi06 gunicorn[441]:   File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
Feb 21 22:34:19 phspi06 gunicorn[441]:   File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
Feb 21 22:34:19 phspi06 gunicorn[441]:   File "<frozen importlib._bootstrap_external>", line 728, in exec_module
Feb 21 22:34:19 phspi06 gunicorn[441]:   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
Feb 21 22:34:19 phspi06 gunicorn[441]:   File "/home/drgrumpy/Mycodo/mycodo/start_flask_ui.py", line 11, in <module>

drgrumpy, 2020-02-21:
So to be clear, I have reverted to the original tsl2561.py, and mycodo.db before I attempted to test the fix. 
And after starting deamon:
2020-02-21 23:06:44,591 - INFO - mycodo.daemon - Mycodo daemon v8.2.5 starting
2020-02-21 23:06:47,979 - INFO - mycodo.controllers.controller_output - Output f24d6e27 (GPIO) setup on pin 17 and turned OFF (OFF=LOW)
2020-02-21 23:06:47,984 - INFO - mycodo.controllers.controller_output - Output 88295c3b (GPIO) setup on pin 27 and turned OFF (OFF=LOW)
2020-02-21 23:06:47,989 - INFO - mycodo.controllers.controller_output - Output 7a081430 (GPIO) setup on pin 22 and turned OFF (OFF=LOW)
2020-02-21 23:06:47,994 - INFO - mycodo.controllers.controller_output - Output d16aa0f3 (GPIO) setup on pin 10 and turned OFF (OFF=LOW)
2020-02-21 23:06:48,153 - INFO - mycodo.controllers.controller_output - Output 3330a2eb (PWM) setup on pin 19
2020-02-21 23:06:48,156 - INFO - mycodo.controllers.controller_output - Setting Output f24d6e27 startup state to OFF
2020-02-21 23:06:48,159 - INFO - mycodo.controllers.controller_output - Setting Output 88295c3b startup state to OFF
2020-02-21 23:06:48,162 - INFO - mycodo.controllers.controller_output - Setting Output 7a081430 startup state to OFF
2020-02-21 23:06:48,165 - INFO - mycodo.controllers.controller_output - Setting Output d16aa0f3 startup state to OFF
2020-02-21 23:06:48,170 - INFO - mycodo.controllers.controller_output - Activated in 1051.6 ms
2020-02-21 23:06:48,675 - INFO - mycodo.daemon - All activated Conditional controllers started
2020-02-21 23:06:48,677 - INFO - mycodo.daemon - All activated Trigger controllers started
2020-02-21 23:06:51,862 - ERROR - mycodo.controllers.controller_input_8e7783a7 - initialize_variables() Exception: 'NoneType' object has no attribute 'loader'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 74, in run
    self.initialize_variables()
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 248, in initialize_variables
    self.dict_inputs = parse_input_information()
  File "/var/mycodo-root/mycodo/utils/inputs.py", line 100, in parse_input_information
    input_custom = load_module_from_file(full_path, 'inputs')
  File "/var/mycodo-root/mycodo/utils/modules.py", line 17, in load_module_from_file
    module_custom = importlib.util.module_from_spec(spec)
  File "<frozen importlib._bootstrap>", line 580, in module_from_spec
AttributeError: 'NoneType' object has no attribute 'loader'
2020-02-21 23:06:51,911 - INFO - mycodo.controllers.controller_input_8e7783a7 - Activated in 718.0 ms
2020-02-21 23:06:51,917 - ERROR - mycodo.controllers.controller_input_8e7783a7 - loop() Error
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 88, in run
    self.loop()
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 154, in loop
    if (now > self.next_measurement and
TypeError: '>' not supported between instances of 'float' and 'NoneType'
2020-02-21 23:06:51,943 - ERROR - mycodo.controllers.controller_input_8e7783a7 - Run Error
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 94, in run
    time.sleep(self.sample_rate)
TypeError: an integer is required (got type NoneType)
2020-02-21 23:06:51,965 - INFO - mycodo.controllers.controller_input_8e7783a7 - Deactivated in 0.1 ms

kizniche, 2020-02-21:
Not sure what's wrong. I just released 8.3.0. Try upgrading and see if that fixes the issue.

drgrumpy, 2020-02-21:
OK. Without the web interface, what do I need to run to upgrade?

drgrumpy, 2020-02-22:
Okay got it run:
upgrade_download.sh
then
upgrade_install.sh

kizniche, 2020-02-22:
```sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_download.sh upgrade-release-major 8```

drgrumpy, 2020-02-22:
Had to make executable first, in progress...

kizniche, 2020-02-22:
If you preface the script with ```/bin/bash``` in my command, it doesn't need to be executable.

drgrumpy, 2020-02-22:
Of course, my bad.
So the upgrade reported success, and mycodo daemon is apparently running without error....
But mycodoflask still fails to start this time with a different error, and without anything in the system logs:

> 
>  sudo systemctl status -l mycodoflask
> ● mycodoflask.service - gunicorn daemon for Mycodo flask web user interface
>    Loaded: loaded (/home/drgrumpy/Mycodo/install/mycodoflask.service; enabled; vendor preset: enabled)
>    Active: failed (Result: signal) since Sat 2020-02-22 01:12:36 GMT; 3min 58s ago
>   Process: 3095 ExecStart=/var/mycodo-root/env/bin/gunicorn --workers 1 --worker-class gthread --threads 2 --timeout 300 --pid /var/run
>  Main PID: 3095 (code=killed, signal=SEGV)
> 
> Feb 22 01:12:36 phspi06 systemd[1]: Started gunicorn daemon for Mycodo flask web user interface.
> Feb 22 01:12:36 phspi06 systemd[1]: mycodoflask.service: Main process exited, code=killed, status=11/SEGV
> Feb 22 01:12:36 phspi06 systemd[1]: mycodoflask.service: Failed with result 'signal'

Weird that it was working fine up until I tested the new tsl file (no other changes made at that time)

kizniche, 2020-02-22:
What's the output after the following commands:

```bash
cd ~/Mycodo/mycodo
sudo /var/mycodo-root/env/bin/gunicorn \
--workers 1 \
--worker-class gthread \
--threads 2 \
--timeout 300 \
--pid /var/run/mycodoflask.pid \
--bind unix:/var/run/mycodoflask.sock start_flask_ui:app
```

drgrumpy, 2020-02-22:
After shutdown overnight and then a reboot mycodoflask is now working.
Also I can confirm that with 8.3.0 I am getting data from the TSL2561
But now I have database mismatch issue....
I tried running
`~/Mycodo/env/bin/alembic history`
but this results in Python errors:

> Traceback (most recent call last):
>   File "/home/drgrumpy/Mycodo/env/bin/alembic", line 8, in <module>
>     sys.exit(main())
>   File "/home/drgrumpy/Mycodo/env/lib/python3.7/site-packages/alembic/config.py", line 577, in main
>     CommandLine(prog=prog).main(argv=argv)
>   File "/home/drgrumpy/Mycodo/env/lib/python3.7/site-packages/alembic/config.py", line 571, in main
>     self.run_cmd(cfg, options)
>   File "/home/drgrumpy/Mycodo/env/lib/python3.7/site-packages/alembic/config.py", line 551, in run_cmd
>     **dict((k, getattr(options, k, None)) for k in kwarg)
>   File "/home/drgrumpy/Mycodo/env/lib/python3.7/site-packages/alembic/command.py", line 430, in history
>     _display_history(config, script, base, head)
>   File "/home/drgrumpy/Mycodo/env/lib/python3.7/site-packages/alembic/command.py", line 399, in _display_history
>     base=base or "base", head=head or "heads"
>   File "/home/drgrumpy/Mycodo/env/lib/python3.7/site-packages/alembic/script/base.py", line 215, in walk_revisions
>     head, base, inclusive=True, assert_relative_length=False
>   File "/home/drgrumpy/Mycodo/env/lib/python3.7/site-packages/alembic/script/revision.py", line 765, in _iterate_revisions
>     requested_lowers = self.get_revisions(lower)
>   File "/home/drgrumpy/Mycodo/env/lib/python3.7/site-packages/alembic/script/revision.py", line 321, in get_revisions
>     resolved_id, branch_label = self._resolve_revision_number(id_)
>   File "/home/drgrumpy/Mycodo/env/lib/python3.7/site-packages/alembic/script/revision.py", line 501, in _resolve_revision_number
>     self._revision_map
>   File "/home/drgrumpy/Mycodo/env/lib/python3.7/site-packages/alembic/util/langhelpers.py", line 230, in __get__
>     obj.__dict__[self.__name__] = result = self.fget(obj)
>   File "/home/drgrumpy/Mycodo/env/lib/python3.7/site-packages/alembic/script/revision.py", line 123, in _revision_map
>     for revision in self._generator():
>   File "/home/drgrumpy/Mycodo/env/lib/python3.7/site-packages/alembic/script/base.py", line 112, in _load_revisions
>     script = Script._from_filename(self, vers, file_)
>   File "/home/drgrumpy/Mycodo/env/lib/python3.7/site-packages/alembic/script/base.py", line 906, in _from_filename
>     module = util.load_python_file(dir_, filename)
>   File "/home/drgrumpy/Mycodo/env/lib/python3.7/site-packages/alembic/util/pyfiles.py", line 98, in load_python_file
>     module = load_module_py(module_id, path)
>   File "/home/drgrumpy/Mycodo/env/lib/python3.7/site-packages/alembic/util/compat.py", line 173, in load_module_py
>     spec.loader.exec_module(module)
>   File "<frozen importlib._bootstrap_external>", line 728, in exec_module
>   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
>   File "/home/drgrumpy/Mycodo/databases/alembic/versions/2e416233221b_add_daemon_debug_mode_config_option.py", line 15, in <module>
>     from databases.alembic_post_utils import write_revision_post_alembic
>   File "/home/drgrumpy/Mycodo/databases/__init__.py", line 1, in <module>
>     nstall
> NameError: name 'nstall' is not defined

Deleted and then re-created an empty __init__.py file
Then successfully followed the wiki instructions to upgrade db stamp

drgrumpy, 2020-02-25:
Fixed in version 8.3.0

closed 2020-02-25 by drgrumpy