kizniche/Mycodo #752: DS18B20 Unable to select device id when adding new device

drgrumpy, 2020-02-29:
Version 8.3
Recently installed on Pi Zero
Buster Lite

Data 
Add new DS18B20 (w1thermosensor) output
1-wire device id dropdown is an empty list
Expect to see one or more connected sensors listed

Sensors are connected and are detected by the system at boot, and are detected by w1thermosensor test script..

kizniche, 2020-02-29:
I think I know what this bug is. You'll have to use the owshell input for the time being until I can make a fix and new release.

kizniche, 2020-02-29:
Turned out to be an easier fix than I anticipated. If you want to make that edit, restart your frontend, and test, let me know if it worked.

drgrumpy, 2020-02-29:
I made the change.... but now mycodoflask fails to start:

> 
> sudo /var/mycodo-root/env/bin/python /var/mycodo-root/mycodo/start_flask_ui.py --debug
> Traceback (most recent call last):
>   File "/var/mycodo-root/mycodo/start_flask_ui.py", line 11, in <module>
>     from mycodo.mycodo_flask.app import create_app
>   File "/var/mycodo-root/mycodo/mycodo_flask/app.py", line 32, in <module>
>     from mycodo.mycodo_flask import routes_page
>   File "/var/mycodo-root/mycodo/mycodo_flask/routes_page.py", line 90, in <module>
>     from mycodo.mycodo_flask.forms import forms_input
>   File "/var/mycodo-root/mycodo/mycodo_flask/forms/forms_input.py", line 34, in <module>
>     class InputAdd(FlaskForm):
>   File "/var/mycodo-root/mycodo/mycodo_flask/forms/forms_input.py", line 36, in InputAdd
>     dict_inputs = parse_input_information()
>   File "/var/mycodo-root/mycodo/utils/inputs.py", line 100, in parse_input_information
>     input_custom = load_module_from_file(full_path, 'inputs')
>   File "/var/mycodo-root/mycodo/utils/modules.py", line 17, in load_module_from_file
>     module_custom = importlib.util.module_from_spec(spec)
>   File "<frozen importlib._bootstrap>", line 580, in module_from_spec
> AttributeError: 'NoneType' object has no attribute 'loader'

kizniche, 2020-02-29:
This looks like the same error you had previously, unrelated to my changes. In fact, my change could not possibly cause any errors to appear. You seem to have done something else to cause this issue.

kizniche, 2020-02-29:
See https://github.com/kizniche/Mycodo/issues/750#issuecomment-589875185

drgrumpy, 2020-02-29:
Okay thanks, I find the inner workings of python a bit obscure....
The only things I did were:
1. To have a look at input.html (but it is reverted to original, and presumably this would not cause a python issue)
2. I also had mess with ds18b20.py, but then reverted back to the original, again I would not expect this to cause the issue, however after some digging around, it seems that when restart mycodo with
`sudo systemctl restart mycodo `
the related pyc been recompiled (as expected) but is now owned by root with permissions of 640
(why are all the other pyc files owned by mycodo and not root as above?)
Presumably mycodoflask is unable to access....
So changed the owner and group of the pyc file to mycodo and set it to 644
and now mycodoflask runs.....

BUT sadly your fix has not fixed the issue!

I do not want to mess with owfs - this has broken my set ups before, so how can I downgrade/revert to the combined file and edit out the one line as I did before?

kizniche, 2020-03-01:
I see I didn't fix it correctly. I'll push a fix shortly.

kizniche, 2020-03-01:
Also, I'm not sure if it makes a difference, but I always use ```sudo service mycodoflask restart``` instead of ```systemctl``` and never have issues with permissions.

drgrumpy, 2020-03-01:
So after that latest edit, the ds18b20 addresses are now showing up....

But unlike before I am now seeing ds2438 address also showing up (prefix of 26 below):
$ ls /sys/bus/w1/devices/
26-000001ca9633  28-000005f35388  28-000005f3e46c  w1_bus_master1
Maybe not a good idea to remove the prefix to avoid confusion?

Yes I realised that I needed to restart flask, but I had previously restarted mycodo after making changes to ds18b20.py (and then reverting)
As I understand on the 'service' command is a wrapper that works out what init system is running, so on a systemd based system is actually running 'systemctl', so it should not make any difference.
Also 'systemctl can do more that 'service'

I really appreciate the speed with which you are able to deal with these fixes.

kizniche, 2020-03-01:
The method of listing 1-wire devices is to simply convert the files in /sys/bus/w1/devices/ to a list for the ID selection in the Input options. I don't see a point in parsing any further, as this is just added complexity that is likely not needed. The user should know what devices are connected and which one to select. I'm often torn with development because I feel I make Mycodo ***too*** approachable.

drgrumpy, 2020-03-02:
OK, I certainly wouldn't want to add complexity.
How easy would it be to display the prefix, i.e. the 26- or the 28- ?

kizniche, 2020-03-15:
Sure. Just added and will be available in the next release.

frodus17, 2020-03-17:
I'm also not able to see any 1-wire. My SD card crashed, so I got a new card, installed buster, installed Mycodo, installed both of the 1-wire dependencies, and neither of them work. Nothing has changed hardware-wise. 

It seems every time something happens, I'm back to square 1, trying to debug several things why it won't work.... sort of problematic, as my fridge goes down each time and I spend 3-4 hours getting it back running.

kizniche, 2020-03-17:
This fix has been committed but a release hasn't yet been made. If you [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) it should work.

frodus17, 2020-03-17:
Thanks, back up running.

kizniche, 2020-03-17:
This may be of interest: Boot a Raspberry Pi4 with an SSD to make it reliable and fast

https://youtu.be/gp6XW-fGVjo

closed 2020-03-23 by kizniche