kizniche/Mycodo #753: Unable to access dashboard

drgrumpy, 2020-03-01:
Version 8.3 (plus ds18b20 fix)
Pi Zero W
Buster Lite

Recent install, still trying to set up.

Unable to access dashboard, simply clicked on default dasboard in data menu.
This is the first time I tried to access the dashboard, results in the following error:

Error (Full Traceback):
Traceback (most recent call last):
  File "/home/drgrumpy/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1246, in _execute_context
    cursor, statement, parameters, context
  File "/home/drgrumpy/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/default.py", line 588, in do_execute
    cursor.execute(statement, parameters)
sqlite3.OperationalError: no such column: widget.stops

I guess I have missing column in the sqlite db table.

kizniche, 2020-03-01:
That error is indicative of something wrong with the database. I would guess you're not running a vanilla install of Mycodo. I would recommend [Upgrading to Master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) and seeing if that fixes your issue.

drgrumpy, 2020-03-02:
Since upgrading from 8.25 to 8.3 the only changes I made were to fix issues as reported in other threads.
I followed the instruction for Upgrading to Master and it proceeded without errors.
But following the upgrade, I still get the same error.
Can I manually add the column in the database?

drgrumpy, 2020-03-02:
More info:
```
$ ~/Mycodo/env/bin/alembic history
0a8a5eb1be4b -> a91e6a71028f (head), Add stops to gauge widgets
20174b717c2e -> 0a8a5eb1be4b, split DS18B20 input module for separate libraries
69960a0722a7 -> 20174....
```b717c2e, add widget options

If I understand correctly, this suggests that the change has already been applied?

drgrumpy, 2020-03-05:
I used sqlite3 to check the actual structure of the widget table, and even though the upgrade reported to no errors, it had not added the extra stops column....
so I added it manually
ALTER TABLE widget ADD COLUMN stops int
and hey presto, dasboard now works.

closed 2020-03-23 by kizniche