kizniche/Mycodo #754: Cannot Upgrade

jacopienaar, 2020-03-01:
When trying to access the upgrade page I get this message:

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 2446, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1951, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1820, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1949, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1935, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_admin.py", line 515, in admin_upgrade
    is_internet=is_internet)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/templating.py", line 140, in render_template
    ctx.app,
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/templating.py", line 120, in _render
    rv = template.render(context)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/environment.py", line 1008, in render
    return self.environment.handle_exception(exc_info, True)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/environment.py", line 780, in handle_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/_compat.py", line 37, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/admin/upgrade.html", line 2, in top-level template code
    {% set active_page = "upgrade" %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/layout.html", line 262, in top-level template code
    {%- block body %}{% endblock -%}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/admin/upgrade.html", line 126, in block "body"
    You are {{releases_behind}} release{% if releases_behind > 1 %}s{% endif %} behind the latest version.
TypeError: unorderable types: NoneType() > int()

kizniche, 2020-03-01:
I suspect your Pi doesn't have internet access. Either that or it can't communicate with Github for some reason. Do you get a response when you execute "ping yahoo.com" when logged into a terminal on your Pi?

jacopienaar, 2020-03-01:
Thanks for the super fast response - this is not urgent  - just noticed it erlier when wanting to upgrade. I do get a response when pining yahoo:

[image]

kizniche, 2020-03-01:
What version Mycodo are you running?

jacopienaar, 2020-03-02:
Version 7.6.3

kizniche, 2020-03-02:
And running on Raspbian Stretch, Jessie, or Buster?

jacopienaar, 2020-03-02:
I will have to check - set it up last year about this time - probably Jessie. Not at home now so can't check the version.  I'll update Raspbian to the latest version later tonight (I'm in New Zealand) to see if that fixes it and get back to you.

kizniche, 2020-03-02:
You're running a very old version, and you're likely not on Buster, so you won't be able to upgrade to the latest version anyway. To get the latest version, you'll have to flash Raspbian Buster, then install Mycodo.

jacopienaar, 2020-03-02:
I'm running stretch.  If I have to do a clean install I guess that resolves the issue and it can be closed. I have a mushroom grow room running Mycodo and don't want to risk updating at least till the weekend when I have more time available to fix issues when something goes pear shaped.

closed 2020-03-23 by kizniche