kizniche/Mycodo #755: TCA9548A I2C multiplexer setup issues

bndaquino, 2020-03-08:
Hi Kyle,

I'm trying to use a TCA95481 I2C multiplexer to get readings from my sensors, but I'm stuck on setting up new I2C buses (from these instructions in the manual: https://github.com/Theoi-Meteoroi/GPIO-pca9548).

After following the instructions, I can see both that the proper two lines were added to /boot/config.txt and that two files i2c_gpio-pca9548.dtbo and i2c-mux-pca9545a.dtbo were added to /boot/overlays. However, it seems like nothing happens after I reboot my pi. `dmesg` returns nothing about adding new multiplexed buses and `sudo i2cdetect -l` only shows i2c-3 and i2c-1. System Information in Mycodo also only shows buses 1 and 3.

Any ideas on what I might be doing wrong? I'm pretty new to raspberry pi so I'm not sure what else to try. If it matters, I was running Mycodo version 8.3.0.

Thanks in advance.

kizniche, 2020-03-08:
What version Pi are you using?

bndaquino, 2020-03-08:
3 B+

kizniche, 2020-03-08:
@theoi-meteoroi discovered an issue with the latest Raspbian release (Buster) and his device tree overlay: https://kylegabriel.com/forum/general-discussion/i2c-multiplexer-support/#post-209. It works with Stretch, so a quick fix would be to install the latest Stretch release of Raspbian, then install Mycodo. Though, Mycodo will disable the API if running Stretch due to a Python version incompatibility. Not an ideal solution, but it will work.

Have you looked elsewhere for support for this multiplexer with Buster?

bndaquino, 2020-03-09:
I haven't been able to find anything at all online about using this multiplexer with Buster. I'll try switching to Stretch, and I'll let you know how it goes. Thanks again.

Theoi-Meteoroi, 2020-03-26:
It seems I have some quiet time, so I looked at this in detail and built a system to debug this problem.   I may have figured out how to make it work on the RPi 4.
This was tested with Buster 2020-02-13 on a Rpi 4 4MB v1.2 

```[   11.013609] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   11.028070] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  2 2020 23:30:41 version 7.45.202 (r724630 CY) FWID 01-72f6ece2
[   11.041913] random: crng init done
[   11.041926] random: 7 urandom warning(s) missed due to ratelimiting
[   11.102055] i2c i2c-7: Added multiplexed i2c bus 8
[   11.102403] i2c i2c-7: Added multiplexed i2c bus 9
[   11.102746] i2c i2c-7: Added multiplexed i2c bus 10
[   11.103074] i2c i2c-7: Added multiplexed i2c bus 11
[   11.103408] i2c i2c-7: Added multiplexed i2c bus 12
[   11.103805] i2c i2c-7: Added multiplexed i2c bus 13
[   11.104139] i2c i2c-7: Added multiplexed i2c bus 14
[   11.108851] i2c i2c-7: Added multiplexed i2c bus 15
[   11.108869] pca954x 7-0070: registered 8 multiplexed busses for I2C switch pca9548
[   11.110512] i2c i2c-1: Added multiplexed i2c bus 16
[   11.110857] i2c i2c-1: Added multiplexed i2c bus 17
[   11.111186] i2c i2c-1: Added multiplexed i2c bus 18
[   11.111517] i2c i2c-1: Added multiplexed i2c bus 19
[   11.111835] i2c i2c-1: Added multiplexed i2c bus 20
[   11.112162] i2c i2c-1: Added multiplexed i2c bus 21
[   11.112488] i2c i2c-1: Added multiplexed i2c bus 22
[   11.112795] i2c i2c-1: Added multiplexed i2c bus 23
[   11.112809] pca954x 1-0070: registered 8 multiplexed busses for I2C switch pca9548
[   13.292710] uart-pl011 fe201000.serial: no DMA platform data
[   13.569893] 8021q: 802.1Q VLAN Support v1.8 

This configuration has a 9548 mux on the built-in hardware i2c-arm ( I2C-1 ) and another 9548 mux on a GPIO i2c bus ( I2C-7 )  

This may still be a hack - I found the kernel symbol I needed to use was missing when I specified the bus number for the GPIO bus.  I did have to make some edits to the dts file and recompile a dtbo - so I'll have to release a source code change.

Theoi-Meteoroi, 2020-03-26:
Forgot to mention - they think they fixed the clock stretching issue in the ARM hardware however I have read there still is some YMMV in there.  The behavior has definitely changed.   
I probably have to fix the rarely used 9545 mux dts for GPIO as well ( I have several of the SDL mux boards ) but that may wait for a bit while I catch up on other projects.

kizniche, 2020-03-28:
Just letting everyone know @Theoi-Meteoroi released an update to his overlay that adds support for the Pi 4 v1.2. No others have been tested.

https://kylegabriel.com/forum/general-discussion/i2c-multiplexer-support/#post-259

closed 2020-05-01 by kizniche