kizniche/Mycodo #757: PID controller ignores Min Duty Cycle

neongrau81, 2020-03-16:
### Describe the problem/bug

**SETUP**
I am running a PID controller to control the temperature in my grow room.
As input I use an "_Average (Last, Multiple Channels)_" controller that uses the temperature values of several temp sensors.

[image]

As output I run a _PWM Linux Command_
curl -s -m 10 http://192.168.179.29/cm?cmnd=Dimmer%20((duty_cycle))

The command tells an ESP8266 that runs Tasmota to control the fan-speed

**PROBLEM**
As I want to run my exhaust fan constantly at a very slow pace (~10%) I set a _Min duty cycle_ of 10.
And because my exhaust fan is a little oversized for my room, I as well set a _Max duty cycle_ of 80.

It seems that mycodo in my case ignores the _Min duty cycle_ setting because my ESP8266 receives a lot of calls that tell the fan to run at 0%
[image]

The Max duty cycle although seems to work fine
[image]

I used wireshark to verify,if either the ESP8266 misinterprets a valid call from mycodo, or if mycodo itself sends out the 0% to the ESP. According to WS the mycodo sends out calls to http://192.168.179.29/cm?cmnd=Dimmer%200 

### Versions:

 - Mycodo Version: 8.0.3
 - Raspberry Pi Version: 3B+
 - Raspbian OS Version: Buster 

### Expected behavior:

As long as PID controller generates an output =< _Min ducty cycle_, the value gets overwritten by the _Min duty cycle_

kizniche, 2020-03-16:
This is the expected behavior. It is modeled after the duration output behavior. A minimum duration or duty cycle does not mean the output is alway outputting at least that amount every PID cycle, but rather the PID output must be at least that amount for the output to be activated for that duration or duty cycle. If it were always at least outputting the min value, it would cause interference with the alternate output if the PID was attempting to regulate in the opposite direction.

That being said, I do see a benefit for PWM Outputs to have an additional option, "Only Between Min/Max" (default: True), that will only allow the duty cycle to be set to a value between min and max.

neongrau81, 2020-03-16:
Wow, thanks a lot for the super fast response and the clarification !

The additional option would be great, so I can be sure to always have a "negative pressure" in the room by a constantly running exhaust fan. -> less skunky flavor for the neighbors :-)

Thanks for the great work you do with mycodo, I absolutely love this system

kizniche, 2020-03-16:
I just pushed a commit that adds this feature. If you would like to test before the release, you can [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch).

closed 2020-03-23 by kizniche

neongrau81, 2020-04-04:
Hey @kizniche I was a bit shy upgrading to master as my system was running smooth after a while of tweaking ;-)

I updated to 8.4.0 two weeks ago and started testing.

[image]
I have checked the "Always min" checkbox, what (as far as I understood) should prevent the output to send any value below this threshold. But it seems like the checkbox is ignored, my exhaust fan still goes down to 0% every now and than.

Do you have any idea?

kizniche, 2020-04-04:
Can you provide a graph showing the following series when the 0 duty cycle occurs: temperature input, duty cycle of the output as well as the P, I and D measurements, PID setpoint,

neongrau81, 2020-04-05:
Here you go :-) Sorry, I had to wait for a night cycle to show a proper graph.
[image]
The little spike of the _PWM up_ on the right is due to a second PID controller that uses the same output. As I don't have a dehumidifier, I use my exhaust fan as well to reduce the humidty as soon as it exceeds a certain threshold (56% humidity).