kizniche/Mycodo #758: triggers does not work

drgrumpy, 2020-03-23:
I have two triggers set: to turn on/off my fan relay when the pwm output from cooling pid is greater than 10% or less than 9%
The trigger is not turning the relay on...

The PWM value is now at 95% and this is what I see in the logs....

> 2020-03-23 15:04:28,443 - ERROR - mycodo.daemon - Could not turn output on: unsupported operand type(s) for +=: 'float' and 'NoneType'
> Traceback (most recent call last):
>   File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 751, in output_on
>     trigger_conditionals=trigger_conditionals)
>   File "/var/mycodo-root/mycodo/controllers/controller_output.py", line 512, in output_on_off
>     current_amps = self.current_amp_load()
>   File "/var/mycodo-root/mycodo/controllers/controller_output.py", line 1319, in current_amp_load
>     amp_load += each_output_amps
> TypeError: unsupported operand type(s) for +=: 'float' and 'NoneType'

drgrumpy, 2020-03-23:
Version etc.:
Mycodo Version: 8.3.0
Python Version: 3.7.3 (default, Dec 20 2019, 18:57:59) [GCC 8.3.0]

kizniche, 2020-03-23:
Thanks for the bug report. I just pushed a fix and will be releasing 8.4.0 shortly.

closed 2020-03-23 by kizniche

drgrumpy, 2020-03-23:
Just made the change and restarted backend - seems to be fixed