kizniche/Mycodo #759: Graph shows PWM as >100%

drgrumpy, 2020-03-23:
Version 8.3

Not critical

When displaying PWM output Duty Cycle I expect the values to be displayed are in the range 0 to 100%, actual value displayed is up to 450%. Presumably some kind of scaling issue?

kizniche, 2020-03-23:
Displayed where?

drgrumpy, 2020-03-23:
Sorry dashboard

kizniche, 2020-03-23:
There are numerous places on a dashboard duty cycle can be displayed. Can you be specific?

drgrumpy, 2020-03-23:
Actually I think I am perhaps misunderstanding:
The value when I hover over it is shown as e.g. 475% but it is actually truncated to 100% for display.
I'll post a screenshot...

drgrumpy, 2020-03-23:
[image]

drgrumpy, 2020-03-23:
So perhaps it is as intended?

kizniche, 2020-03-23:
It's intended if you have data grouping enabled.

drgrumpy, 2020-03-23:
Apologies if I missed it, but I couldn't find or figure out what data grouping is for/supposed to do? Now I understand, and indeed if turn off data grouping I get the 'old' (?) expected behaviour

closed 2020-03-23 by drgrumpy

kizniche, 2020-03-23:
Data grouping has always been on by default. The option to disable it was only added this past year.