kizniche/Mycodo #760: Graph display ignores custom colours

drgrumpy, 2020-03-23:
Version 8.3

Custom colours seem to be ignored and replaced with another colour.

Custom colours checkbox is checked and I select various colours for outputs, pid settings etc.
click save, then some of the colours seem to get changed...
I think it only affects outputs.....(i.e. inputs seem to keep their assigned colours)

drgrumpy, 2020-03-23:
This is in the dashboard.

drgrumpy, 2020-03-23:
See screenshot in #759 where the colours of the outputs are not the same as those that I selected.

kizniche, 2020-03-23:
I can't replicate this behavior. Can you supply a screenshot of both your graph as well as your configuration for that graph?

drgrumpy, 2020-03-23:
This is before saving:
[image]

drgrumpy, 2020-03-23:
This is after saving the above colour settings...
[image]
SSR1 should be red, but has taken the colour above....
Lights is red but should be pale yellow
PWM should be blue but is pale yellow..
I am wondering if there is a for loop that is indexed incorrectly ?

drgrumpy, 2020-03-23:
And if I make a change to another value (like changing data grouping) and save again then I get another change to the colours....
[image]

kizniche, 2020-03-23:
I'm not understanding. The screenshots you sent (https://github.com/kizniche/Mycodo/issues/760#issuecomment-602744893 and https://github.com/kizniche/Mycodo/issues/760#issuecomment-602748253) have the correct colors from the configuration matching the graph series.

kizniche, 2020-03-23:
The issue is the ordering. The colors are correctly being applied to the graph series in the exact same order as in the configuration, however PIDs come before Outputs in the configuration and PIDs come after Outputs in the graph series.

kizniche, 2020-03-23:
Fixed the order, now working on another issue: displaying the proper color from the database on the configuration page.

kizniche, 2020-03-23:
Found the last issue and pushed fixes. They'll be available in the next release.

drgrumpy, 2020-03-23:
Great, many thanks.

closed 2020-04-03 by kizniche