kizniche/Mycodo #761: Trigger fails to turn off relay

drgrumpy, 2020-03-23:
I have a trigger that should switch off relay if PWM value is less than 9%, 
but it does not work, I suspect because the PID output value has become NULL
In version 6 I had an additional condition to set switch off the relay if there was no PWM value, but now in version 8, this cannot be done in the trigger....
I guess I need to create a condition, but these are now more complicated than in V6, so it would be good to have an additional option for if the value is null or does not exist....

Sorry for raising all the issues at once... it's because the sun came out today!

drgrumpy, 2020-03-23:
If I understand correctly I can achieve the same effect as a trigger in a condition... i.e.  a trigger is a more limited specific condition?

kizniche, 2020-03-23:
Can you screenshot the configuration options?

drgrumpy, 2020-03-23:
That is tricky, as the dropdown disappears when activating screenshot:
[image]
The if dropdown options are: <, >, =

kizniche, 2020-03-23:
Found the bug. It wasn't triggering when a PWM was set to a 0 % duty cycle if the "Less Than" option was selected, even though 0 is less than 9. I just pushed a fix for recognizing 0 %.

drgrumpy, 2020-03-23:
Presume it more efficient to do in a single conditional with three if, elif statements, think I figured it out...

kizniche, 2020-03-24:
> Is it more efficient to do in a single conditional several if elif statements, think I figured it out...

It basically does the exact same thing. Your single conditional would require two trigger functions. Also, in your code you haven't accounted for when the measurement is between 9 and 10. You could change it to "if measurement < 9" and "elif measurement > 9".

drgrumpy, 2020-03-24:
Yes, at present I have two trigger functions. I have made the change to the code for the triggers, and the sun is out again, so they should get tested today. I was aware that there is a dead spot between 9 and 10, which was intentional as I presumed that this would mean the current status would be maintained.

drgrumpy, 2020-03-24:
So I have just been through a cycle of the cooling fan on/off via triggers in response to pwm levels, and can confirm that the fix works.

closed 2020-03-24 by drgrumpy