kizniche/Mycodo #762: Downsampling and retention policies for Influxdb

drgrumpy, 2020-04-02:
My perception (perhaps wrongly) is that influxdb is the weak point in the system, e.g. system hangs etc can often be tracked to influxdb stopping for some reason...
Currently as the systems runs.... the database just keeps getting bigger, and bigger...
I am currently sampling/recording data at 1 min intervals... so it rapidly accumulates, this makes extracting visualizing long term historical data slow (and risky to the system).
Whilst having the data at 1 min intervals is good for accurate pid control in the short term... for long term logging of conditions this is/may be unnecessary, e.g. long term having an hourly average, max and min may be perfectly adequate. 

This could be done with continuous queries and retention policies in influxdb...
see [](https://atomstar.tweakblogs.net/blog/17748/influxdb-retention-policy-and-data-downsampling)

I am planning to try and set this up 'by hand' but wondering what the implications will be?

Probably it is better to implement when the influxdb database is first created?

It is also tricky to figure out the data in the db because the device_ids are uninformative (are they random? for ds18b20s etc, could the id just use the actual address of the device as these are already unique?):

> C,channel=0,device_id=118e128d-01de-4523-9f59-4f3ac61de63c,measure=temperature
> C,channel=0,device_id=290c02ab-6452-4da5-bb76-ecc223209514,measure=temperature

kizniche, 2020-04-03:
For viewing data longer than a few days, I recommend using the synchronous graph page, as this was designed for displaying large amount of data with virtually no data or processing requirement, as only 700 data points per series are ever loaded onto one graph at a time.

Influxdb retention policies are something I never bothered with but could be appealing in certain circumstances. Even on my most active systems, I typically don't worry about space, as long as I have a decently-sized SD card in use (>8 GB min, > 16 GB preferred). Influxdb is generally good about minimizing db size, and I haven't run into the stability issues you mention.

If you find any interesting or useful configurations of influxdb, please share. I'm not that familiar with its settings, but I'm sure there are ways to improve performance, storage, etc.

All devices (Inputs, Outputs, Functions, LCDs, etc.) have a randomly-generated unique-ID when they're created.

https://github.com/kizniche/Mycodo/blob/72790a9f9010fe6eec85af3ed79523ab5390ac66/mycodo/databases/models/input.py#L8-L13

https://github.com/kizniche/Mycodo/blob/72790a9f9010fe6eec85af3ed79523ab5390ac66/mycodo/databases/__init__.py#L40-L42

For determining which ID is associated with each device, you can typically hover your mouse over the disabled button for the Input, Output, etc. and it will show the unique ID (and copy it to the clipboard if clicked).

> for ds18b20s etc, could the id just use the actual address of the device as these are already unique?

For consistency, no. All Inputs, regardless of what they are, use the same system for identification to the rest of the system. It wouldn't make sense for a particular sensor to deviate form this. It would make the code confusing and difficult to maintain, as well as a nightmare to refactor every other part of the system to recognize it. This is why there is the Name entry for each device, so you can give it a friendly name that you can use to identify one device from another. This name is typically used throughout the system wherever you go to distinguish devices.

drgrumpy, 2020-04-03:
Hi Kyle,
To some extent I was just thinking aloud, and thinking/hoping maybe perhaps someone else had already addressed these issues...

I am running on Pi Zeroes, and influxdb takes an age to load when booting (like 5 mins), and seems to use most of the mem and cpu, and is presumably writing most to the sd card.

Ultimately what I am wanting to do is to save the data (from experiments) elsewhere for data analysis and archiving... and trying to figure out the best way to automate this....

So I am clear that the asynch graphs can be used to visualise (and even download) summaries, but 
(a) there is no control over the time points
(b) it cannot be automated (?) to download the data, and would be very tedious to do manually on a regular basis.
(I cannot do retrospective summaries like hourly max/min) 

Appreciate the argument about device-id assignment, it was just a thought. Problem is that manually getting all the names (e.g. if more than just one or two sensors is a bit tedious, with ds18b20s I can potentially have dozens on a single bus) ...and the names are not stored in the influxdb database. In my arduino data loggers I actually use the maxim ids as default and create unique maxim style address ids for all my none-maxim sensors, according to some simple rules.

I'm also new to influxdb, and will be happy to share any insights...
In the end I guess I need to push a summarised backup of the influxdb to a remote server.... on a regular basis using a cron job.

Thanks as ever for bringing us Mycodo and your continuing work on it.

kizniche, 2020-04-03:
> Ultimately what I am wanting to do is to save the data

What about the Export Measurement Data as CSV feature on the Import/Export page? The /export endpoint can be automatically requested with the measurement and data rage at a regular interval by a script.

kizniche, 2020-04-03:
I could add an export endpoint to the API that would make the process of automating exporting easier.

drgrumpy, 2020-04-03:
Yes I looked at that, but that measurement data is at short time intervals, and results in separate file for each sensor ( I would then ultimately import back into a db to summarise....so might as well just export/copy the influxdb.... )
It seems more appropriate to use influxdb to handle the summary of time series data as that is what is designed for...
I suggest you don't do anything yet.... (it will be a while until I get to grips with influxdb....)

drgrumpy, 2020-04-03:
Short answer - I don't know.

The only thing I have used Pi zeroes for is running mycodo to control 'home made' growth cabinet or incubators. I have spent most of my life working with relational type MySQL databases (the data from my data loggers gets uploaded to MySQL), this is my first exposure to a 'NoSQL' database, but it does seem cumbersome.. (I have tended to work on the basis that it is more efficient to program in the language you already know, rather than having to learn something new in order to figure out if is any better, and there seem endless debates about which is best for particular purposes.... I think there is nothing perfect) .. although there may be some tuning that can be done...

kizniche, 2020-05-09:
I've been reading about [InfluxDB retention policies](https://docs.influxdata.com/influxdb/v1.8/guides/downsample_and_retain/), and it seems pretty straightforward to apply retention policies.

For instance, to create a new retention policy that only stores data for 26 weeks (half year), and make it the default retention policy for the Mycodo db, you can do this:

```
pi@tango:~ $ influx
Connected to http://localhost:8086 version 1.7.9
InfluxDB shell version: 1.7.9
> CREATE RETENTION POLICY "half_year" ON "mycodo_db" DURATION 26w REPLICATION 1 DEFAULT
```

not5, 2020-05-13:
@kizniche maybe a solution to the issue @drgrumpy is having is to not store any sensor data locally but instead using an MQTT publishing approach whereby the onus to record data is on whatever is subscribed to the MQTT channel that the data is published to. This would fix the issue of constant read/writes to the local SD card which can lead to data corruption. It would then be on the end user on how they chose to store (or not store) the measurements/data.

If data isn't stored locally, you wouldn't be able to use some of the functionality of mycodo.
A potential solution to this would be if mycodo had a back-end setting to call a non-localhost database. Now that I think about it, mycodo _could_ have a non-localhost influxdb setup from the get-go with a lightweight localhost database as a fall-back (for local settings like users etc).

drgrumpy, 2020-05-13:
Apologies for my lack of activity - too many other priorities...
So in response to @kizniche the approach I had in mind was something like:
- First downsample the data e.g. to hourly max, min, avg temp, humidity, light, etc., this data would be kept indefinitely and/or it would be pushed/uploaded to a server (if/when it is available)
- Then retain the recorded data locally for say a week/month ?
Probably this can all be done outside of Mycodo, i.e. by modifying the influxdb directly and creating appropriate daily/hourly cron job to check for server
For me the base system needs to be capable of working self-contained/stand-alone indefinitely, so the default should be local storage and the non-local is a fall-back or add on...

closed 2021-02-02 by kizniche