kizniche/Mycodo #763: new action to shutdown the Pi

leoneo, 2020-04-03:
Hi,
I think that it will be great to have a new action to shutdown properly the Pi before disconnect the power with minimal interface user.
I have add a switch on a GPIO and I read the GPIO state, I want if it is activated that it turn the Pi shutdown properly.  And I can disconnect the power withou risk.

Thanks

kizniche, 2020-04-03:
I was actually thinking about system restart/shutdown actions the other day, but didn't write it down. I'll add it in the next release. Thanks for the recommendation.

leoneo, 2020-04-03:
Great thanks Kyle

kizniche, 2020-04-03:
Although this should be able to be done with the Linux Command Action, it is simpler to use the System Restart/Shutdown Actions. I just added these two actions and they'll be available in the next release.

closed 2020-06-01 by kizniche