kizniche/Mycodo #765: Ds18b20 Max age exceeded

dutchnerd, 2020-04-12:
Version 8.4
Installed raspberry pi 3b
Buster Lite
Ds18b20 waterproof

I have install yesterday a new raspberry pi with buster lite. En install ds18b20 and in works until this morgin. And now I get max age exceeded.
This error do I get in the logs. I reboot the raspberry pi

2020-04-12 10:24:49,444 - ERROR - mycodo.inputs.ds18b20_de4f8b2d - InputModule raised an exception when taking a reading: float() argument must be a string or a number, not 'NoneType'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 127, in read
    self._measurements = self.get_measurement()
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20.py", line 92, in get_measurement
    self.value_set(0, temperature)
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 237, in value_set
    self.return_dict[chan]['value'] = float(value)
TypeError: float() argument must be a string or a number, not 'NoneType'
2020-04-12 10:24:53,604 - ERROR - mycodo.inputs.ds18b20_de4f8b2d - InputModule raised an exception when taking a reading: float() argument must be a string or a number, not 'NoneType'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 127, in read
    self._measurements = self.get_measurement()
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20.py", line 92, in get_measurement
    self.value_set(0, temperature)
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 237, in value_set
    self.return_dict[chan]['value'] = float(value)
TypeError: float() argument must be a string or a number, not 'NoneType'
2020-04-12 10:25:01,683 - ERROR - mycodo.inputs.ds18b20_e42411fe - InputModule raised an exception when taking a reading: float() argument must be a string or a number, not 'NoneType'
2020-04-12 10:25:01,685 - ERROR - mycodo.controllers.controller_input_e42411fe - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-04-12 10:25:08,564 - ERROR - mycodo.inputs.ds18b20_de4f8b2d - InputModule raised an exception when taking a reading: float() argument must be a string or a number, not 'NoneType'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 127, in read
    self._measurements = self.get_measurement()
  File "/home/pi/Mycodo/mycodo/inputs/ds18b20.py", line 92, in get_measurement
    self.value_set(0, temperature)
  File "/var/mycodo-root/mycodo/inputs/base_input.py", line 237, in value_set
    self.return_dict[chan]['value'] = float(value)
TypeError: float() argument must be a string or a number, not 'NoneType'
2020-04-12 10:25:08,566 - ERROR - mycodo.controllers.controller_input_de4f8b2d - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.

kizniche, 2020-04-12:
Is your sensor detected? You can list the detected devices in the w1 directory:

```bash
ls -l /sys/devices/w1\ bus\ master/
```

dutchnerd, 2020-04-12:
Yes. And with command line I get a reading. it’s working

kizniche, 2020-04-12:
So it's not working? This suggests it's a hardware issue, not a software issue.

I would suggest checking your connections. A loose connection can cause communication issues. Also, try to reduce your wire length as much as possible. If you're using long wires, you can also use 5 volts instead of 3.3 volts. The longer wires will increase voltage drop, but using a higher voltage will reduce the drop. This may or may not be safe for the Pi, but I've never experienced an issue doing this with my DS18B20s.

dutchnerd, 2020-04-12:
It’s working now. I think it’s the 3,3v it’s a 10m cable. And I have no loose connections

closed 2020-04-12 by dutchnerd