kizniche/Mycodo #766: Mycodo fails to start

drgrumpy, 2020-04-13:
### Versions:
 - Mycodo Version: 8.4
 - Raspberry Pi Version: zero w
 - Raspbian OS Version: Buster Lite

Mycodo fails to start:

Apr 13 22:23:38 phspi06 python[2604]:   File "<frozen importlib._bootstrap>", line 983, in _find_and_load
Apr 13 22:23:38 phspi06 python[2604]:   File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
Apr 13 22:23:38 phspi06 python[2604]:   File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
Apr 13 22:23:38 phspi06 python[2604]:   File "<frozen importlib._bootstrap_external>", line 724, in exec_module
Apr 13 22:23:38 phspi06 python[2604]:   File "<frozen importlib._bootstrap_external>", line 857, in get_code
Apr 13 22:23:38 phspi06 python[2604]:   File "<frozen importlib._bootstrap_external>", line 525, in _compile_bytecode
Apr 13 22:23:38 phspi06 python[2604]: ValueError: bad marshal data (invalid reference)
Apr 13 22:23:39 phspi06 systemd[1]: mycodo.service: Control process exited, code=exited, status=1/FAILURE
Apr 13 22:23:39 phspi06 systemd[1]: mycodo.service: Failed with result 'exit-code'.
Apr 13 22:23:39 phspi06 systemd[1]: Failed to start Mycodo server.

All been running fine for 3 weeks or so....
I upgraded from 8.3 to 8.4 about a week ago, and all seemingly ok....
but then I realised that the fix from #761 was not implemented in 8.4
so I manually made the changes to controller_output.py
and restarted the daemon...
However since then I have been unable to start with the errors above...

I thought it might be permissions issue as the controller_output was not getting compiled to pyc and the __pycache__ directory was owned by root.....
so I changed all the perms to mycodo:mycodo...
with no effect
In a final desperation I upgraded to master from the command line
`sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_download.sh force-upgrade-master`
the upgrade was apparently successful, except the mycodo and mycodoflask still fail to start with above errors...

but again:
__pycache__ seems to have acquired root permissions, and only contains 3 files...
~/Mycodo/mycodo/controllers/__pycache__ $ ls -l
total 20
-rw-r--r-- 1 root root 4478 Apr 13 22:23 base_controller.cpython-37.pyc
-rw-r--r-- 1 root root 5161 Apr 13 22:23 controller_conditional.cpython-37.pyc
-rw-r--r-- 1 root root  134 Apr 13 22:23 __init__.cpython-37.pyc

After a bit of googling it seems this may not be a mycodo issue but a python issue...
so I tried:
```
sudo find ~/Mycodo/mycodo -name '*.pyc' -delete
sudo find /usr/ -name '*.pyc' -delete
```
still fails to start with same errors...
then after
`sudo find /Mycodo/env/ -name '*.pyc' -delete`

different error:
Apr 14 00:09:45 phspi06 python[2258]:   File "/var/mycodo-root/mycodo/databases/models/user.py", line 3, in <module>
Apr 14 00:09:45 phspi06 python[2258]:     from flask_login import UserMixin
Apr 14 00:09:45 phspi06 python[2258]:   File "/var/mycodo-root/env/lib/python3.7/site-packages/flask_login/__init__.py", line 16, in <module>
Apr 14 00:09:45 phspi06 python[2258]:     from .login_manager import LoginManager
Apr 14 00:09:45 phspi06 python[2258]:   File "/var/mycodo-root/env/lib/python3.7/site-packages/flask_login/login_manager.py", line 24, in <mod
Apr 14 00:09:45 phspi06 python[2258]:     from .utils import (_get_user, login_url as make_login_url, _create_identifier,
Apr 14 00:09:45 phspi06 python[2258]: ValueError: source code string cannot contain null bytes

More googling and attempt to remove any null bytes with:
`sed -i 's/\x0//g' login_manager.py`
and same result...

kizniche, 2020-04-13:
I've never seen that error before. I suspect corrupt data on your SD card is responsible. I'd recommend testing with a different SD card.

drgrumpy, 2020-04-13:
Yes that is where I am getting too... but of course that is non trivial....meanwhile it is a cold night and nearly 1 am local time!

Can you clarify what should the permission be on the __pycache__ directories and files within?

drgrumpy, 2020-04-14:
Some progress... reinstalled python3:
`sudo apt --reinstall install python3`
and now at least the daemon runs, but flask still gives errors, but different again:
r 14 02:21:43 phspi06 gunicorn[1858]:   File "/home/drgrumpy/Mycodo/env/lib/python3.7/site-packages/flask_wtf/csrf.py", line 11, in <module>
Apr 14 02:21:43 phspi06 gunicorn[1858]:     from wtforms import ValidationError
Apr 14 02:21:43 phspi06 gunicorn[1858]:   File "/home/drgrumpy/Mycodo/env/lib/python3.7/site-packages/wtforms/__init__.py", line 11, in <modul
Apr 14 02:21:43 phspi06 gunicorn[1858]:     from wtforms import validators, widgets
Apr 14 02:21:43 phspi06 gunicorn[1858]: ValueError: source code string cannot contain null bytes

kizniche, 2020-04-14:
I'm not sure about the permissions, but you shouldn't need to change them. This isn't likely to be the issue.

You can delete the Mycodo/env directory and rerun the setup.sh script to regenerate the virtual environment and install the required python packages.

```bash
sudo rm -rf ~/Mycodo/env
sudo /bin/bash ~/Mycodo/install/setup.sh
```

closed 2020-05-01 by kizniche

chamroeunbm, 2021-01-26:
Faced same issue. It's due to some corrupted pyc file. remove all .pyc files and it works for me.