kizniche/Mycodo #768: Fantastic project, but I'm stuck.... Need help, or is it a bug?

matthall69, 2020-05-03:
### How does one select multiple inputs for the Math:Redundancy feature?

### Versions:
 - Mycodo Version: 8.4.0
 - Raspberry Pi Version: 3B
 - Raspbian OS Version: Raspbian GNU/Linux 10 (buster)

### Reproducibility

New fresh install of Mycodo
Setup 2 DHT sensors or 2 DS18B20 sensors
Try to add 2 of these inputs into a Math:Redundancy controller 

### Expected behavior

By clicking Ctrl, and selecting more than 1 input, multiple inputs should be shown in the "Order of Use" list. Unfortunately I can't seem to get the list to populate. Tried saving after selecting multiple inputs, tried pressing enter, tried dragging. What am I missing? I can't seem to select multiple inputs, and then order them.

### Screenshots

[image]

### Additional context

Not sure if this is a bug, or if I'm just missing doing something!

kizniche, 2020-05-03:
Save, then they should appear in the order selection box.

matthall69, 2020-05-03:
Thanks for the prompt reply Kyle.

I've tried pressing save :-)

The screenshot above was taken after I'd pressed save, and gone back in to the controller definition.

kizniche, 2020-05-03:
Probably has remnants of old code before the measurement system was updated. I'll check it out.

kizniche, 2020-05-04:
I just pushed a fix that will be incorporated into the next release. If you want to test it before release, you can [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch).

matthall69, 2020-05-05:
All working now :-)

Thank you so much Kyle for the very quick turn around. I've tested this now, and it's working well. 

Thanks again for such an amazingly useful and polished project.

closed 2020-05-05 by matthall69