kizniche/Mycodo #769: Issues with integrating a VL53L0X "Time of light sensor"

NWH0, 2020-05-04:
@kizniche Thank you a lot for your great project. I'm using it for my hydroponics garden and I'm impatiently waiting for your project update on your hydroponics garden. 

### Describe the problem/bug
To measure tank levels and detecting leaks I wanted to use a VL53L0X - distance sensor in my water tank. I could manage to upload a custom input but don't get readings with Mycodo. I get proper readings via the command line with 

`   import board
    import busio
    import adafruit_vl53l0x
    i2c = busio.I2C(board.SCL, board.SDA)
    sensor = adafruit_vl53l0x.VL53L0X(i2c)
    print('Range: {}mm'.format(sensor.range))
`

but not in Mycodo. 

Could you help integrating this sensor? Any help is much apriciated.

Ressources for the VL53L0X sensor are: 

https://learn.adafruit.com/adafruit-vl53l0x-micro-lidar-distance-sensor-breakout/python-circuitpython

https://www.st.com/resource/en/datasheet/vl53l0x.pdf

### Versions:

 - Mycodo Version: 8.4.0
 - Raspberry Pi Version: 3B+
 - Raspbian OS Version: Buster

### Reproducibility

1. Uploading my custom input and start this input.

### Additional context

Custom Imput:

`# coding=utf-8
# Copyright (c) 2014 Adafruit Industries
# Author: Tony DiCola
# Based on the BMP280 driver with SHT31 changes provided by
# David J Taylor, Edinburgh (www.satsignal.eu)
import math
import time

from flask_babel import lazy_gettext

from mycodo.inputs.base_input import AbstractInput

# Measurements
measurements_dict = {
    0: {
        'measurement': 'Millimeter',
        'unit': 'mm'
    }
}

# Input information
INPUT_INFORMATION = {
    'input_name_unique': 'vl53l0x',
    'input_manufacturer': 'STMicroelectronics',
    'input_name': 'vl53l0x (Time of light)',
    'measurements_name': 'Millimeter',
    'measurements_dict': measurements_dict,

    'options_enabled': [
        'i2c_location',
        'measurements_select',
        'period',
        'pre_output',
        'log_level_debug'
    ],
    'options_disabled': ['interface'],

    'dependencies_module': [
        ('pip-pypi', 'adafruit_vl53l0x', 'adafruit-circuitpython-vl53l0x')
    ],

    'interfaces': ['I2C'],
    'i2c_location': [
        '0x29'
    ],
    'i2c_address_editable': False,
}

class InputModule(AbstractInput):
    """
    A sensor support class that measures the SHT31's humidity and temperature,
    them calculates the dew point and vapor pressure deficit
    """

    def __init__(self, input_dev, testing=False):
        super(InputModule, self).__init__()
        self.logger = logging.getLogger('mycodo.inputs.vl53l0x')
        self.name = INPUT_INFORMATION['input_name_unique']
        self.measurement_count = 0

        if not testing:
            from adafruit_vl53l0x import VL53L0X
            self.i2c_address = int(str(input_dev.i2c_location), 16)
            self.i2c_bus = input_dev.i2c_bus
            self.sensor = VL53L0X(
                address=self.i2c_address,
                busnum=self.i2c_bus)

    def get_measurement(self):
        self.return_dict = measurements_dict.copy()

        Millimeter = None
        success = False
        for _ in range(3):  # Three attempts
            try:
                Millimeter = self.sensor.range
            except OSError as e:
                self.logger.debug("OSError: {}".format(e))
                self.logger.debug("Attempting reset of sensor and another measurement")
                try:
                    self.sensor.reset()
                    try:
                        Millimeter = self.sensor.range
                    except Exception as e:
                        self.logger.debug("Measurement unsuccessful after reset")
                except Exception:
                    self.logger.debug("Reset command unsuccessful")
            if (None in [Millimeter]):
                self.logger.debug(
                    "At least one measurement is not a number: Millimeter: {}, ".format(Millimeter))
            else:
                success = True
                break
            time.sleep(0.1)

        if not success:
            self.logger.debug("Could not obtain measurements after 3 tries")
            return

        if self.is_enabled(0):
            self.value_set(0, Millimeter)

        return self.return_dict

    def status(self):
        status = self.sensor.read_status()
        is_data_crc_error = self.sensor.is_data_crc_error()
        is_command_error = self.sensor.is_command_error()
        is_reset_detected = self.sensor.is_reset_detected()

        self.logger.info('Status           = {:04X}'.format(status))
        self.logger.info('  Data CRC Error = {}'.format(is_data_crc_error))
        self.logger.info('  Command Error  = {}'.format(is_command_error))
        self.logger.info('  Reset Detected = {}'.format(is_reset_detected))
        time.sleep(0.1)
`

kizniche, 2020-05-04:
I just pushed a commit that adds support for this sensor, but since I don't have this sensor, it is currently untested. If you're willing to test, could you please [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch), test the new input, and report back whether it works or not? Also, while it's active, check the daemon log for any errors. Thanks.

NWH0, 2020-05-05:
Dear Kyle thank you a lot for your effortsa, for shure I will test it. Nice, that you integrated the accuracy as well. Sadly the sensor does not work yet. By the way why do think is the dependency you used better than the one I supposed?

If you need any further information pleas let me know.

The deamon log last 300 lines: 

```
2020-05-05 09:09:49,588 - INFO - mycodo.daemon - Mycodo daemon v8.4.0 starting
2020-05-05 09:09:50,070 - INFO - mycodo.controllers.controller_output - Output e7d4abb5 (GPIO) setup on pin 5 and turned OFF (OFF=HIGH)
2020-05-05 09:09:50,071 - INFO - mycodo.controllers.controller_output - Setting Output e7d4abb5 startup state to OFF
2020-05-05 09:09:50,071 - INFO - mycodo.controllers.controller_output - Activated in 127.2 ms
2020-05-05 09:09:50,923 - INFO - mycodo.controllers.controller_conditional_a8e66f84 - Activated in 113.4 ms
2020-05-05 09:09:50,930 - INFO - mycodo.daemon - All activated Conditional controllers started
2020-05-05 09:09:51,228 - ERROR - mycodo.controllers.controller_conditional_a8e66f84 - Exception executing check_conditionals() on attempt 1 of 3. Waiting 10 seconds and trying again.
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 124, in attempt_execute
    func()
  File "/var/mycodo-root/mycodo/controllers/controller_conditional.py", line 198, in check_conditionals
    self.conditional_run.conditional_code_run()
  File "/home/pi/Mycodo/mycodo/user_python_code/conditional_a8e66f84-abbf-4729-a31e-dbb81d8492ad.py", line 43, in conditional_code_run
    measurement = self.condition("29fccf96-7c08-4387-a1df-05d11e13421b")
  File "/home/pi/Mycodo/mycodo/user_python_code/conditional_a8e66f84-abbf-4729-a31e-dbb81d8492ad.py", line 26, in condition
    return control.get_condition_measurement(condition_id)
  File "/var/mycodo-root/mycodo/mycodo_client.py", line 211, in get_condition_measurement
    return self.proxy().get_condition_measurement(condition_id)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/client.py", line 475, in __call__
    return self.__send(self.__name, args, kwargs)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/Pyro5/client.py", line 242, in _pyroInvoke
    raise data  # if you see this in your traceback, you should probably inspect the remote traceback as well
AttributeError: 'NoneType' object has no attribute 'split'
2020-05-05 09:09:51,470 - INFO - mycodo.controllers.controller_trigger_0056b6e5 - Activated in 117.9 ms
2020-05-05 09:09:51,997 - INFO - mycodo.controllers.controller_trigger_4002c9ea - Activated in 118.1 ms
2020-05-05 09:09:52,003 - INFO - mycodo.daemon - All activated Trigger controllers started
2020-05-05 09:09:52,480 - DEBUG - mycodo.trigger_action_6dab38b5 - Message: 2020-05-05 09:09:51
[Trigger 4002c9ea-cc2c-46e8-96ba-fef64f9641a4 (Trigger: Timer (Daily Span))]
[Action 6dab38b5, Output (Duration)]: Turn output e7d4abb5-01e0-45a4-9423-93568bdb689c (1, Ein / Aus (GPIO)) on.
2020-05-05 09:09:52,481 - DEBUG - mycodo.trigger_action_6dab38b5 - Note Tags: []
2020-05-05 09:09:52,482 - DEBUG - mycodo.trigger_action_6dab38b5 - Email Recipients: []
2020-05-05 09:09:52,482 - DEBUG - mycodo.trigger_action_6dab38b5 - Attachment Files: None
2020-05-05 09:09:52,483 - DEBUG - mycodo.trigger_action_6dab38b5 - Attachment Type: None
2020-05-05 09:09:52,483 - DEBUG - mycodo.trigger_function_actions_4002c9ea - Message: 2020-05-05 09:09:51
[Trigger 4002c9ea-cc2c-46e8-96ba-fef64f9641a4 (Trigger: Timer (Daily Span))]
[Action 6dab38b5, Output (Duration)]: Turn output e7d4abb5-01e0-45a4-9423-93568bdb689c (1, Ein / Aus (GPIO)) on.
2020-05-05 09:09:52,921 - INFO - mycodo.controllers.controller_input_cde3a618 - Activated in 425.6 ms
2020-05-05 09:09:53,587 - INFO - mycodo.controllers.controller_input_d11844c0 - Activated in 298.9 ms
2020-05-05 09:09:54,100 - INFO - mycodo.controllers.controller_input_8240dc87 - Activated in 258.2 ms
2020-05-05 09:09:54,833 - INFO - mycodo.controllers.controller_input_531f1b90 - Activated in 429.0 ms
2020-05-05 09:09:55,396 - INFO - mycodo.controllers.controller_input_e39f1c00 - Activated in 283.6 ms
2020-05-05 09:09:55,399 - INFO - mycodo.daemon - All activated Input controllers started
2020-05-05 09:09:55,399 - INFO - mycodo.daemon - All activated Math controllers started
2020-05-05 09:09:55,400 - INFO - mycodo.daemon - All activated PID controllers started
2020-05-05 09:09:55,402 - INFO - mycodo.daemon - All activated LCD controllers started
2020-05-05 09:09:55,402 - INFO - mycodo.daemon - All activated Custom controllers started
2020-05-05 09:09:55,904 - INFO - mycodo.daemon - Mycodo daemon started in 6.314 seconds
2020-05-05 09:09:55,911 - INFO - mycodo.daemon - 47.98 MB RAM in use
2020-05-05 09:10:52,376 - DEBUG - mycodo.trigger_action_6dab38b5 - Message: 2020-05-05 09:10:52
[Trigger 4002c9ea-cc2c-46e8-96ba-fef64f9641a4 (Trigger: Timer (Daily Span))]
[Action 6dab38b5, Output (Duration)]: Turn output e7d4abb5-01e0-45a4-9423-93568bdb689c (1, Ein / Aus (GPIO)) on.
2020-05-05 09:10:52,376 - DEBUG - mycodo.trigger_action_6dab38b5 - Note Tags: []
2020-05-05 09:10:52,377 - DEBUG - mycodo.trigger_action_6dab38b5 - Email Recipients: []
2020-05-05 09:10:52,378 - DEBUG - mycodo.trigger_action_6dab38b5 - Attachment Files: None
2020-05-05 09:10:52,379 - DEBUG - mycodo.trigger_action_6dab38b5 - Attachment Type: None
2020-05-05 09:10:52,382 - DEBUG - mycodo.trigger_function_actions_4002c9ea - Message: 2020-05-05 09:10:52
[... cut ...]
2020-05-05 09:15:52,532 - DEBUG - mycodo.trigger_action_6dab38b5 - Email Recipients: []
2020-05-05 09:15:52,532 - DEBUG - mycodo.trigger_action_6dab38b5 - Attachment Files: None
2020-05-05 09:15:52,533 - DEBUG - mycodo.trigger_action_6dab38b5 - Attachment Type: None
2020-05-05 09:15:52,534 - DEBUG - mycodo.trigger_function_actions_4002c9ea - Message: 2020-05-05 09:15:52
[Trigger 4002c9ea-cc2c-46e8-96ba-fef64f9641a4 (Trigger: Timer (Daily Span))]
[Action 6dab38b5, Output (Duration)]: Turn output e7d4abb5-01e0-45a4-9423-93568bdb689c (1, Ein / Aus (GPIO)) on.
```

NWH0, 2020-05-05:
I tested the suggested dipendency (https://github.com/pimoroni/VL53L0X-python) via command line and it works.

kizniche, 2020-05-05:
I just pushed a commit that should fix the Python library issue. However, you will have to do the following to fully fix the issue because you currently have a broken library installed. Log in to your Pi and execute the following command to uninstall the current library:

```bash
~/Mycodo/env/bin/pip uninstall VL53L0X-rasp-python
```

Then, restart the Mycoo Frontend from the Config menu. Then, delete your current Input and add the input again and you will be prompted in Mycodo to install the dependency. It should work after that.

kizniche, 2020-05-05:
> By the way why do think is the dependency you used better than the one I supposed?

I chose to use a different one because I was unable to find a way to select the I2C bus using Circuitpython. This is a severe limitation of Circuitpython, in my opinion, as using multiple I2C buses on a Raspberry Pi is very common. If you happen to know how to select the bus, I'm willing to potentially use the Adafruit library, but as of now, this library allows a bus number to be passed to the class to select which bus to use.

NWH0, 2020-05-06:
Thanks for the update, everything works now!

kizniche, 2020-05-08:
Great. Thanks for testing!

closed 2020-06-01 by kizniche