kizniche/Mycodo #770: Duplication Buttons on the Web Interface

DarthPleurotus, 2020-05-04:
I am adding several inputs/outputs/functions to my Mycodo setup, some of which have very similar parameters to others. It would save some time and repetition if there was an option to duplicate an input/output/function entry so that it could then simply be modified slightly instead of creating a new entry from scratch.

kizniche, 2020-05-05:
Can you elaborate and provide an example how this would save a significant amount of time?

kizniche, 2020-05-08:
This has been raised in discussion a few times and I just don't see that much benefit, though I'm open to hear anyone's particular use that may persuade me to believe otherwise. I don't see more than a few seconds or minutes saved by doing this. I have to weigh how much time it would take to build and maintain the feature.

To accomplish duplicating an input, for instance, there's an Input database (db) entry that needs to be duplicated, and then there is a db entry for each measurement that needs to be duplicated, then the initial Input ID needs to be changed to a unique value, then each measurement entry needs to change the reference ID to match that Input ID to maintain association with it. This is not consistent across different controller types, so there would need to be custom handling for each controller that would need to be duplicated.

DarthPleurotus, 2020-05-10:
I requested this with my own build in mind, but also thinking about a friend who is currently trying to set up some small climate-controlled chambers for a grad school experiment and is looking into using Mycodo. For a build like his, where he is trying to do simple temp/humidity control for several chambers using the same RPi, the duplication feature could make things quicker and easier. I hear you though-- given the issues you described, the cost-benefit tradeoff may not favor making this change. I just wanted to put it out there. I am a big Mycodo fan and want to provide whatever feedback I can that might be useful.

kizniche, 2020-05-10:
There is a settings export/import feature that will allow you to essentially clone a setup on another Mycodo instance.

closed 2020-07-23 by kizniche