kizniche/Mycodo #771: Reset LCD gives Error 500 (when reset pin not configured)

samsixtysix, 2020-05-07:
### Describe the problem/bug
128x64 OLED (SD1306)
Clicking 'reset LCD' when 'Pin: Reset' is not configured gives below error:

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 2446, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1951, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_restx/api.py", line 600, in error_router
    return original_handler(e)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1820, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1949, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask/app.py", line 1935, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 1111, in page_lcd
    utils_lcd.lcd_reset_flashing(form_lcd_mod.lcd_id.data)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/utils/utils_lcd.py", line 229, in lcd_reset_flashing
    return_value, return_msg = control.lcd_flash(lcd_id, False)
TypeError: cannot unpack non-iterable NoneType object

### Versions:

 - Mycodo Version: 8.0.4
 - Raspberry Pi Version: 3B+

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Go to Setup/LCD
2. Add 128x64 OLED (SD1306)
3. Configure without reset pin
2. Click on 'Reset LCD'
4. See error

kizniche, 2020-05-08:
What does the daemon log show when this error occurs? The reset pin has nothing to do with the "Reset LCD" button (ironically). Reset LCD merely stops the LCD from flashing (after the flashing function action is activated). What you're likely experiencing is the result of a misconfigured LCD, which should have a traceback in the daemon log.

samsixtysix, 2020-05-08:
Found the problem following your pointer:
For the 128x32 OLED you have a check that warns you if you have not selected anything for one of the lines.  For the 128x64 the check is not there and I had left some lines with no selection specified.  Maybe the dropdown should default to 'Blank Line' rather than 'Select One'.

In case you need it Daemon log:

2020-05-08 09:44:30,849 - ERROR - mycodo.daemon - Could not flash LCD (False): 'NoneType' object has no attribute 'lcd_backlight'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 553, in lcd_flash
    return self.controller['LCD'][lcd_id].lcd_flash(state)
  File "/var/mycodo-root/mycodo/controllers/controller_lcd.py", line 571, in lcd_flash
    self.lcd_backlight(True)
  File "/var/mycodo-root/mycodo/controllers/controller_lcd.py", line 558, in lcd_backlight
    self.lcd_out.lcd_backlight(state)
AttributeError: 'NoneType' object has no attribute 'lcd_backlight'

This was presumably preventing the reset from working so my OLED was just blank/did not initialise.  Combine this with the fact that the first OLED I was trying to use was also faulty.

kizniche, 2020-05-08:
That's the issue! Thanks for helping hunt that down. I just pushed a fix for checking for blank lines on that 8-line display.

closed 2020-06-01 by kizniche