kizniche/Mycodo #772: Inverted PWM reported wrong

dylandn, 2020-05-11:
### Describe the problem/bug

I have two PWM fans that I am controlling, they are different brands/models.

The first fan is on pin 18 using hardware PWM, does not requires the "Invert Signal" button be checked, and it works great. 
The second fan, pin 23 "Any Pin" library, does requires the "Invert Signal" button be checked, and it's quirky.

If I set the duty cycle to 15%, I get two popups in the upper right corner. 

The first popup, colored blue, says "Command sent to turn output {blah} on with a duty cycle of 15%"

The second popup, colored green, says Output {blah}: SUCCESS: PWM 10 (Exhaust fan) set to a duty cycle of 85.00% at 10000 Hz" and 85% is reported everywhere. 

If I set a duty cycle of 25%, it reports 75%, etc.

### Versions:
Mycodo 8.4, Pi 3B+, Raspbian latest

[image]

kizniche, 2020-05-11:
What is the issue? You instruct a duty cycle of 15 % with a signal inversion, which turns the duty cycle to 80 %.

dylandn, 2020-05-11:
I guess I expected to see it report as 15%. I set the fan to 15%, the fan spins at 15%, but to have that reported as 85% is counterintuitive. If I uncheck the Invert Signal button, then when I set the fan to 15%, it's actually spinning at 85% and reported as 15%.

kizniche, 2020-05-11:
You appear to have a fan that spins slower as the duty cycle approaches 100 %. These types of devices are why the Invert Signal option exists. For instance, PID controllers will operate assuming 0 % duty cycle is off and 100 % is always on. Inverted devices would not operate properly unless the signal is inverted.

kizniche, 2020-05-11:
I suppose the duty cycle inversion could happen at the PID controller and leave the output alone.

dylandn, 2020-05-12:
You're right, this fan is quirky and runs at full speed at 0% and stops at 100% when the signal is not inverted. Inverting the signal works fine to fix that, but it's just reporting the wrong value.

kizniche, 2020-05-12:
What do you mean by "it's reporting the wrong value"? It's reporting the duty cycle being generated.

dylandn, 2020-05-12:
Yeah, I set the duty cycle to 15%, the fan's running at 15% but it's telling me 85%.

So if I look at a dashboard and it says 85%, I can't trust that value. I'll just have to remember to substract from 100% whenever I look at a dashboard.

kizniche, 2020-05-12:
Whatever it shows is what duty cycle is being produced. Whatever you input (when invert signal is enabled) will be inverted before the duty cycle is set.

kizniche, 2020-05-12:
85 % duty cycle is the correct value. Whatever is occurring on the hardware end past that is between you and the engineer thay designed it. You can trust when Mycodo says 85 % duty cycle, there's an 85 % duty cycle being generated on that GPIO pin.

kizniche, 2020-05-12:
Duty cycle is not synonymous with speed. It is merely a signal that can be interpreted in different ways by the hardware receiving the signal.

dylandn, 2020-05-12:
I think the problem is the fan is quirky and runs opposite of the duty cycle. 

I think what I need to do is duplicate the PWM output controller and add a line that's basically value=(100-value), just for this one quirky fan.

kizniche, 2020-05-12:
I don't think you're understanding what a duty cycle is. The fan isn't quirky, it's operating exactly how it was intended to operate.

dylandn, 2020-05-12:
I dont think you understand what my problem is. I'll just remember to substract the reported value from 100 to know what speed the fan's at.

closed 2020-05-12 by dylandn

kizniche, 2020-05-12:
Whatever the problem is, it's not with Mycodo.

kizniche, 2020-05-12:
I'm not opposed to changing anything. However, I don't see an issue there hasn't been described an issue other than your hardware not interpreting a PWM signal in a way you desire. If you have a suggestion how you think Mycodo can be improved, please share. If this is just a misunderstanding about how your device interprets the PWM signal, then I believe this issue is solved.

dylandn, 2021-12-04:
Thank you for adding the "Invert Stored Signal" feature, now my graphs make sense and report the actual speed of the fan.

dylandn, 2022-02-06:
Actually, Invert Stored Signal doesn't help. If a PWM output (which has "Invert Signal" checked) is set to, say, 65%, it still shows up as 35% on graphs.