kizniche/Mycodo #773: Self Watering Kit

smguth, 2020-05-14:
**Is your feature request related to a problem? Please describe.**
Hi guys, I puchased a kit on amazon thinking maybe I could get it going with an arduino or some other sort of device. **linked below** I'm looking for support/help integrating Capacitive Soil Moisture Sensor and Water Pump with mycodo.   I was hoping to connect the sensor to a esp32 but when I went into the list of supported devices I did not find it and searching the forums indicated people were doing it but I could figure out how. 
 
https://www.amazon.com/gp/product/B07TLRYGT1/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

**Describe the solution you'd like**
Basically I want help getting started.  It would be nice using a esp32 or esp8266 like they show in the photos for the product on amazon.   Ideally i'd have a few of these watering my house plants at different locations talking to mycodo. 

**Describe alternatives you've considered**
I was thinking maybe this could work with the sonoff TH10/16 since it is listed as a supported device but the inputs don't seem like they would work with this sensor.

kizniche, 2020-05-14:
The soil moisture sensor and pump look easy enough to interface with an arduino or ESP32. The output voltage of the sensor is 0 to 3 VDC and can be measured by an analog pin. From a cursory search, it looks like MQTT may be the easiest way to receive measurements from an ESP32 to a Pi and send output command from the Pi to the ESP32 to run the pump (see https://randomnerdtutorials.com/esp32-mqtt-publish-subscribe-arduino-ide/).

kizniche, 2020-05-14:
For interfacing with the Pi, you would install and start a Mosquito MQTT Server, which you can then use with an "MQTT Protocol (paho)" Input that can subscribe to the soil moisture measurement topic and store any measurements sent by the ESP32. The Pi can publish output commands with the "Linux Command" or the "Python Code" Output and the ESP32 can subscribe to those messages and turn on the pump for the duration value it receives.

smguth, 2020-05-14:
Thanks Kyle for pointing me in the right direction.  I'll have a bit of
reading/learning ahead of me but i'm looking forward to learning something
new.  Hoping I can get this going in short order!
Thanks for your quick response!
smguth

closed 2021-02-02 by kizniche