kizniche/Mycodo #774: Unable to use GPIO Outputs after upgrade to 8.5.1

adkodak, 2020-05-30:
When trying to turn on GPIO outputs after upgrading to 8.5.1 I'm receiving the 'ERROR: Could not turn output on: ' GPIO is not in use for PWM'

pigpio.error: 'GPIO is not in use for PWM'
2020-05-30 13:51:45,268 - ERROR - mycodo.inputs.rpi_signal_pwm_ce5e0598 - Could not connect to pigpiod.Ensure it is running and try again.
2020-05-30 13:51:46,591 - ERROR - mycodo.controllers.controller_pid_6757a1e4 - Exception executing check_pid() on attempt 3 of 3.
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 124, in attempt_execute
    func()
  File "/var/mycodo-root/mycodo/controllers/controller_pid.py", line 366, in check_pid
    self.manipulate_output()
  File "/var/mycodo-root/mycodo/controllers/controller_pid.py", line 645, in manipulate_output
    if ('output_types' in self.dict_outputs[self.lower_output_type] and
KeyError: None
[image]

kizniche, 2020-05-30:
Please screenshot the Output settings

adkodak, 2020-05-30:
[image]

kizniche, 2020-05-30:
What version did you upgrade from? Can you provide your upgrade log excerpt of the upgrade?

kizniche, 2020-05-30:
FYI, you can restore a backed up version that was created during an upgrade. I haven't yet been able to replicate your issue. All of my systems upgraded without issue.

adkodak, 2020-05-30:
[2020-05-30 13:01:43] 
[2020-05-30 13:01:43] Installed version: 8.4.0
[2020-05-30 13:01:43] Latest version: 8.5.1
[2020-05-30 13:01:43] 
[2020-05-30 13:01:43] #### Beginning Upgrade: Stage 1 of 3 ####
[2020-05-30 13:01:43] 
[2020-05-30 13:01:57] Downloading latest Mycodo version 8.5.1 to /home/pi/mycodo-8.5.1.tar.gz...Done.
[2020-05-30 13:01:57] Creating /tmp/Mycodo-8.5.1...Done.
[2020-05-30 13:01:58] Extracting /home/pi/mycodo-8.5.1.tar.gz to /tmp/Mycodo-8.5.1...Done.
[2020-05-30 13:01:58] Removing /home/pi/mycodo-8.5.1.tar.gz...Done.
[2020-05-30 13:01:58] 
[2020-05-30 13:01:58] #### Completed Upgrade Stage 1 of 3 in 15 seconds ####
[2020-05-30 13:01:58] 
[2020-05-30 13:01:58] #### Beginning Upgrade Stage 2 of 3 ####
[2020-05-30 13:01:58] 
[2020-05-30 13:01:58] RELEASE_WIPE = false
[2020-05-30 13:01:58] Found currently-installed version of Mycodo. Checking version...
[2020-05-30 13:01:59] Mycodo version found installed: 8.4.0
[2020-05-30 13:01:59] 
[2020-05-30 13:01:59] #### Beginning pre-upgrade checks ####
[2020-05-30 13:01:59] 
[2020-05-30 13:02:00] Checking Python version...
[2020-05-30 13:02:00] Python >= 3.6 found. Continuing with the upgrade.
[2020-05-30 13:02:00] All pre-upgrade checks passed. Proceeding with upgrade.
[2020-05-30 13:02:00] 
[2020-05-30 13:02:01] Upgrading Mycodo to version 8.5.1
[2020-05-30 13:02:01] 
[2020-05-30 13:02:08] Stopping the Mycodo daemon...Done.
[2020-05-30 13:02:08] Moving env directory...Done.
[2020-05-30 13:02:08] Copying databases...Done.
[2020-05-30 13:02:08] Copying flask_secret_key...Done.
[2020-05-30 13:02:08] Copying SSL certificates...Done.
[2020-05-30 13:02:08] Copying mycodo/controllers/custom_controllers...Done.
[2020-05-30 13:02:08] Copying mycodo/inputs/custom_inputs...Done.
[2020-05-30 13:02:08] Copying mycodo/user_python_code...Done.
[2020-05-30 13:02:08] Moving cameras directory...Done.
[2020-05-30 13:02:08] Moving current Mycodo intstall from /home/pi/Mycodo to /var/Mycodo-backups/Mycodo-backup-2020-05-30_13-01-58-8.4.0...Done.
[2020-05-30 13:02:08] Moving downloaded Mycodo version from /tmp/Mycodo-8.5.1 to /home/pi/Mycodo...Done.
[2020-05-30 13:02:38] 
[2020-05-30 13:02:38] #### Completed Upgrade Stage 2 of 3 in 40 seconds ####
[2020-05-30 13:02:38] 
[2020-05-30 13:02:38] #### Beginning Upgrade Stage 3 of 3 ####
[2020-05-30 13:02:38] 
[2020-05-30 13:02:38] Running post-upgrade script...
[2020-05-30 13:02:38] 
[2020-05-30 13:02:38] #### Running initialization
[2020-05-30 13:02:38] 
[2020-05-30 13:02:38] #### Creating mycodo user
[2020-05-30 13:02:38] useradd: user 'mycodo' already exists
[2020-05-30 13:02:38] The user `mycodo' is already a member of `adm'.
[2020-05-30 13:02:38] The user `mycodo' is already a member of `dialout'.
[2020-05-30 13:02:39] The user `mycodo' is already a member of `gpio'.
[2020-05-30 13:02:39] The user `mycodo' is already a member of `i2c'.
[2020-05-30 13:02:39] The user `mycodo' is already a member of `kmem'.
[2020-05-30 13:02:39] The user `mycodo' is already a member of `video'.
[2020-05-30 13:02:39] The user `pi' is already a member of `mycodo'.
[2020-05-30 13:02:39] The user `mycodo' is already a member of `pi'.
[2020-05-30 13:02:39] 
[2020-05-30 13:02:39] #### Compiling mycodo_wrapper
[2020-05-30 13:02:40] /home/pi/Mycodo/mycodo/scripts/mycodo_wrapper.c: In function âupgrade_commandsâ:
[2020-05-30 13:02:40] /home/pi/Mycodo/mycodo/scripts/mycodo_wrapper.c:17:5: warning: âstrncatâ specified bound 255 equals destination size [-Wstringop-overflow=]
[2020-05-30 13:02:40]      strncat(full_cmd, "/upgrade_commands.sh ", sizeof(full_cmd));
[2020-05-30 13:02:40]      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[2020-05-30 13:02:40] /home/pi/Mycodo/mycodo/scripts/mycodo_wrapper.c: In function âmainâ:
[2020-05-30 13:02:40] /home/pi/Mycodo/mycodo/scripts/mycodo_wrapper.c:35:4: warning: âstrncatâ specified bound 255 equals destination size [-Wstringop-overflow=]
[2020-05-30 13:02:40]     strncat(restoreScript, "/upgrade_commands.sh backup-create", sizeof(restoreScript));
[2020-05-30 13:02:40]     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[2020-05-30 13:02:40] 
[2020-05-30 13:02:40] #### Creating symlinks to Mycodo executables
[2020-05-30 13:02:40] 
[... 574 lines cut ...]
[2020-05-30 13:07:10] Stage 3 execution time summary:
[2020-05-30 13:07:10] initialize:                   4 s
[2020-05-30 13:07:10] update-swap-size:             0 s
[2020-05-30 13:07:10] setup-virtualenv:             0 s
[2020-05-30 13:07:10] update-apt:                   22 s
[2020-05-30 13:07:10] update-packages:              21 s
[2020-05-30 13:07:10] web-server-update:            2 s
[2020-05-30 13:07:10] update-logrotate:             0 s
[2020-05-30 13:07:10] update-pip3:                  12 s
[2020-05-30 13:07:10] update-pip3-packages:         60 s
[2020-05-30 13:07:10] update-dependencies:          66 s
[2020-05-30 13:07:10] update-influxdb:              56 s
[2020-05-30 13:07:10] update-alembic:               2 s
[2020-05-30 13:07:10] update-alembic-post:          3 s
[2020-05-30 13:07:10] update-mycodo-startup-script: 1 s
[2020-05-30 13:07:10] compile-translations:         4 s
[2020-05-30 13:07:10] update-cron:                  0 s
[2020-05-30 13:07:10] update-permissions:           1 s
[2020-05-30 13:07:10] restart-daemon:               9 s
[2020-05-30 13:07:10] web-server_reload:            5 s
[2020-05-30 13:07:10] web-server-connect:           4 s
[2020-05-30 13:07:10] 
[2020-05-30 13:07:10] #### Completed Upgrade Stage 3 of 3 in 272 seconds ####
[2020-05-30 13:07:10] 
[2020-05-30 13:07:10] Upgrade completed successfully without errors.

kizniche, 2020-05-30:
Your upgrade log looks good. I'll keep investigating.

kizniche, 2020-06-01:
I still have no idea how you're getting that kind of error. By now there have been many systems that have upgraded and no one else has made an issue like this or found their way here. I'm not really sure what to tell you.

closed 2020-06-06 by kizniche