kizniche/Mycodo #775: error 502

koenaal, 2020-05-30:
i have got a problem to conect to my mycodo i get a error 502.
it happend after i have update, can you help me fix it? i am new with mycodo .

kind regarts.
koen

kizniche, 2020-05-30:
Log into the Pi and run these commands:

```
sudo service mycodoflask stop
cd ~/Mycodo/mycodo
sudo /var/mycodo-root/env/bin/gunicorn \
--workers 1 \
--worker-class gthread \
--threads 2 \
--timeout 300 \
--pid /var/run/mycodoflask.pid \
--bind unix:/var/run/mycodoflask.sock start_flask_ui:app
```

Then refresh the web UI, and copy the error that appears in the terminal.

koenaal, 2020-05-30:
i get this after the command:

pi@raspberrypi:~ $ > --workers 1 \
> > --worker-class gthread \
> > --threads 2 \
> > --timeout 300 \
> > --pid /var/run/mycodoflask.pid \
> > --bind unix:/var/run/mycodoflask.sock start_flask_ui:app 
bash: 1: opdracht niet gevonden
pi@raspberrypi:~ $ [2020-05-30 22:37:15 +0200] [1567] [INFO] Starting gunicorn 20.0.4
bash: [2020-05-30: opdracht niet gevonden
pi@raspberrypi:~ $ [2020-05-30 22:37:15 +0200] [1567] [INFO] Listening at: unix:/var/run/mycodoflask.sock (1567)
bash: syntaxfout nabij onverwacht symbool '('
pi@raspberrypi:~ $ [2020-05-30 22:37:15 +0200] [1567] [INFO] Using worker: gthread
bash: [2020-05-30: opdracht niet gevonden
pi@raspberrypi:~ $ [2020-05-30 22:37:15 +0200] [1571] [INFO] Booting worker with pid: 1571
bash: [2020-05-30: opdracht niet gevonden
pi@raspberrypi:~ $ [2020-05-30 22:37:16 +0200] [1571] [ERROR] Exception in worker process
bash: [2020-05-30: opdracht niet gevonden
pi@raspberrypi:~ $ Traceback (most recent call last):
bash: syntaxfout nabij onverwacht symbool 'most'
pi@raspberrypi:~ $   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/arbiter.py", line 583, in spawn_worker
bash: File: opdracht niet gevonden
pi@raspberrypi:~ $     worker.init_process()
>   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/workers/gthread.py", line 92, in init_process
bash: syntaxfout nabij onverwacht symbool 'File'
pi@raspberrypi:~ $     super().init_process()
bash: syntaxfout nabij onverwacht symbool '.init_process'
pi@raspberrypi:~ $   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/workers/base.py", line 119, in init_process
bash: File: opdracht niet gevonden
pi@raspberrypi:~ $     self.load_wsgi()
>   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/workers/base.py", line 144, in load_wsgi
bash: syntaxfout nabij onverwacht symbool 'File'
pi@raspberrypi:~ $     self.wsgi = self.app.wsgi()
bash: syntaxfout nabij onverwacht symbool '('
pi@raspberrypi:~ $   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/app/base.py", line 67, in wsgi
bash: File: opdracht niet gevonden
pi@raspberrypi:~ $     self.callable = self.load()
bash: syntaxfout nabij onverwacht symbool '('
pi@raspberrypi:~ $   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/app/wsgiapp.py", line 49, in load
bash: File: opdracht niet gevonden
pi@raspberrypi:~ $     return self.load_wsgiapp()
bash: syntaxfout nabij onverwacht symbool '('
pi@raspberrypi:~ $   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/app/wsgiapp.py", line 39, in load_wsgiapp
bash: File: opdracht niet gevonden
pi@raspberrypi:~ $     return util.import_app(self.app_uri)
bash: syntaxfout nabij onverwacht symbool '('
pi@raspberrypi:~ $   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/util.py", line 358, in import_app
bash: File: opdracht niet gevonden
pi@raspberrypi:~ $     mod = importlib.import_module(module)
bash: syntaxfout nabij onverwacht symbool '('
pi@raspberrypi:~ $   File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
bash: File: opdracht niet gevonden
pi@raspberrypi:~ $     return _bootstrap._gcd_import(name[level:], package, level)
bash: syntaxfout nabij onverwacht symbool '('
pi@raspberrypi:~ $   File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
bash: File: opdracht niet gevonden
pi@raspberrypi:~ $   File "<frozen importlib._bootstrap>", line 983, in _find_and_load
bash: File: opdracht niet gevonden
pi@raspberrypi:~ $   File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
bash: File: opdracht niet gevonden
pi@raspberrypi:~ $   File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
bash: File: opdracht niet gevonden
pi@raspberrypi:~ $   File "<frozen importlib._bootstrap_external>", line 728, in exec_module
bash: File: opdracht niet gevonden
pi@raspberrypi:~ $   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
bash: File: opdracht niet gevonden
pi@raspberrypi:~ $   File "/home/pi/Mycodo/mycodo/start_flask_ui.py", line 11, in <module>
bash: syntaxfout nabij onverwacht symbool 'newline'
pi@raspberrypi:~ $     from mycodo.mycodo_flask.app import create_app
from: can't read /var/mail/mycodo.mycodo_flask.app
pi@raspberrypi:~ $   File "/home/pi/Mycodo/mycodo/mycodo_flask/app.py", line 17, in <module>
bash: syntaxfout nabij onverwacht symbool 'newline'
pi@raspberrypi:~ $     from flask_limiter import Limiter
from: can't read /var/mail/flask_limiter
pi@raspberrypi:~ $   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_limiter/__init__.py", line 8, in <module>
bash: syntaxfout nabij onverwacht symbool 'newline'
pi@raspberrypi:~ $     from .errors import RateLimitExceeded
from: can't read /var/mail/.errors
pi@raspberrypi:~ $   File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_limiter/errors.py", line 6, in <module>
bash: syntaxfout nabij onverwacht symbool 'newline'
pi@raspberrypi:~ $     from pkg_resources import get_distribution
from: can't read /var/mail/pkg_resources
pi@raspberrypi:~ $ ImportError: cannot import name 'get_distribution' from 'pkg_resources' (/home/pi/Mycodo/env/lib/python3.7/site-packages/pkg_resources/__init__.py)
bash: syntaxfout nabij onverwacht symbool '('
pi@raspberrypi:~ $ [2020-05-30 22:37:16 +0200] [1571] [INFO] Worker exiting (pid: 1571)
bash: syntaxfout nabij onverwacht symbool '('
pi@raspberrypi:~ $ [2020-05-30 22:37:17 +0200] [1567] [INFO] Shutting down: Master
bash: [2020-05-30: opdracht niet gevonden
pi@raspberrypi:~ $ [2020-05-30 22:37:17 +0200] [1567] [INFO] Reason: Worker failed to boot.
bash: [2020-05-30: opdracht niet gevonden

kizniche, 2020-05-30:
That doesn't look like it was executed properly

kizniche, 2020-05-30:
You seem to have deleted the "\" from the command. Copy and paste it. It must be the exact command I wrote.

koenaal, 2020-05-30:
pi@raspberrypi:~ $ sudo service mycodoflask stop
pi@raspberrypi:~ $ cd ~/Mycodo/mycodo
pi@raspberrypi:~/Mycodo/mycodo $ sudo /var/mycodo-root/env/bin/gunicorn \
> --workers 1 \
> --worker-class gthread \
> --threads 2 \
> --timeout 300 \
> --pid /var/run/mycodoflask.pid \
> --bind unix:/var/run/mycodoflask.sock start_flask_ui:app
[2020-05-30 23:02:44 +0200] [2266] [INFO] Starting gunicorn 20.0.4
[2020-05-30 23:02:44 +0200] [2266] [INFO] Listening at: unix:/var/run/mycodoflask.sock (2266)
[2020-05-30 23:02:44 +0200] [2266] [INFO] Using worker: gthread
[2020-05-30 23:02:44 +0200] [2269] [INFO] Booting worker with pid: 2269
[2020-05-30 23:02:45 +0200] [2269] [ERROR] Exception in worker process
Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/arbiter.py", line 583, in spawn_worker
    worker.init_process()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/workers/gthread.py", line 92, in init_process
    super().init_process()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/workers/base.py", line 119, in init_process
    self.load_wsgi()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/workers/base.py", line 144, in load_wsgi
    self.wsgi = self.app.wsgi()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/app/base.py", line 67, in wsgi
    self.callable = self.load()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/app/wsgiapp.py", line 49, in load
    return self.load_wsgiapp()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/app/wsgiapp.py", line 39, in load_wsgiapp
    return util.import_app(self.app_uri)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/util.py", line 358, in import_app
    mod = importlib.import_module(module)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/pi/Mycodo/mycodo/start_flask_ui.py", line 11, in <module>
    from mycodo.mycodo_flask.app import create_app
  File "/home/pi/Mycodo/mycodo/mycodo_flask/app.py", line 17, in <module>
    from flask_limiter import Limiter
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_limiter/__init__.py", line 8, in <module>
    from .errors import RateLimitExceeded
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/flask_limiter/errors.py", line 6, in <module>
    from pkg_resources import get_distribution
ImportError: cannot import name 'get_distribution' from 'pkg_resources' (/home/pi/Mycodo/env/lib/python3.7/site-packages/pkg_resources/__init__.py)
[2020-05-30 23:02:45 +0200] [2269] [INFO] Worker exiting (pid: 2269)
[2020-05-30 23:02:45 +0200] [2266] [INFO] Shutting down: Master
[2020-05-30 23:02:45 +0200] [2266] [INFO] Reason: Worker failed to boot.

koenaal, 2020-05-30:
sorry this is what i got after the command

kizniche, 2020-05-30:
looks like your python environment may be messed up somehow. You can rebuild it with the following commands:

```
mv ~/Mycodo/env ~/Mycodo/env-old
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh setup-virtualenv
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh update-pip3-packages
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh update-dependencies
sudo service mycodo restart
sudo service mycodoflask restart
```

Check if it's working. If not, use the commands in [this comment](https://github.com/kizniche/Mycodo/issues/775#issuecomment-636381483) again.

koenaal, 2020-05-31:
it didn't work if i used the command in your first comment i get this:

pi@raspberrypi:~ $ sudo service mycodoflask stop
pi@raspberrypi:~ $ cd ~/Mycodo/mycodo
pi@raspberrypi:~/Mycodo/mycodo $ sudo /var/mycodo-root/env/bin/gunicorn \
> --workers 1 \
> --worker-class gthread \
> --threads 2 \
> --timeout 300 \
> --pid /var/run/mycodoflask.pid \
> --bind unix:/var/run/mycodoflask.sock start_flask_ui:app
[2020-05-31 09:48:11 +0200] [1924] [INFO] Starting gunicorn 20.0.4
[2020-05-31 09:48:11 +0200] [1924] [INFO] Listening at: unix:/var/run/mycodoflask.sock (1924)
[2020-05-31 09:48:11 +0200] [1924] [INFO] Using worker: gthread
[2020-05-31 09:48:11 +0200] [1927] [INFO] Booting worker with pid: 1927
[2020-05-31 09:48:26 +0200] [1927] [ERROR] Exception in worker process
Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1284, in _execute_context
    cursor, statement, parameters, context
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/default.py", line 590, in do_execute
    cursor.execute(statement, parameters)
sqlite3.OperationalError: no such column: roles.reset_password

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/arbiter.py", line 583, in spawn_worker
    worker.init_process()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/workers/gthread.py", line 92, in init_process
    super().init_process()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/workers/base.py", line 119, in init_process
    self.load_wsgi()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/workers/base.py", line 144, in load_wsgi
    self.wsgi = self.app.wsgi()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/app/base.py", line 67, in wsgi
    self.callable = self.load()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/app/wsgiapp.py", line 49, in load
    return self.load_wsgiapp()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/app/wsgiapp.py", line 39, in load_wsgiapp
    return util.import_app(self.app_uri)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/util.py", line 358, in import_app
    mod = importlib.import_module(module)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/pi/Mycodo/mycodo/start_flask_ui.py", line 13, in <module>
    app = create_app()  # required by the wsgi config and main()
  File "/home/pi/Mycodo/mycodo/mycodo_flask/app.py", line 58, in create_app
    register_extensions(app)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/app.py", line 80, in register_extensions
    populate_db()
  File "/home/pi/Mycodo/mycodo/databases/models/__init__.py", line 136, in populate_db
    known_roles = {r.name: r for r in Role.query.all()}
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/orm/query.py", line 3319, in all
    return list(self)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/orm/query.py", line 3481, in __iter__
    return self._execute_and_instances(context)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/orm/query.py", line 3506, in _execute_and_instances
    result = conn.execute(querycontext.statement, self._params)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1020, in execute
    return meth(self, multiparams, params)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/sql/elements.py", line 298, in _execute_on_connection
    return connection._execute_clauseelement(self, multiparams, params)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1139, in _execute_clauseelement
    distilled_params,
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1324, in _execute_context
    e, statement, parameters, cursor, context
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1518, in _handle_dbapi_exception
    sqlalchemy_exception, with_traceback=exc_info[2], from_=e
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/util/compat.py", line 178, in raise_
    raise exception
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1284, in _execute_context
    cursor, statement, parameters, context
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/default.py", line 590, in do_execute
    cursor.execute(statement, parameters)
sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) no such column: roles.reset_password
[SQL: SELECT roles.id AS roles_id, roles.unique_id AS roles_unique_id, roles.name AS roles_name, roles.edit_settings AS roles_edit_settings, roles.edit_controllers AS roles_edit_controllers, roles.edit_users AS roles_edit_users, roles.view_settings AS roles_view_settings, roles.view_camera AS roles_view_camera, roles.view_stats AS roles_view_stats, roles.view_logs AS roles_view_logs, roles.reset_password AS roles_reset_password 
FROM roles]
(Background on this error at: http://sqlalche.me/e/e3q8)
[2020-05-31 09:48:26 +0200] [1927] [INFO] Worker exiting (pid: 1927)
[2020-05-31 09:48:27 +0200] [1924] [INFO] Shutting down: Master
[2020-05-31 09:48:27 +0200] [1924] [INFO] Reason: Worker failed to boot.

kizniche, 2020-05-31:
It did fix your python virtualenv. This is a different error, and indicates your database is not the correct version. Sometime during an upgrade in the past your database developed an issue. Have you ever upgraded to the master branch in the past?

Your only real recourse is to delete your database and start anew. To fix this kind of issue, you would likely need to manually edit the database.

kizniche, 2020-06-01:
From #775, it seems this may fix your issue:

```
cd ~/Mycodo/databases
../env/bin/alembic upgrade head
```

closed 2020-06-06 by kizniche