kizniche/Mycodo #776: Error 502 after Upgrade

BenjiBuilt, 2020-06-01:
Hey,

I just updated mycodo and after that happened I get a Error 502. 
I just put HOURS into setting up mycodo and I dont really want to start again.

I have performed this test as you have mentioned in many of these other threads.

sudo service mycodoflask stop
cd ~/Mycodo/mycodo
sudo /var/mycodo-root/env/bin/gunicorn \
--workers 1 \
--worker-class gthread \
--threads 2 \
--timeout 300 \
--pid /var/run/mycodoflask.pid \
--bind unix:/var/run/mycodoflask.sock start_flask_ui:app

Which Resulted in this in the pi terminal:

pi@raspberrypi:~ $ sudo service mycodoflask stop
pi@raspberrypi:~ $ cd ~/Mycodo/mycodo
pi@raspberrypi:~/Mycodo/mycodo $ sudo /var/mycodo-root/env/bin/gunicorn \
> --workers 1 \
> --worker-class gthread \
> --threads 2 \
> --timeout 300 \
> --pid /var/run/mycodoflask.pid \
> --bind unix:/var/run/mycodoflask.sock start_flask_ui:app
[2020-06-01 14:05:42 +1000] [3846] [INFO] Starting gunicorn 20.0.4
[2020-06-01 14:05:42 +1000] [3846] [INFO] Listening at: unix:/var/run/mycodoflask.sock (3846)
[2020-06-01 14:05:42 +1000] [3846] [INFO] Using worker: gthread
[2020-06-01 14:05:42 +1000] [3849] [INFO] Booting worker with pid: 3849
[2020-06-01 14:05:45 +1000] [3849] [ERROR] Exception in worker process
Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1246, in _execute_context
    cursor, statement, parameters, context
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/default.py", line 588, in do_execute
    cursor.execute(statement, parameters)
sqlite3.OperationalError: no such column: roles.reset_password

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/arbiter.py", line 583, in spawn_worker
    worker.init_process()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/workers/gthread.py", line 92, in init_process
    super().init_process()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/workers/base.py", line 119, in init_process
    self.load_wsgi()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/workers/base.py", line 144, in load_wsgi
    self.wsgi = self.app.wsgi()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/app/base.py", line 67, in wsgi
    self.callable = self.load()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/app/wsgiapp.py", line 49, in load
    return self.load_wsgiapp()
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/app/wsgiapp.py", line 39, in load_wsgiapp
    return util.import_app(self.app_uri)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/gunicorn/util.py", line 358, in import_app
    mod = importlib.import_module(module)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/pi/Mycodo/mycodo/start_flask_ui.py", line 13, in <module>
    app = create_app()  # required by the wsgi config and main()
  File "/home/pi/Mycodo/mycodo/mycodo_flask/app.py", line 58, in create_app
    register_extensions(app)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/app.py", line 80, in register_extensions
    populate_db()
  File "/home/pi/Mycodo/mycodo/databases/models/__init__.py", line 136, in populate_db
    known_roles = {r.name: r for r in Role.query.all()}
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/orm/query.py", line 3233, in all
    return list(self)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/orm/query.py", line 3389, in __iter__
    return self._execute_and_instances(context)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/orm/query.py", line 3414, in _execute_and_instances
    result = conn.execute(querycontext.statement, self._params)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 982, in execute
    return meth(self, multiparams, params)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/sql/elements.py", line 293, in _execute_on_connection
    return connection._execute_clauseelement(self, multiparams, params)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1101, in _execute_clauseelement
    distilled_params,
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1250, in _execute_context
    e, statement, parameters, cursor, context
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1476, in _handle_dbapi_exception
    util.raise_from_cause(sqlalchemy_exception, exc_info)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/util/compat.py", line 398, in raise_from_cause
    reraise(type(exception), exception, tb=exc_tb, cause=cause)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/util/compat.py", line 152, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1246, in _execute_context
    cursor, statement, parameters, context
  File "/home/pi/Mycodo/env/lib/python3.7/site-packages/sqlalchemy/engine/default.py", line 588, in do_execute
    cursor.execute(statement, parameters)
sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) no such column: roles.reset_password
[SQL: SELECT roles.id AS roles_id, roles.unique_id AS roles_unique_id, roles.name AS roles_name, roles.edit_settings AS roles_edit_settings, roles.edit_controllers AS roles_edit_controllers, roles.edit_users AS roles_edit_users, roles.view_settings AS roles_view_settings, roles.view_camera AS roles_view_camera, roles.view_stats AS roles_view_stats, roles.view_logs AS roles_view_logs, roles.reset_password AS roles_reset_password 
FROM roles]
(Background on this error at: http://sqlalche.me/e/e3q8)
[2020-06-01 14:05:45 +1000] [3849] [INFO] Worker exiting (pid: 3849)
[2020-06-01 14:05:45 +1000] [3846] [INFO] Shutting down: Master
[2020-06-01 14:05:45 +1000] [3846] [INFO] Reason: Worker failed to boot.
pi@raspberrypi:~/Mycodo/mycodo $ 

Any help would be appreciated please!

kizniche, 2020-06-01:
What's the output from the following commands?

```
cd ~/Mycodo/databases
../env/bin/alembic upgrade head
```

BenjiBuilt, 2020-06-01:
pi@raspberrypi:~ $ cd ~/Mycodo/databases
pi@raspberrypi:~/Mycodo/databases $ ../env/bin/alembic upgrade head
bash: ../env/bin/alembic: No such file or directory
pi@raspberrypi:~/Mycodo/databases $

kizniche, 2020-06-01:
And this?

```bash
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh update-pip3-packages
```

BenjiBuilt, 2020-06-01:
Its taking a long time but this is where its at right now.

pi@raspberrypi:~ $ sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh update-pip3-packages

#### Installing pip requirements
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='www.piwheels.org', port=443): Read timed out. (read timeout=15)")': /simple/pip/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='www.piwheels.org', port=443): Read timed out. (read timeout=15)")': /simple/pip/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='www.piwheels.org', port=443): Read timed out. (read timeout=15)")': /simple/pip/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='www.piwheels.org', port=443): Read timed out. (read timeout=15)")': /simple/pip/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='www.piwheels.org', port=443): Read timed out. (read timeout=15)")': /simple/pip/
Requirement already up-to-date: pip in ./env/lib/python3.7/site-packages (20.1.1)
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='www.piwheels.org', port=443): Read timed out. (read timeout=15)")': /simple/setuptools/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='www.piwheels.org', port=443): Read timed out. (read timeout=15)")': /simple/setuptools/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='www.piwheels.org', port=443): Read timed out. (read timeout=15)")': /simple/setuptools/
^CERROR: Operation cancelled by user
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='www.piwheels.org', port=443): Read timed out. (read timeout=15)")': /simple/alembic/

kizniche, 2020-06-01:
Your python environment appears to be broken and your internet connection doesn't seem to be working to update the pip packages.

BenjiBuilt, 2020-06-01:
Hmmmm well im posting all of this information to you from the raspberry pi itself.... So i dont know how I dont have internet?

kizniche, 2020-06-01:
for whatever reason, it's unable to connect for pip to operate properly. That's what the above errors indicate.

BenjiBuilt, 2020-06-01:
Okay so is there anything else you want me to try? I just went on youtube on the Pi and it loads fine.

kizniche, 2020-06-01:
You can try completely rebuilding your virtualenv:

```
sudo service mycodo stop
sudo service mycodoflask stop
mv ~/Mycodo/env ~/Mycodo/env-old
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh setup-virtualenv
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh update-pip3-packages
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh update-dependencies
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh update-permissions
```

kizniche, 2020-06-01:
You can always perform a restore of a backup (created before the upgrade started) by running the command:

```bash
sudo /bin/bash ~/Mycodo/mycodo/scripts/mycodo_backup_restore.sh /var/Mycodo-backups/Mycodo-backup-2020-03-23_11-47-31-8.3.0/
```

Replacing the path with the actual path to your backup. Note: Do not perform a restore while you are attempting to regenerate the virtualenv.
```

BenjiBuilt, 2020-06-01:
Hey so i just restarted my wifi connection and tried the last command you said

sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh update-pip3-packages

and got this

pi@raspberrypi:~ $ sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh update-pip3-packages

#### Installing pip requirements
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='www.piwheels.org', port=443): Read timed out. (read timeout=15)")': /simple/pip/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='www.piwheels.org', port=443): Read timed out. (read timeout=15)")': /simple/pip/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='www.piwheels.org', port=443): Read timed out. (read timeout=15)")': /simple/pip/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='www.piwheels.org', port=443): Read timed out. (read timeout=15)")': /simple/pip/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='www.piwheels.org', port=443): Read timed out. (read timeout=15)")': /simple/pip/
Requirement already up-to-date: pip in ./env/lib/python3.7/site-packages (20.1.1)
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='www.piwheels.org', port=443): Read timed out. (read timeout=15)")': /simple/setuptools/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='www.piwheels.org', port=443): Read timed out. (read timeout=15)")': /simple/setuptools/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='www.piwheels.org', port=443): Read timed out. (read timeout=15)")': /simple/setuptools/
^CERROR: Operation cancelled by user
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='www.piwheels.org', port=443): Read timed out. (read timeout=15)")': /simple/alembic/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='www.piwheels.org', port=443): Read timed out. (read timeout=15)")': /simple/alembic/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='www.piwheels.org', port=443): Read timed out. (read timeout=15)")': /simple/alembic/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='www.piwheels.org', port=443): Read timed out. (read timeout=15)")': /simple/alembic/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='www.piwheels.org', port=443): Read timed out. (read timeout=15)")': /simple/alembic/
Collecting alembic==1.4.2
  Downloading alembic-1.4.2.tar.gz (1.1 MB)
     |████████████████████████████████| 1.1 MB 919 kB/s 
  Installing build dependencies ... error
  ERROR: Command errored out with exit status 1:
   command: /home/pi/Mycodo/env/bin/python /home/pi/Mycodo/env/lib/python3.7/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-ce7_eoyf/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- 'setuptools>=40.8.0' wheel
       cwd: None
  Complete output (20 lines):
  Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://www.piwheels.org/simple
  WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='www.piwheels.org', port=443): Read timed out. (read timeout=15)")': /simple/setuptools/
  WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='www.piwheels.org', port=443): Read timed out. (read timeout=15)")': /simple/setuptools/
  WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='www.piwheels.org', port=443): Read timed out. (read timeout=15)")': /simple/setuptools/
  WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='www.piwheels.org', port=443): Read timed out. (read timeout=15)")': /simple/setuptools/
[... cut ...]
Requirement already satisfied, skipping upgrade: zipp>=0.5 in /usr/local/lib/python3.7/dist-packages (from importlib-metadata>=0.12; python_version < "3.8"->pytest==5.4.2->-r /home/pi/Mycodo/install/requirements-testing.txt (line 4)) (3.1.0)
Requirement already satisfied, skipping upgrade: text-unidecode==1.3 in ./env/lib/python3.7/site-packages (from Faker>=0.7.0->factory_boy==2.12.0->-r /home/pi/Mycodo/install/requirements-testing.txt (line 5)) (1.3)
Requirement already satisfied, skipping upgrade: python-dateutil>=2.4 in ./env/lib/python3.7/site-packages (from Faker>=0.7.0->factory_boy==2.12.0->-r /home/pi/Mycodo/install/requirements-testing.txt (line 5)) (2.8.1)

kizniche, 2020-06-01:
You seem to have issues with your internet connection, as it's failing multiple times to download required files.

kizniche, 2020-06-01:
Let me build a virtualenv and see if I can.

BenjiBuilt, 2020-06-01:
Okay ill do the commands that you said before now to rebuild virtualenv

kizniche, 2020-06-01:
Downloaded fine for me, so the python package servers appear to be operational.

BenjiBuilt, 2020-06-01:
This is what I got 

pi@raspberrypi:~ $ sudo service mycodo stop
pi@raspberrypi:~ $ sudo service mycodoflask stop
pi@raspberrypi:~ $ mv ~/Mycodo/env ~/Mycodo/env-old
mv: cannot move '/home/pi/Mycodo/env' to '/home/pi/Mycodo/env-old/env': Permission denied
pi@raspberrypi:~ $ mv ~/Mycodo/env ~/Mycodo/env-old
mv: cannot move '/home/pi/Mycodo/env' to '/home/pi/Mycodo/env-old/env': Permission denied
pi@raspberrypi:~ $ sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh setup-virtualenv

#### Checking python 3 virtualenv
#### Virtualenv already exists, skipping creation
pi@raspberrypi:~ $ sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh update-pip3-packages

#### Installing pip requirements
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: pip in ./env/lib/python3.7/site-packages (20.1.1)
Requirement already up-to-date: setuptools in ./env/lib/python3.7/site-packages (47.1.1)
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: alembic==1.4.2 in ./env/lib/python3.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 2)) (1.4.2)
Requirement already up-to-date: argparse==1.4.0 in ./env/lib/python3.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 3)) (1.4.0)
Requirement already up-to-date: bcrypt==3.1.7 in ./env/lib/python3.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 4)) (3.1.7)
Requirement already up-to-date: daemonize==2.5.0 in ./env/lib/python3.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 5)) (2.5.0)
Requirement already up-to-date: distro==1.5.0 in ./env/lib/python3.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 6)) (1.5.0)
Requirement already up-to-date: email_validator==1.1.1 in ./env/lib/python3.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 7)) (1.1.1)
Requirement already up-to-date: filelock==3.0.12 in /usr/local/lib/python3.7/dist-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 8)) (3.0.12)
Requirement already up-to-date: Flask==1.1.2 in ./env/lib/python3.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 9)) (1.1.2)
Requirement already up-to-date: Flask_Accept==0.0.6 in ./env/lib/python3.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 10)) (0.0.6)
Requirement already up-to-date: Flask_Babel==1.0.0 in ./env/lib/python3.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 11)) (1.0.0)
Requirement already up-to-date: Flask_Compress==1.4.0 in ./env/lib/python3.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 12)) (1.4.0)
Requirement already up-to-date: Flask_Limiter==1.1.0 in ./env/lib/python3.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 13)) (1.1.0)
Requirement already up-to-date: Flask_Login==0.4.1 in ./env/lib/python3.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 14)) (0.4.1)
Requirement already up-to-date: Flask_Marshmallow==0.10.1 in ./env/lib/python3.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 15)) (0.10.1)
Requirement already up-to-date: Flask_RESTX==0.2.0 in ./env/lib/python3.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 16)) (0.2.0)
Requirement already up-to-date: Flask_SQLAlchemy==2.4.1 in ./env/lib/python3.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 17)) (2.4.1)
Requirement already up-to-date: Flask_SSLify==0.1.5 in ./env/lib/python3.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 18)) (0.1.5)
Requirement already up-to-date: Flask_WTF==0.14.3 in ./env/lib/python3.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 19)) (0.14.3)
Requirement already up-to-date: geocoder==1.38.1 in ./env/lib/python3.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 20)) (1.38.1)
Requirement already up-to-date: gunicorn==20.0.4 in ./env/lib/python3.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 21)) (20.0.4)
Requirement already up-to-date: influxdb==5.3.0 in ./env/lib/python3.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 22)) (5.3.0)
Requirement already up-to-date: marshmallow_sqlalchemy==0.21.0 in ./env/lib/python3.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 23)) (0.21.0)
Requirement already up-to-date: pyro5==5.10 in ./env/lib/python3.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 24)) (5.10)
Requirement already up-to-date: pyserial in /usr/lib/python3/dist-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 25)) (3.4)
Requirement already up-to-date: python-dateutil==2.8.1 in ./env/lib/python3.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 26)) (2.8.1)
Requirement already up-to-date: requests in ./env/lib/python3.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 27)) (2.23.0)
Requirement already up-to-date: SQLAlchemy==1.3.17 in ./env/lib/python3.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 28)) (1.3.17)
Requirement already up-to-date: Werkzeug==1.0.1 in ./env/lib/python3.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 29)) (1.0.1)
[... cut ...] ation AS output_uart_location, output.baud_rate AS output_baud_rate, output.pwm_hertz AS output_pwm_hertz, output.pwm_library AS output_pwm_library, output.pwm_invert_signal AS output_pwm_invert_signal, output.flow_rate AS output_flow_rate, output.custom_options AS output_custom_options 
FROM output]
(Background on this error at: http://sqlalche.me/e/e3q8)
pi@raspberrypi:~ $ sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh update-permissions

#### Setting permissions
pi@raspberrypi:~ $

kizniche, 2020-06-01:
This is good... Now run:

```
cd ~/Mycodo/databases
../env/bin/alembic upgrade head
```

BenjiBuilt, 2020-06-01:
pi@raspberrypi:~ $ cd ~/Mycodo/databases
pi@raspberrypi:~/Mycodo/databases $ ../env/bin/alembic upgrade head
INFO  [alembic.runtime.migration] Context impl SQLiteImpl.
INFO  [alembic.runtime.migration] Will assume non-transactional DDL.
INFO  [alembic.runtime.migration] Running upgrade dc9eddfc845d -> 267dc913a062, Add log_level_debug, message_include_code, force_command
INFO  [alembic.runtime.migration] Running upgrade 267dc913a062 -> 4b5f6207cbdf, Add Camera options
INFO  [alembic.runtime.migration] Running upgrade 4b5f6207cbdf -> f5b77ef5f17c, Change SMTP DB options
INFO  [alembic.runtime.migration] Running upgrade f5b77ef5f17c -> 61a0d0568d24, Add password reset capabilities
INFO  [alembic.runtime.migration] Running upgrade 61a0d0568d24 -> e5bd30e798b8, Add Linux Command Output option: user to execute with
INFO  [alembic.runtime.migration] Running upgrade e5bd30e798b8 -> caf50eb4f236, Add Output Log Level option
INFO  [alembic.runtime.migration] Running upgrade caf50eb4f236 -> 840c4d18e38c, Add output custom_options
pi@raspberrypi:~/Mycodo/databases $

kizniche, 2020-06-01:
Excellent. Now these commands:

```
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh update-dependencies
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh update-permissions
sudo service mycodo restart
sudo service mycodoflask restart
```

You should then have everything working.

BenjiBuilt, 2020-06-01:
Got this

pi@raspberrypi:~ $ sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh update-dependencies

#### Checking for updates to dependencies
Unmet dependencies found: [('w1thermsensor', 'pip-pypi w1thermsensor', 'pip-pypi', 'w1thermsensor')]

#### Installing/updating w1thermsensor (pip-pypi)
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting w1thermsensor
  Using cached w1thermsensor-1.3.0-py2.py3-none-any.whl (14 kB)
Requirement already satisfied, skipping upgrade: click in /usr/lib/python3/dist-packages (from w1thermsensor) (7.0)
Installing collected packages: w1thermsensor
Successfully installed w1thermsensor-1.3.0

dpkg-query: no packages found matching bluez-hcidump
dpkg-query: no packages found matching libglib2.0-dev
dpkg-query: no packages found matching libjpeg-dev
dpkg-query: no packages found matching libglib2.0-dev
dpkg-query: no packages found matching fswebcam
dpkg-query: no packages found matching python3-opencv
dpkg-query: no packages found matching python3-opencv
dpkg-query: no packages found matching python3-opencv
dpkg-query: no packages found matching liblircclient-dev
dpkg-query: no packages found matching liblircclient-dev
dpkg-query: no packages found matching libjpeg-dev
dpkg-query: no packages found matching libjpeg-dev
E: Sub-process /usr/bin/dpkg returned an error code (1)
Reading package lists...
Building dependency tree...
Reading state information...
lirc is already the newest version (0.10.1-5.2).
python3-dev is already the newest version (3.7.3-1).
python3-numpy is already the newest version (1:1.16.2-1).
python3-psutil is already the newest version (5.5.1-1).
zlib1g-dev is already the newest version (1:1.2.11.dfsg-1).
bluez is already the newest version (5.50-1.2~deb10u1+rpt1).
bluez set to manually installed.
The following packages were automatically installed and are no longer required:
  libpython-all-dev python-all-dev python-configparser python-entrypoints
  python-keyring python-keyrings.alt python-secretstorage python-wheel
  python-xdg
Use 'sudo apt autoremove' to remove them.
Suggested packages:
  owserver
The following packages will be upgraded:
  ow-shell pi-bluetooth
2 upgraded, 0 newly installed, 0 to remove and 178 not upgraded.
1 not fully installed or removed.
Need to get 34.8 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://archive.raspberrypi.org/debian buster/main armhf pi-bluetooth all 0.1.13 [4,908 B]
Get:2 http://raspbian.melbourneitmirror.net/raspbian buster/main armhf ow-shell armhf 3.2p3+dfsg1-2+deb10u1 [29.9 kB]
apt-listchanges: Reading changelogs...
Fetched 34.8 kB in 1s (35.4 kB/s)
(Reading database ... 154704 files and directories currently installed.)
Preparing to unpack .../ow-shell_3.2p3+dfsg1-2+deb10u1_armhf.deb ...
Unpacking ow-shell (3.2p3+dfsg1-2+deb10u1) over (3.2p3+dfsg1-2) ...
Preparing to unpack .../pi-bluetooth_0.1.13_all.deb ...
Unpacking pi-bluetooth (0.1.13) over (0.1.12) ...
Setting up pi-bluetooth (0.1.13) ...
Setting up lirc (0.10.1-5.2) ...
Job for lircd.service failed because a fatal signal was delivered to the control process.
See "systemctl status lircd.service" and "journalctl -xe" for details.
invoke-rc.d: initscript lircd, action "start" failed.
● lircd.service - Flexible IR remote input/output application support
   Loaded: loaded (/lib/systemd/system/lircd.service; disabled; vendor preset: enabled)
   Active: failed (Result: signal) since Mon 2020-06-01 14:52:27 AEST; 29ms ago
     Docs: man:lircd(8)
           http://lirc.org/html/configure.html
[... 62 lines cut ...]
Collecting Pillow
  Downloading https://www.piwheels.org/simple/pillow/Pillow-7.1.2-cp37-cp37m-linux_armv7l.whl (1.3 MB)
Requirement already up-to-date: picamera in /usr/lib/python3/dist-packages (from -r /home/pi/Mycodo/install/requirements-generated.txt (line 6)) (1.13)
Collecting python-lirc
  Downloading python-lirc-1.2.3.tar.gz (36 kB)
Requirement already satisfied, skipping upgrade: click in /usr/lib/python3/dist-packages (from w1thermsensor->-r /home/pi/Mycodo/install/requirements-generated.txt (line 1)) (7.0)
Building wheels for collected packages: python-lirc
  Building wheel for python-lirc (setup.py): started
  Building wheel for python-lirc (setup.py): finished with status 'error'
  Running setup.py clean for python-lirc
Failed to build python-lirc
Installing collected packages: Pillow, python-lirc
  Attempting uninstall: Pillow
    Found existing installation: Pillow 5.4.1
    Not uninstalling pillow at /usr/lib/python3/dist-packages, outside environment /home/pi/Mycodo/env
    Can't uninstall 'Pillow'. No files were found to uninstall.
    Running setup.py install for python-lirc: started
    Running setup.py install for python-lirc: finished with status 'error'

pi@raspberrypi:~ $ sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh update-permissions

#### Setting permissions
pi@raspberrypi:~ $ sudo service mycodo restart
pi@raspberrypi:~ $ sudo service mycodoflask restart
pi@raspberrypi:~ $

BenjiBuilt, 2020-06-01:
But now the UI is working again !!! You absolute legend!!

I need this thing to be really stable..

Would this have caused any instability issues now? 
Is it worth me making abackup of the setup and reinstalling mycodo fresh or something?

kizniche, 2020-06-01:
Did yo by chance upgrade to 8.5.0 before 8.5.1?

BenjiBuilt, 2020-06-01:
I just used the upgrade feature in the UI Im pretty sure it said upgrade 8.3 - 8.5.1

kizniche, 2020-06-01:
I'm not sure what files during your upgrade. I suspect there was some file corruption in your virtualenv. In any case, it appears your virtualenv is fixed (it was completely rebuilt) and your database has been upgraded. There shouldn't be any lasting issues from this issue, except if you have some sort of filesystem problem (e.g. SD going bad). I would recommend exporting your settings database from the inport/export menu and save it as a backup that you can restore later if something bad happens. It's always a good idea to export it before doing an upgrade ni case something like this happens. That way you can always reinstall Mycodo fresh and import your settings database and be exactly back where you were.

BenjiBuilt, 2020-06-01:
Mate, you are an absolute champion and have really saved my butt today!
Thankyou so much for the fast response and indepth help!

Can I donate to the mycodo cause somewhere? You really deserve it

kizniche, 2020-06-01:
You're welcome. Glad you caught me right before I turned off my computer to go to sleep. I have a donate link in the readme at https://github.com/kizniche/Mycodo/blob/master/README.rst#donate Thanks for the consideration.

BenjiBuilt, 2020-06-01:
Just sent a donation, it's not alot but its all I could afford. Thankyou sir.

closed 2020-06-01 by kizniche