kizniche/Mycodo #777: CCS811 sensor works fine in the CLI python test script but will not show any data in Mycodo

infinisean, 2020-06-03:
Hi All.
New here so please be gentle  :)

I have a new Pi4b with the latest version of Mycodo and Buster and my CCS811 sensor will not read any data. It works fine in Arduino code and also on the CLI with the included python library and test script (see below).

Any guidance as to what I could be doing wrong would be greatly appreciated.
Any other specific info you need to help troubleshoot, please let me know and I will get it posted right away.

Thanks!

--Sean

### Describe the problem/bug
ccs811 sensor is setup and activated but it will not read any data. 

**i2c address is detected correctly in Mycodo (5a) and matches what i2cdetect shows on the CLI**

pi@MushPi4:/var/log $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- 5a -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

It also shows "sensor error" in daemon logs

2020-06-03 05:38:42,991 - ERROR - mycodo.controllers.controller_input_25f3fb05 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-06-03 05:38:46,023 - ERROR - mycodo.inputs.ccs811_25f3fb05 - Sensor error
2020-06-03 05:38:49,068 - ERROR - mycodo.inputs.ccs811_25f3fb05 - Sensor error

Using the simple test script provided with the library works fine from the command-line:
(py3) pi@MushPi4:~ $ python3 ccs.py
CO2: 0 PPM, TVOC: 0 PPB
CO2: 0 PPM, TVOC: 0 PPB
CO2: 434 PPM, TVOC: 5 PPB
CO2: 428 PPM, TVOC: 4 PPB
CO2: 428 PPM, TVOC: 4 PPB
CO2: 428 PPM, TVOC: 4 PPB
CO2: 428 PPM, TVOC: 4 PPB
CO2: 424 PPM, TVOC: 3 PPB
CO2: 424 PPM, TVOC: 3 PPB
CO2: 428 PPM, TVOC: 4 PPB
CO2: 428 PPM, TVOC: 4 PPB
CO2: 428 PPM, TVOC: 4 PPB
CO2: 428 PPM, TVOC: 4 PPB
CO2: 428 PPM, TVOC: 4 PPB
CO2: 428 PPM, TVOC: 4 PPB
CO2: 428 PPM, TVOC: 4 PPB
CO2: 434 PPM, TVOC: 5 PPB

A clear and concise description of what you were trying to do and what the problem/bug is.

### Versions:
Mycodo Version: **8.5.2**
Python Version: 3.7.3 (default, Dec 20 2019, 18:57:59) [GCC 8.3.0]
Database Version: 840c4d18e38c
Daemon Status: Running
Frontend RAM Usage: 63.9 MB
Frontend Virtualenv: Yes

 - Raspberry Pi Version: [4B]
 - Raspbian OS Version: [Buster Lite]

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

Add data input for CCS811 in Mycodo, configure, activate (all done successfully)
Check live data (none) and logs (error pasted above)

### Expected behavior
See valid data...

### Screenshots

The data above should be sufficient but I am willing to provide anything else specifically needed

kizniche, 2020-06-06:
What is ccs.py?

infinisean, 2020-06-06:
That is the simple test script included with the library which works fine on the CLI (so I know the hardware , library and wiring are good)

kizniche, 2020-06-06:
do you have a link or attachment?

infinisean, 2020-06-06:
**Guide:** 
https://learn.adafruit.com/adafruit-ccs811-air-quality-sensor/raspberry-pi-wiring-test

**Git:**
git clone https://github.com/adafruit/Adafruit_CCS811_python.git
cd Adafruit_CCS811_python/examples
sudo python CCS811_example.py

**Code:**
pi@MuchPi4:~/Adafruit_CCS811_python/examples $ cat CCS811_example.py
#this example reads and prints CO2 equiv. measurement, TVOC measurement, and temp every 2 seconds

from time import sleep
from Adafruit_CCS811 import Adafruit_CCS811

ccs =  Adafruit_CCS811()

while not ccs.available():
        pass
temp = ccs.calculateTemperature()
ccs.tempOffset = temp - 25.0

while(1):
        if ccs.available():
            temp = ccs.calculateTemperature()
            if not ccs.readData():
              print "CO2: ", ccs.geteCO2(), "ppm, TVOC: ", ccs.getTVOC(), " temp: ", temp

            else:
              print "ERROR!"
              while(1):
                pass
        sleep(2)

kizniche, 2020-06-06:
The issue is likely from my attempt to set the bus number:

https://github.com/kizniche/Mycodo/blob/09f451f19eaf9d14a3cab876593edfab70730b48/mycodo/inputs/ccs811.py#L64-L66

You can try removing "busnum=self.i2c_bus", saving, restarting the backend, and seeing if it works.

kizniche, 2020-06-06:
Though, it should work (see below).

https://github.com/adafruit/Adafruit_CCS811_python/blob/b75196d2f887247a316a631b81c4a909da395f3d/Adafruit_CCS811/Adafruit_CCS811.py#L58-L59

https://github.com/adafruit/Adafruit_CCS811_python/blob/b75196d2f887247a316a631b81c4a909da395f3d/Adafruit_CCS811/Adafruit_CCS811.py#L65-L69

https://github.com/adafruit/Adafruit_Python_GPIO/blob/a12fee39839665966bd124fd22588b2c87ced9d2/Adafruit_GPIO/I2C.py#L59-L66

infinisean, 2020-06-11:
I actually switched to another sensor (SCD30) which has Temp, Humidity and CO2 all in one and is more accurate (via infrared CO2 measurement) and also more stable than the CCS811.

I'm currently using the linux bash command to query a RAM-based file the sensor's driver creates with all the values and it is working beautifully, though native sensor support would obviously be cleaner and more user friendly.

If you'd be interested in adding support for this sensor, I'd be happy to provide you with any information you need from my working environment to do so as it is a fantastic sensor.

kizniche, 2020-06-11:
Do you have any links to literature on the SCD30 and what specific board/sensor(s) you're using?

infinisean, 2020-06-11:
This is the library I am using (which creates a file in RAM with the 3 
values to read with grep / awk)

https://github.com/sparkfun/SparkFun_SCD30_Arduino_Library

This is the board:

https://www.sparkfun.com/products/15112

Datasheet

https://cdn.sparkfun.com/assets/1/a/7/9/9/Sensirion_CO2_Sensors_SCD30_Preliminary-Datasheet.pdf

Interface

https://cdn.sparkfun.com/assets/f/e/d/0/1/Sensirion_CO2_Sensors_SCD30_Interface_Description.pdf

Everything you need is in the driver though as it is accessed via I2C.

Let me know what else I can get for you from my working environment.

Happy to help

--Sean

kizniche, 2020-06-11:
You linked what appears to be an arduino library. Are you not using it on a Raspberry Pi?

kizniche, 2020-06-11:
From your description of how you're reading the sensor, it appears you're using this: https://github.com/UnravelTEC/Raspi-Driver-SCD30

kizniche, 2020-06-11:
Could you possibly test this? https://github.com/postronium/SCD30-Raspberry-Pi

kizniche, 2020-06-16:
Have you had a chance to test [postronium/SCD30-Raspberry-Pi](https://github.com/postronium/SCD30-Raspberry-Pi)?

kizniche, 2020-06-16:
Okay, I've given up on that library and am using Raspi-Driver-SCD30. Can you provide the output from the command ```cat /run/sensors/scd30/last```?

kizniche, 2020-06-16:
Better yet, can you add this code to the file ```test_read_scd30.py``` and execute it with ```~/Mycodo/env/bin/python ~/test_read_scd30.py```?

```python
with open('/run/sensors/scd30/last') as scd30_read:
    print("'{}'".format(scd30_read.read()))
```

infinisean, 2020-06-16:
Yes, I should be able to do this tomorrow. I'll get back to you with 
results.

infinisean, 2020-06-17:
pi@MushPi4:~/Mycodo/env/bin $ cat /run/sensors/scd30/last

gas_ppm{sensor="SCD30",gas="CO2"} 1544.25573730
temperature_degC{sensor="SCD30"} 29.05889893
humidity_rel_percent{sensor="SCD30"} 78.48205566

kizniche, 2020-06-17:
Attached is a module for testing. You can import it from the Configure -> Inputs page, then add it from the Setup -> Data page. You can enable Log Level: Debug in the Input options to see debug line output in the Daemon Log.

[scd30.py.txt](https://github.com/kizniche/Mycodo/files/4795504/scd30.py.txt)

infinisean, 2020-06-18:
I did that but I am getting an error saying unmet dependencies (pigpiod)
but I have that installed.,.

pi@MushPi4:~/Mycodo $ grep -i pigpio ../installed
libpigpio-dev/testing,now 1.71-0~rpt1 armhf [installed]
libpigpio1/testing,now 1.71-0~rpt1 armhf [installed]
libpigpiod-if-dev/testing,now 1.71-0~rpt1 armhf [installed]
libpigpiod-if1/testing,now 1.71-0~rpt1 armhf [installed]
libpigpiod-if2-1/testing,now 1.71-0~rpt1 armhf [installed]
pigpio-tools/testing,now 1.71-0~rpt1 armhf [installed]
pigpio/testing,now 1.71-0~rpt1 armhf [installed]
pigpiod/testing,now 1.71-0~rpt1 armhf [installed]
python-pigpio/testing,now 1.71-0~rpt1 all [installed]
python3-pigpio/testing,now 1.71-0~rpt1 all [installed]

kizniche, 2020-06-18:
If you didn't install pigpiod from Mycodo, it doesn't know it's installed, since Mycodo handles setting the sample rate. Go ahead and let Mycodo install pigpio, as this will fulfill the dependency and shouldn't pose an issue for anything else using pigpiod.

infinisean, 2020-07-07:
Kyle,

Sorry I haven't gotten back to you in a while... had some family stuff 
going on so I am just now getting back to this project.

I'll give the pigpiod a shot tomorrow and let you know what I find with 
that. Thanks for the info.

Just ran into a strange issue trying to get my first PID working... was 
hoping you might be able to point me in the right direction how to 
troubleshoot something like this.

2020-07-06 23:55:29,662 - ERROR - mycodo.controllers.controller_pid_f72bde02 - Exception executing check_pid() on attempt 2 of 3. Waiting 10 seconds and trying again.
Traceback (most recent call last):
   File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 124, in attempt_execute
     func()
   File "/var/mycodo-root/mycodo/controllers/controller_pid.py", line 316, in check_pid
     duration_sec=self.setpoint_tracking_max_age)
   File "/var/mycodo-root/mycodo/utils/influx.py", line 468, in read_last_influxdb
     measure=measure, duration_sec=duration_sec, value='LAST')
   File "/var/mycodo-root/mycodo/utils/influx.py", line 440, in read_influxdb_single
     last_measurement = client.query(query).raw
   File "/var/mycodo-root/env/lib/python3.7/site-packages/influxdb/client.py", line 518, in query
     expected_response_code=expected_response_code
   File "/var/mycodo-root/env/lib/python3.7/site-packages/influxdb/client.py", line 369, in request
     raise InfluxDBClientError(err_msg, response.status_code)
influxdb.exceptions.InfluxDBClientError: 400: {"error":"error parsing query: found s, expected ; at line 1, char 151"}
2020-07-06 23:55:39,749 - ERROR - mycodo.controllers.controller_pid_f72bde02 - Exception executing check_pid() on attempt 3 of 3.
Traceback (most recent call last):
   File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 124, in attempt_execute
     func()
   File "/var/mycodo-root/mycodo/controllers/controller_pid.py", line 316, in check_pid
     duration_sec=self.setpoint_tracking_max_age)
   File "/var/mycodo-root/mycodo/utils/influx.py", line 468, in read_last_influxdb
     measure=measure, duration_sec=duration_sec, value='LAST')
   File "/var/mycodo-root/mycodo/utils/influx.py", line 440, in read_influxdb_single
     last_measurement = client.query(query).raw
   File "/var/mycodo-root/env/lib/python3.7/site-packages/influxdb/client.py", line 518, in query
     expected_response_code=expected_response_code
   File "/var/mycodo-root/env/lib/python3.7/site-packages/influxdb/client.py", line 369, in request
     raise InfluxDBClientError(err_msg, response.status_code)
influxdb.exceptions.InfluxDBClientError: 400: {"error":"error parsing query: found s, expected ; at line 1, char 151"}
2020-07-06 23:55:49,517 - ERROR - mycodo.controllers.controller_pid_f72bde02 - Exception executing check_pid() on attempt 1 of 3. Waiting 10 seconds and trying again.
Traceback (most recent call last):
   File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 124, in attempt_execute
     func()
   File "/var/mycodo-root/mycodo/controllers/controller_pid.py", line 316, in check_pid
     duration_sec=self.setpoint_tracking_max_age)
   File "/var/mycodo-root/mycodo/utils/influx.py", line 468, in read_last_influxdb
     measure=measure, duration_sec=duration_sec, value='LAST')
   File "/var/mycodo-root/mycodo/utils/influx.py", line 440, in read_influxdb_single

kizniche, 2020-07-07:
I was just informed about a bug with the pigpiod install within Mycodo (#783), so currently it is broken. I am working on getting this released, but I'm right in the middle of a big code change with the Output and PID system.

I'll look into your PID errors.

infinisean, 2020-07-07:
Understood. No worries.

Thanks for the quick reply.

Let me know if I can help at all with testing or whatever.

kizniche, 2020-07-07:
I just released v8.5.7, a bugfix release that fixes the pigpio install issue.

infinisean, 2020-07-07:
Is that the cause of the pid issue as well?

kizniche, 2020-07-07:
I actually just released v8.5.8 that really fixes the pigpio install issue. I haven't yet looking into your PID issue.

kizniche, 2020-07-23:
I just released v8.6.0, which fixes even more pigpiod issues I found recently. After upgrade, I'd recommend going to Configure -> Raspberry Pi and uninstalling pigpio and reinstalling it, from the web interface.

closed 2020-08-11 by kizniche