kizniche/Mycodo #778: Humidity Sensor and EZO PH Sensor keep dropping out

BenjiBuilt, 2020-06-05:
Hey Kizniche,

You helped me the other day rebuilding my environment after the update.

I have this recurring issue.

After about 1 day of being online the DHT22 sensor drops out and says NO DATA. Then about 6 hours later the PH Reading Drops out and goes either 0pH or 14pH.

If I give the whole raspberry Pi a reboot it resolves itself until that time passes again. But if I give back and front end a reboot from the mycodo GUI it doesnt fix anything.

Do you have any ideas what could be going on??

Thanks

Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue.

### Describe the problem/bug

A clear and concise description of what you were trying to do and what the problem/bug is.

### Versions:

 - Mycodo Version: [e.g. 8.0.3]
 - Raspberry Pi Version: [e.g. 3B+]
 - Raspbian OS Version: [e.g. Buster Lite]

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior

A clear and concise description of what you expected to happen.

### Screenshots

If applicable, add screenshots to help explain your problem.

### Additional context

Is there anything that should be added to make it easier to address this issue?

kizniche, 2020-06-05:
Are there errors in the Daemon Log?

BenjiBuilt, 2020-06-05:
2020-06-05 21:47:43,401 - ERROR - mycodo.controllers.controller_input_74b5cda3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-06-05 21:49:13,375 - ERROR - mycodo.controllers.controller_input_74b5cda3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-06-05 21:50:43,467 - ERROR - mycodo.controllers.controller_input_74b5cda3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-06-05 21:52:13,395 - ERROR - mycodo.controllers.controller_input_74b5cda3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-06-05 21:53:43,373 - ERROR - mycodo.controllers.controller_input_74b5cda3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-06-05 21:55:13,376 - ERROR - mycodo.controllers.controller_input_74b5cda3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-06-05 21:56:43,471 - ERROR - mycodo.controllers.controller_input_74b5cda3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-06-05 21:58:13,458 - ERROR - mycodo.controllers.controller_input_74b5cda3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-06-05 21:59:43,443 - ERROR - mycodo.controllers.controller_input_74b5cda3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-06-05 22:00:25,146 - INFO - mycodo.controllers.controller_trigger_221a803c - Activated in 97.4 ms
2020-06-05 22:00:25,268 - INFO - mycodo.controllers.controller_trigger_f4d6b37e - Deactivated in 80.0 ms
2020-06-05 22:00:32,686 - INFO - mycodo.controllers.controller_trigger_f4d6b37e - Activated in 170.0 ms
2020-06-05 22:00:32,865 - INFO - mycodo.controllers.controller_trigger_221a803c - Deactivated in 119.3 ms
2020-06-05 22:00:54,996 - INFO - mycodo.controllers.controller_trigger_221a803c - Activated in 101.8 ms
2020-06-05 22:00:55,244 - INFO - mycodo.controllers.controller_trigger_f4d6b37e - Deactivated in 228.2 ms
2020-06-05 22:01:13,466 - ERROR - mycodo.controllers.controller_input_74b5cda3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-06-05 22:02:43,466 - ERROR - mycodo.controllers.controller_input_74b5cda3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-06-05 22:04:13,447 - ERROR - mycodo.controllers.controller_input_74b5cda3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-06-05 22:05:43,455 - ERROR - mycodo.controllers.controller_input_74b5cda3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-06-05 22:07:13,433 - ERROR - mycodo.controllers.controller_input_74b5cda3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-06-05 22:08:43,421 - ERROR - mycodo.controllers.controller_input_74b5cda3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-06-05 22:10:13,440 - ERROR - mycodo.controllers.controller_input_74b5cda3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-06-05 22:11:43,418 - ERROR - mycodo.controllers.controller_input_74b5cda3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-06-05 22:13:13,394 - ERROR - mycodo.controllers.controller_input_74b5cda3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-06-05 22:14:43,401 - ERROR - mycodo.controllers.controller_input_74b5cda3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-06-05 22:16:13,469 - ERROR - mycodo.controllers.controller_input_74b5cda3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-06-05 22:17:43,462 - ERROR - mycodo.controllers.controller_input_74b5cda3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-06-05 22:19:13,443 - ERROR - mycodo.controllers.controller_input_74b5cda3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-06-05 22:20:43,436 - ERROR - mycodo.controllers.controller_input_74b5cda3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-06-05 22:22:13,404 - ERROR - mycodo.controllers.controller_input_74b5cda3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-06-05 22:23:43,433 - ERROR - mycodo.controllers.controller_input_74b5cda3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-06-05 22:25:13,405 - ERROR - mycodo.controllers.controller_input_74b5cda3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
[... cut ...] ler_input_74b5cda3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-06-05 22:28:13,471 - ERROR - mycodo.controllers.controller_input_74b5cda3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-06-05 22:29:43,400 - ERROR - mycodo.controllers.controller_input_74b5cda3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.

kizniche, 2020-06-05:
It seems Mycodo is operating properly, and there's been no change to the input modules, so this is likely an issue with the hardware or your system itself (Raspbian). I'd recommend performing ```sudo apt update && sudo apt upgrade``` and reboot to upgrade your kernel and firmware.

BenjiBuilt, 2020-06-05:
Im running that now, so Ill let you know when its done how it goes. 

Im not really certain on how it could be hardware if its fixed when I reboot it but youre the guru.

kizniche, 2020-06-05:
> Im not really certain on how it could be hardware if its fixed when I reboot

> or your system itself (Raspbian)

kizniche, 2020-06-05:
fyi, my system that has the same pH sensor and is on the latest version has had no issues.

BenjiBuilt, 2020-06-06:
2020-06-06 15:26:26,393 - ERROR - mycodo.controllers.controller_input_74b5cda3 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.

still doing this after full update. 
So you think it could be hardware related. I wonder what would cause it to be intermittent on PH and DHT22?

kizniche, 2020-06-06:
There are a lot of hardware-related issues that can cause a sensor not to work properly. I'd suggest disconnecting everything from your Pi and only connecting each input one at a time on the shortest wires possible to test each in isolation.

closed 2020-07-23 by kizniche