kizniche/Mycodo #779: GPIO Outputs won't show in dashboard after upgrade to 8.5.1

Bart9999, 2020-06-06:
I also have some issues with the GPIO outputs. I've configured the system in version 8.4.0 Everything works fine. Today i have added an output: Ontvochtiger
[image]

I can refer to this output in the PID controller:
[image]

Unfortunately i can not choose this output in the dashboard (not in the Output widget and also not in the Indicator widget)
[image]
[image]

For some reason only the outputs are visible that were created before the upgrade. Any idea what i'm doing wrong?

kizniche, 2020-06-06:
Thank you for the detailed info. I'll investigate and get back to you.

Bart9999, 2020-06-06:
BTW, I also tried unistalling Mycodo and after that installed Mycodo completely from the beginning by the command:  curl -L https://kizniche.github.io/Mycodo/install | bash

The same issue appeared.

kizniche, 2020-06-06:
I was able to replicate the issue. Working on a fix.

kizniche, 2020-06-06:
Just committed a fix and released v8.5.3. Thanks for the bug report.

closed 2020-06-06 by kizniche

Bart9999, 2020-06-06:
Thanks. Outputs are showing again in the dashboard :-)

kizniche, 2020-06-06:
Thanks for confirming the fix works out in the wild.