kizniche/Mycodo #780: New Project: Automated Hydroponic System Build

kizniche, 2020-06-09:
For anyone interested to see how I set up my automated hydroponic system, I just released an article and video detailing the process.

Article: https://kylegabriel.com/projects/2020/06/automated-hydroponic-system-build.html

Video: https://youtu.be/nyqykZK2Ev4

zsole2, 2020-06-10:
This looks absolutely fantastic! 
I am getting nostalgic how I followed your write-up on mushroom cultivation back in 2015 (wow, eons ago!) to apply for cheese-making. Maybe this time around if I get the itch I just simply copy this, it looks as perfect as it can be. Great job, as usual!

Theoi-Meteoroi, 2020-06-17:
I watch youtube, alot.   This is really awesome work there, Kyle.  
I was actually looking at another implementation that used Home Assistant and played with that for a bit but I have so much time invested in Mycodo that I balked, eventually.   
You have inspired me (again) with this work to engage.   I've come to find that sourcing parts is becoming more of an issue and I'll see if I can help.  Folks are poking me to restock and that hasn't happened before.

kizniche, 2020-06-18:
@Theoi-Meteoroi Thanks! I think I watched the same video (was it https://www.youtube.com/watch?v=Q9fjKeYOyqU ?). In the middle of that vid, my head was spinning from its complexity (not difficulty, but the amount of devices and setup required). I've seen a lot of products ordered from my build writeup in the few days it's been live. When you get your parts in stock, I'd be happy to add a link to your store to the materials list, if you would like.

Surferpup, 2021-03-23:
This looks fantastic!!!  I am going to use your ideas to build my system.  I have seen  a few users that use NFT for leafy vegetables and add a tomato plant in a dutch bucket -- all off of the same reservoir.  Have you had any luck with that?

kizniche, 2021-03-26:
> Have you had any luck with that?

I haven't yet modified the NFT for any add-ons yet. I would like to add some wider channels and an additional small pump to start growing microgreens along side the lettuces and herbs.

Surferpup, 2021-03-26:
I am also looking into using NIR (Near Infrared) imaging and AI Image analysis to try and determine health of plants -- maybe to determine Nitrogen content of leaves.  Do you know of any sensors that can analyze nutritional makeup of leaves?  I am trying to come up with ways to inform the system to either backoff on nutrients or increase them.  I am the hardware guy, and my son is a AI Computer Scientist.  We are both really combing over your project.

Ben408, 2021-03-30:
> I am also looking into using NIR (Near Infrared) imaging and AI Image analysis to try and determine health of plants -- maybe to determine Nitrogen content of leaves. Do you know of any sensors that [...]

I've been looking at this as well. Seedo was doing this in their factory in Israel, they seem to have recently gone bankrupt. There is some equipment out there for this:
https://cid-inc.com/plant-science-tools/photosynthesis-measurement/ci-340-handheld-photosynthesis-system/accessories/control-modules/ci-510cf-chlorophyll-fluorescence/

I'm new to this project as well, and impressed so far. I plan to start with the reservoir management and see how I can port the irrigation management to what I need. I have 2 systems running that use a more basic "count the green pixels" from the camera to control nutrient dosing.
https://www.instagram.com/bjcornelius/

closed 2021-09-20 by kizniche