kizniche/Mycodo #781: Daemon does not start

RynFlutsch, 2020-06-15:
Hey

Upgraded to 8.5.4.
No Daemon logs since than.
Error:
Error: cannot connect to ('127.0.0.1',9090): [Errno 111] Connection refused

kizniche, 2020-06-15:
Did you inspect the upgrade log?
Check if any errors result from starting the daemon manually with ```~/Mycodo/env/bin/python ~/Mycodo/mycodo/mycodo_daemon.py```

RynFlutsch, 2020-06-16:
Yes no errors.

Web.log:
gunicorn[568]: 2020-06-16 16:02:47,145 URL for 'daemon_active' raised and error: cannot connect to ('127.0.0.1', 9090): [Errno 111] Connection refused

~/Mycodo/env/bin/python ~/Mycodo/mycodo/mycodo_daemon.py
shows following error:
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/mycodo_daemon.py", line 40, in <module>
    from Pyro5.api import serve
ImportError: cannot import name 'serve' from 'Pyro5.api' (/home/pi/Mycodo/env/lib/python3.7/site-packages/Pyro5/api.py)

kizniche, 2020-06-16:
Looks like you have some file corruption or an issue with your python package Pyro5. This is usually the case when the daemon doesn't start. I'd recommend either starting on a new SD card or use the same SD that could be going bad and either reinstall Raspbian, Mycodo, or the virtualenv.

You can find the commands to rebuild the virtualenv [here](https://github.com/kizniche/Mycodo/issues/776#issuecomment-636606371).

kizniche, 2020-06-16:
I would first try rebuilding the virtualenv before the other recommendations.

RynFlutsch, 2020-06-16:
Hm dont know i downgraded to 8.4 now it working again. I upgrade later if i have time to fix Problems.

closed 2020-06-16 by RynFlutsch