kizniche/Mycodo #782: No Data from AM2315

RynFlutsch, 2020-06-23:
8.5.4

No PID activated and don't get measurements.

Demon.log

`2020-06-22 17:31:41,723 - ERROR - mycodo.controllers.controller_input_5fea3f94 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-06-22 17:31:48,208 - ERROR - mycodo.inputs.am2315_5fea3f94 - All measurements returned failed CRC
2020-06-22 17:31:56,595 - ERROR - mycodo.inputs.am2315_5fea3f94 - All measurements returned failed CRC
2020-06-22 17:32:05,086 - ERROR - mycodo.inputs.am2315_5fea3f94 - All measurements returned failed CRC
2020-06-22 17:32:11,773 - ERROR - mycodo.daemon - Could not set PID setpoint_band: 'd35d66b7-afae-4742-af53-bc16f15329d4'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 645, in pid_get
    return self.controller['PID'][pid_id].get_setpoint_band()
KeyError: 'd35d66b7-afae-4742-af53-bc16f15329d4'
2020-06-22 17:32:11,783 - ERROR - mycodo.daemon - Could not set PID setpoint_band: 'f7f18fb5-09d3-4fe9-8b8d-d11e250d7cae'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 645, in pid_get
    return self.controller['PID'][pid_id].get_setpoint_band()
KeyError: 'f7f18fb5-09d3-4fe9-8b8d-d11e250d7cae'
2020-06-22 17:32:12,322 - ERROR - mycodo.daemon - Could not set PID setpoint_band: 'd35d66b7-afae-4742-af53-bc16f15329d4'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 645, in pid_get
    return self.controller['PID'][pid_id].get_setpoint_band()
KeyError: 'd35d66b7-afae-4742-af53-bc16f15329d4'
2020-06-22 17:32:12,398 - ERROR - mycodo.daemon - Could not set PID setpoint_band: 'f7f18fb5-09d3-4fe9-8b8d-d11e250d7cae'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 645, in pid_get
    return self.controller['PID'][pid_id].get_setpoint_band()
KeyError: 'f7f18fb5-09d3-4fe9-8b8d-d11e250d7cae'
2020-06-22 17:32:13,471 - ERROR - mycodo.inputs.am2315_5fea3f94 - All measurements returned failed CRC
2020-06-22 17:32:21,958 - ERROR - mycodo.inputs.am2315_5fea3f94 - All measurements returned failed CRC
2020-06-22 17:32:30,345 - ERROR - mycodo.inputs.am2315_5fea3f94 - All measurements returned failed CRC
`

kizniche, 2020-06-23:
Have you tried software I2C discussed in the manual? https://github.com/kizniche/Mycodo/blob/master/mycodo-manual.rst#am2315

RynFlutsch, 2020-06-23:
No i will try tomorrow. It already had worked. Maybe the Update changed something.

kizniche, 2020-06-23:
What update?

RynFlutsch, 2020-06-23:
8.4.0 to 8.5.4

kizniche, 2020-06-23:
Nothing that would break functionality changed with the AM2315 module between those versions: https://github.com/kizniche/Mycodo/compare/v8.4.0...v8.5.4

closed 2020-07-23 by kizniche