kizniche/Mycodo #783: Unable to install PIGPIOD to enable DHT22 and DHT11

trezhunter, 2020-07-06:
I am trying to enable the DHT22 and DHT11, but I am getting an error:

[2020-07-06 12:41:03] Dependency installation beginning. Installing: file-exists /opt/mycodo/pigpio_installed

[2020-07-06 12:41:03] 
[2020-07-06 12:41:03] #### Installing/updating  (pigpio)
[2020-07-06 12:41:03] 
**[2020-07-06 12:41:03] Unrecognized dependency: pigpio**
[2020-07-06 12:41:03] End install of file-exists /opt/mycodo/pigpio_installed

[2020-07-06 12:41:03] 
[2020-07-06 12:41:03] #### Setting permissions

[2020-07-06 12:41:06] #### All Dependencies have been installed.

When I subsequently go to add a DHT, I get the following additional error:
**Error: Add Input: The DHT22 device you're trying to add has unmet dependencies: file-exists /opt/mycodo/pigpio_installed**

I was able to install 1.71-0~rpt1 via the console with APT "sudo apt-get install PIGPIOD", but I am not sure how to make it detect in Mycodo. I uninstalled in configuration, and it looks like the APT-based install is still present and running. when I grep pigpiod.

kizniche, 2020-07-06:
Thanks for the bug report. I'll look into it.

kizniche, 2020-07-07:
I just released v8.5.7, a bugfix release that fixes this issue. Let me know if you experience the same issue after an upgrade.

closed 2020-07-07 by kizniche

trezhunter, 2020-07-07:
Thanks for the fast update!

It looks to run the install now, but I am seeing some makefile and service errors and it still asks me to install pigpiod each time I try to add a DHT sensor with the same error "unmet dependencies: file-exists /opt/mycodo/pigpio_installed". and shows pigpiod as not installed per the dependency list.

[2020-07-07 14:16:56] #### Installing pigpiod
[2020-07-07 14:16:56] Reading package lists...
[2020-07-07 14:16:56] Building dependency tree...
[2020-07-07 14:16:56] Reading state information...
[2020-07-07 14:16:57] python3-pigpio is already the newest version (1.71-0~rpt1).
[2020-07-07 14:16:57] 0 upgraded, 0 newly installed, 0 to remove and 16 not upgraded.
[2020-07-07 14:16:57] gcc -O3 -Wall -pthread -fpic -c -o pigpio.o pigpio.c
[2020-07-07 14:16:57] gcc -O3 -Wall -pthread -fpic -c -o command.o command.c
[2020-07-07 14:16:57] gcc -O3 -Wall -pthread -fpic -c -o pigpiod_if.o pigpiod_if.c
[2020-07-07 14:16:57] gcc -O3 -Wall -pthread -fpic -c -o pigpiod_if2.o pigpiod_if2.c
[2020-07-07 14:16:58] pigpio.c: In function âmbCreateâ:
[2020-07-07 14:16:58] pigpio.c:2682:36: warning: implicit declaration of function âmakedevâ [-Wimplicit-function-declaration]
[2020-07-07 14:16:58]     return mknod(dev, S_IFCHR|0600, makedev(MB_DEV_MAJOR, 0));
[2020-07-07 14:16:58]                                     ^~~~~~~
[2020-07-07 14:16:58] gcc -O3 -Wall -pthread   -c -o x_pigpio.o x_pigpio.c
[2020-07-07 14:17:00] gcc -O3 -Wall -pthread   -c -o x_pigpiod_if.o x_pigpiod_if.c
[2020-07-07 14:17:00] gcc -O3 -Wall -pthread   -c -o x_pigpiod_if2.o x_pigpiod_if2.c
[2020-07-07 14:17:01] gcc -O3 -Wall -pthread   -c -o pig2vcd.o pig2vcd.c
[2020-07-07 14:17:01] gcc -O3 -Wall -pthread   -c -o pigpiod.o pigpiod.c
[2020-07-07 14:17:01] gcc -O3 -Wall -pthread   -c -o pigs.o pigs.c
[2020-07-07 14:17:01] gcc -shared -o libpigpiod_if.so pigpiod_if.o command.o
[2020-07-07 14:17:01] gcc -shared -o libpigpiod_if2.so pigpiod_if2.o command.o
[2020-07-07 14:17:02] strip --strip-unneeded libpigpiod_if.so
[2020-07-07 14:17:02] strip --strip-unneeded libpigpiod_if2.so
[2020-07-07 14:17:02] size     libpigpiod_if.so
[2020-07-07 14:17:02] size     libpigpiod_if2.so
[2020-07-07 14:17:02]    text	   data	    bss	    dec	    hex	filename
[2020-07-07 14:17:02]   47351	   4284	  49244	 100879	  18a0f	libpigpiod_if.so
[2020-07-07 14:17:02] gcc -o pig2vcd pig2vcd.o
[2020-07-07 14:17:02]    text	   data	    bss	    dec	    hex	filename
[2020-07-07 14:17:02]   61050	   4284	   1984	  67318	  106f6	libpigpiod_if2.so
[2020-07-07 14:17:02] gcc -o x_pigpiod_if x_pigpiod_if.o -L. -lpigpiod_if -pthread -lrt
[2020-07-07 14:17:02] gcc -o x_pigpiod_if2 x_pigpiod_if2.o -L. -lpigpiod_if2 -pthread -lrt
[2020-07-07 14:17:02] gcc -o pigs pigs.o command.o
[2020-07-07 14:17:02] strip pig2vcd
[2020-07-07 14:17:02] strip pigs
[2020-07-07 14:17:11] gcc -shared -o libpigpio.so pigpio.o command.o
[2020-07-07 14:17:11] strip --strip-unneeded libpigpio.so
[2020-07-07 14:17:11] size     libpigpio.so
[2020-07-07 14:17:11]    text	   data	    bss	    dec	    hex	filename
[2020-07-07 14:17:11]  222845	   5328	 598336	 826509	  c9c8d	libpigpio.so
[2020-07-07 14:17:11] gcc -o x_pigpio x_pigpio.o -L. -lpigpio -pthread -lrt
[2020-07-07 14:17:11] gcc -o pigpiod pigpiod.o -L. -lpigpio -pthread -lrt
[2020-07-07 14:17:11] /usr/bin/ld: ./libpigpio.so: undefined reference to `makedev'
[2020-07-07 14:17:11] /usr/bin/ld: ./libpigpio.so: undefined reference to `makedev'
[2020-07-07 14:17:11] collect2: error: ld returned 1 exit status
[2020-07-07 14:17:11] make: *** [Makefile:65: pigpiod] Error 1
[2020-07-07 14:17:11] make: *** Waiting for unfinished jobs....
[2020-07-07 14:17:11] collect2: error: ld returned 1 exit status
[2020-07-07 14:17:11] make: *** [Makefile:56: x_pigpio] Error 1
[2020-07-07 14:17:11] gcc -o x_pigpio x_pigpio.o -L. -lpigpio -pthread -lrt
[2020-07-07 14:17:11] /usr/bin/ld: ./libpigpio.so: undefined reference to `makedev'
[2020-07-07 14:17:11] collect2: error: ld returned 1 exit status
[2020-07-07 14:17:11] make: *** [Makefile:56: x_pigpio] Error 1
[2020-07-07 14:17:11] 
[2020-07-07 14:17:11] #### Disabling installed pigpiod startup script
[2020-07-07 14:17:13] Removed /etc/systemd/system/pigpiod_high.service.
[2020-07-07 14:17:13] Removed /etc/systemd/system/pigpiod.service.
[2020-07-07 14:17:13] Removed /etc/systemd/system/multi-user.target.wants/pigpiod_high.service.
[2020-07-07 14:17:13] Failed to disable unit: Unit file pigpiod_low.service does not exist.
[2020-07-07 14:17:13] Failed to disable unit: Unit file pigpiod_high.service does not exist.
[2020-07-07 14:17:13] 
[2020-07-07 14:17:13] #### Enabling pigpiod startup script (5 ms sample rate)
[2020-07-07 14:17:13] Created symlink /etc/systemd/system/pigpiod.service â /home/pi/Mycodo/install/pigpiod_high.service.
[2020-07-07 14:17:13] Created symlink /etc/systemd/system/multi-user.target.wants/pigpiod_high.service â /home/pi/Mycodo/install/pigpiod_high.service.
[2020-07-07 14:17:13] Created symlink /etc/systemd/system/pigpiod_high.service â /home/pi/Mycodo/install/pigpiod_high.service.

[2020-07-07 14:17:14] End install of file-exists /opt/mycodo/pigpio_installed

[2020-07-07 14:17:14] 
[2020-07-07 14:17:14] #### Setting permissions

[2020-07-07 14:17:14] #### All Dependencies have been installed.

kizniche, 2020-07-07:
Okay, I released v8.5.8 with a fix that's been tested this time. You should have an error-free dependency install log like this:

```
[2020-07-07 15:59:33] Dependency installation beginning. Installing: file-exists /opt/mycodo/pigpio_installed

[2020-07-07 15:59:33] 
[2020-07-07 15:59:33] #### Installing/updating pigpio (internal)
[2020-07-07 15:59:33] 
[2020-07-07 15:59:33] #### Installing pigpiod
[2020-07-07 15:59:33] Reading package lists...
[2020-07-07 15:59:34] Building dependency tree...
[2020-07-07 15:59:34] Reading state information...
[2020-07-07 15:59:36] The following packages were automatically installed and are no longer required:
[2020-07-07 15:59:36]   libdrm-etnaviv1 libllvm6.0
[2020-07-07 15:59:36] Use 'apt autoremove' to remove them.
[2020-07-07 15:59:36] The following NEW packages will be installed:
[2020-07-07 15:59:36]   python3-pigpio
[2020-07-07 15:59:38] debconf: unable to initialize frontend: Dialog
[2020-07-07 15:59:38] debconf: (TERM is not set, so the dialog frontend is not usable.)
[2020-07-07 15:59:38] debconf: falling back to frontend: Readline
[2020-07-07 15:59:38] debconf: unable to initialize frontend: Readline
[2020-07-07 15:59:38] debconf: (This frontend requires a controlling tty.)
[2020-07-07 15:59:38] debconf: falling back to frontend: Teletype
[2020-07-07 15:59:38] dpkg-preconfigure: unable to re-open stdin: 
[2020-07-07 15:59:38] 0 upgraded, 1 newly installed, 0 to remove and 41 not upgraded.
[2020-07-07 15:59:38] Need to get 0 B/35.3 kB of archives.
[2020-07-07 15:59:38] After this operation, 175 kB of additional disk space will be used.
[2020-07-07 15:59:38] Selecting previously unselected package python3-pigpio.
[2020-07-07 15:59:39] (Reading database ... 
(Reading database ... 5%
...
(Reading database ... 100%
(Reading database ... 117089 files and directories currently installed.)
[2020-07-07 15:59:39] Preparing to unpack .../python3-pigpio_1.71-0~rpt1_all.deb ...
[2020-07-07 15:59:39] Unpacking python3-pigpio (1.71-0~rpt1) ...
[2020-07-07 15:59:39] Setting up python3-pigpio (1.71-0~rpt1) ...
[2020-07-07 15:59:43] --2020-07-07 15:59:43--  https://github.com/joan2937/pigpio/archive/v77.tar.gz
[2020-07-07 15:59:43] Resolving github.com (github.com)... 140.82.113.4
[2020-07-07 15:59:43] Connecting to github.com (github.com)|140.82.113.4|:443... connected.
[2020-07-07 15:59:44] HTTP request sent, awaiting response... 302 Found
[2020-07-07 15:59:44] Location: https://codeload.github.com/joan2937/pigpio/tar.gz/v77 [following]
[2020-07-07 15:59:44] --2020-07-07 15:59:44--  https://codeload.github.com/joan2937/pigpio/tar.gz/v77
[2020-07-07 15:59:44] Resolving codeload.github.com (codeload.github.com)... 140.82.114.10
[2020-07-07 15:59:44] Connecting to codeload.github.com (codeload.github.com)|140.82.114.10|:443... connected.
[2020-07-07 15:59:44] HTTP request sent, awaiting response... 200 OK
[2020-07-07 15:59:44] Length: unspecified [application/x-gzip]
[2020-07-07 15:59:44] Saving to: âpigpio.tar.gzâ
[2020-07-07 15:59:44] 
[2020-07-07 15:59:44]      0K .......... .......... .......... .......... ..........  799K
...
[2020-07-07 15:59:45]   2800K .......... .......... .......... ....                  5.77M=1.1s
[2020-07-07 15:59:45] 
[2020-07-07 15:59:45] 2020-07-07 15:59:45 (2.55 MB/s) - âpigpio.tar.gzâ saved [2902866]
[2020-07-07 15:59:45] 
[2020-07-07 15:59:45] gcc -O3 -Wall -pthread -fpic -c -o pigpio.o pigpio.c
[2020-07-07 15:59:46] gcc -O3 -Wall -pthread -fpic -c -o command.o command.c
[2020-07-07 15:59:46] gcc -O3 -Wall -pthread -fpic -c -o pigpiod_if.o pigpiod_if.c
[2020-07-07 15:59:46] gcc -O3 -Wall -pthread -fpic -c -o pigpiod_if2.o pigpiod_if2.c
[2020-07-07 15:59:53] gcc -O3 -Wall -pthread   -c -o x_pigpio.o x_pigpio.c
[2020-07-07 15:59:56] gcc -O3 -Wall -pthread   -c -o x_pigpiod_if.o x_pigpiod_if.c
[2020-07-07 15:59:56] gcc -O3 -Wall -pthread   -c -o x_pigpiod_if2.o x_pigpiod_if2.c
[2020-07-07 15:59:58] gcc -O3 -Wall -pthread   -c -o pig2vcd.o pig2vcd.c
[2020-07-07 15:59:59] gcc -O3 -Wall -pthread   -c -o pigpiod.o pigpiod.c
[2020-07-07 15:59:59] gcc -O3 -Wall -pthread   -c -o pigs.o pigs.c
[2020-07-07 15:59:59] gcc -shared -pthread -Wl,-soname,libpigpiod_if.so.1 -o libpigpiod_if.so.1 pigpiod_if.o command.o
[2020-07-07 15:59:59] ln -fs libpigpiod_if.so.1 libpigpiod_if.so
[2020-07-07 15:59:59] strip --strip-unneeded libpigpiod_if.so
[2020-07-07 15:59:59] size     libpigpiod_if.so
[2020-07-07 15:59:59]    text	   data	    bss	    dec	    hex	filename
[2020-07-07 15:59:59]   47778	   5032	  49248	 102058	  18eaa	libpigpiod_if.so
[... 62 lines cut ...]
[2020-07-07 16:00:24] install -m 0755 -d                             /usr/local/man/man1
[2020-07-07 16:00:24] install -m 0644 p*.1                           /usr/local/man/man1
[2020-07-07 16:00:24] install -m 0755 -d                             /usr/local/man/man3
[2020-07-07 16:00:24] install -m 0644 p*.3                           /usr/local/man/man3
[2020-07-07 16:00:24] ldconfig
[2020-07-07 16:00:24] 
[2020-07-07 16:00:24] #### Disabling installed pigpiod startup script
[2020-07-07 16:00:28] Removed /etc/systemd/system/pigpiod.service.
[2020-07-07 16:00:28] Removed /etc/systemd/system/pigpiod_high.service.
[2020-07-07 16:00:28] Removed /etc/systemd/system/multi-user.target.wants/pigpiod_high.service.
[2020-07-07 16:00:29] Failed to disable unit: Unit file pigpiod_low.service does not exist.
[2020-07-07 16:00:29] Failed to disable unit: Unit file pigpiod_high.service does not exist.
[... cut ...]
[2020-07-07 16:00:29] Created symlink /etc/systemd/system/multi-user.target.wants/pigpiod_high.service â /home/pi/Mycodo/install/pigpiod_high.service.
[2020-07-07 16:00:29] Created symlink /etc/systemd/system/pigpiod_high.service â /home/pi/Mycodo/install/pigpiod_high.service.

[2020-07-07 16:00:30] End install of file-exists /opt/mycodo/pigpio_installed

[2020-07-07 16:00:30] 
[2020-07-07 16:00:30] #### Setting permissions

[2020-07-07 16:00:37] #### All Dependencies have been installed.
```

trezhunter, 2020-07-08:
Excellent! It installed now. Thank you 👍 
I am having another issue getting no data on any sensors, will open a separate issue regarding this.