kizniche/Mycodo #784: unable to connect to web server

Mondster, 2020-07-07:
#### Connecting to http://localhost (creates Mycodo database if it doesn't exist)
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 5 times)...

kizniche, 2020-07-07:
Can you provide the rest of the log? It attempts 5 times because the first attempt is often not long enough to allow the server to start, especially on any Pi prior to version 4.

Mondster, 2020-07-08:
sorry my bad, i didnt saw the part using the lite version. i reflash the os to lite and its working now.

Mondster, 2020-07-08:
beside this i got question on the probes, if i just follow this site and get the probes listed, ec, ph, temp i should be able to get the device working without any issue correct? 

https://kylegabriel.com/projects/2020/06/automated-hydroponic-system-build.html?unapproved=652&moderation-hash=29d9dfaad4c6806d4044883e03bf6b9d#comment-652

kizniche, 2020-07-08:
I still need to download the desktop version and see what exactly is different that might be causing install problems. Supposedly others have successfully installed on the desktop version, so there's conflicting information. Nearly all of my PIs are headless. The ones I do have a touch display for I install the lite version and then install X and a window manager so I know I have the most minimal ampunt of space used for the GUI.

kizniche, 2020-07-08:
The sensors I list there I'm currently using with Mycodo, so they're guaranteed to work. You could likely get most sensors to work, but you may have to put some time in figuring out the hardware interfacing and setting up software to read it.

Mondster, 2020-07-08:
I tested the desktop version like 5 times, be it full or the vanilla it wasnt able to work. hence i re-read the steps and found a line you tested on the lite version. hence i reflash and it works. like you mention if i need the gui i just need to install x and get the gui up.

Mondster, 2020-07-08:
> 
> 
> The sensors I list there I'm currently using with Mycodo, so they're guaranteed to work. You could likely get most sensors to work, but you may have to put some time in figuring out the hardware interfacing and setting up software to read it.

i think i just follow your setup becuase i not very pro in hardware on raspi. just that these altas device is a little pricey. :)

however i have a semi automated solution making use of my smart home fibaro setup to trigger chiller base on the temp. I pipe these data to my splunk collector and perform a api call when the parameter i set meet.

Mondster, 2020-07-08:
[image]

Mondster, 2020-07-08:
[image]

Mondster, 2020-07-08:
hopefully i am able to get your device up and working so i can perform more integration.

closed 2020-07-08 by kizniche