kizniche/Mycodo #785: No data : Every sensor returns error : 'NoneType' object has no attribute 'next'

trezhunter, 2020-07-08:
Any sensors I add return error: Error while attempting to read input: 'NoneType' object has no attribute 'next'

I can activate sensors and but I get no data for anything but the Pi GPU temperature.
I tried this with 2 DS18b20, 1 Atlas PH UART, 1 Atlas EC UART, DHT11, and DHT22.
All have worked in 8.5.6 (however the data would show 0 on reboots and manual restart of data device was requried) and much older build 2 years ago.
I feel like I am missing something obvious..

[image]

 Mycodo Version: 8.5.8
 Raspberry Pi Version: 4B
 Raspbian OS Version: Buster Lite 

I am able to pull values and send the find command for the Atlas sensors via UART on minicom -b 9600 /dev/ttyAMA1 and ttyAMA0.

Daemon logs below:

**DHT11**
2020-07-08 01:37:22,744 - INFO - mycodo.controllers.controller_input_2fe2f0ae - Activated in 155.8 ms
2020-07-08 01:37:24,102 - ERROR - mycodo.controllers.controller_input_343acaf0 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 439, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-07-08 01:37:27,185 - ERROR - mycodo.inputs.dht11_2fe2f0ae - Could not acquire a measurement
2020-07-08 01:37:30,612 - ERROR - mycodo.controllers.controller_input_01b176e8 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 439, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-07-08 01:37:30,912 - ERROR - mycodo.controllers.controller_input_77ebe8bb - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 439, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'

**DHT22**
2020-07-08 01:35:31,981 - INFO - mycodo.controllers.controller_input_b8e204a5 - Activated in 153.1 ms
2020-07-08 01:35:39,117 - ERROR - mycodo.controllers.controller_input_343acaf0 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 439, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-07-08 01:35:45,629 - ERROR - mycodo.controllers.controller_input_01b176e8 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 439, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-07-08 01:35:45,930 - ERROR - mycodo.controllers.controller_input_77ebe8bb - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 439, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next''

**DS18b20**
2020-07-08 01:11:00,605 - ERROR - mycodo.controllers.controller_input_01b176e8 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 439, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-07-08 01:11:00,978 - ERROR - mycodo.controllers.controller_input_77ebe8bb - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 439, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-07-08 01:11:15,627 - ERROR - mycodo.controllers.controller_input_01b176e8 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 439, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-07-08 01:11:15,900 - ERROR - mycodo.controllers.controller_input_77ebe8bb - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 439, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'

**Atlas EC**
2020-07-08 01:20:40,931 - INFO - mycodo.controllers.controller_input_343acaf0 - Activated in 1922.1 ms
2020-07-08 01:20:40,932 - ERROR - mycodo.controllers.controller_input_343acaf0 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 439, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-07-08 01:20:45,696 - ERROR - mycodo.controllers.controller_input_01b176e8 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 439, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-07-08 01:20:45,980 - ERROR - mycodo.controllers.controller_input_77ebe8bb - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 439, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'
2020-07-08 01:20:54,152 - ERROR - mycodo.controllers.controller_input_343acaf0 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 439, in update_measure
    measurements = self.measure_input.next()
AttributeError: 'NoneType' object has no attribute 'next'

**Atlas PH**
2020-07-08 01:23:09,159 - ERROR - mycodo.controllers.controller_input_343acaf0 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 439, in update_measure
    measurements = self.measure_input.next()

kizniche, 2020-07-08:
The relevant part of the log has been excluded. Please provide a log excerpt from the first log line of the daemon starting.

kizniche, 2020-07-08:
There's obviously something very wrong to cause all your to have issues. It's likely some corruption of your python virtualenv or something similar. I'll need the full daemon log from the start of the daemon to properly diagnose. The error excerpts you provided are of no use, as the initial error that would cause them would occur prior to seeing those. Therefore, to meaningfully investigate the issue, I'll need more information.

trezhunter, 2020-07-08:
Sorry about the delay. Busy day at the home office.. 

_Here are my logs from the restart of the backend:_

2020-07-08 14:41:37,664 - INFO - mycodo.daemon - Mycodo daemon v8.5.8 starting
2020-07-08 14:41:37,665 - DEBUG - mycodo.daemon - Refreshing camera settings
2020-07-08 14:41:37,688 - DEBUG - mycodo.daemon - Refreshing misc settings
2020-07-08 14:41:37,702 - DEBUG - mycodo.daemon - Anonymous statistics enabled
2020-07-08 14:41:37,719 - INFO - mycodo.pyro_daemon - Starting Pyro5 daemon
2020-07-08 14:41:37,721 - INFO - mycodo.pyro_monitor - Starting Pyro5 daemon monitor (30 min timer)
2020-07-08 14:41:37,839 - DEBUG - mycodo.daemon - Starting Output Controller
2020-07-08 14:41:37,869 - DEBUG - mycodo.controllers.controller_output - Initializing Outputs
2020-07-08 14:41:37,920 - INFO - mycodo.outputs.wired_e3352217 - Output setup on pin 23 and turned OFF (OFF=LOW)
2020-07-08 14:41:37,920 - INFO - mycodo.outputs.wired_e3352217 - Initialized in 4.2 ms
2020-07-08 14:41:37,921 - DEBUG - mycodo.controllers.controller_output - e3352217 (relay 1) Initialized
2020-07-08 14:41:37,922 - INFO - mycodo.outputs.wired_4bca794a - Output setup on pin 24 and turned OFF (OFF=LOW)
2020-07-08 14:41:37,922 - INFO - mycodo.outputs.wired_4bca794a - Initialized in 0.5 ms
2020-07-08 14:41:37,922 - DEBUG - mycodo.controllers.controller_output - 4bca794a (relay 2) Initialized
2020-07-08 14:41:37,923 - INFO - mycodo.outputs.wired_edcb66f5 - Output setup on pin 25 and turned OFF (OFF=LOW)
2020-07-08 14:41:37,923 - INFO - mycodo.outputs.wired_edcb66f5 - Initialized in 0.5 ms
2020-07-08 14:41:37,923 - DEBUG - mycodo.controllers.controller_output - edcb66f5 (relay 4) Initialized
2020-07-08 14:41:37,923 - INFO - mycodo.controllers.controller_output - Setting Output e3352217 startup state to OFF
2020-07-08 14:41:37,924 - DEBUG - mycodo.controllers.controller_output - output_on_off(e3352217-12af-4e7b-9e20-bfeae3f36ef9, off, 0.0, 0.0, 0.0, False)
2020-07-08 14:41:37,924 - INFO - mycodo.controllers.controller_output - Setting Output 4bca794a startup state to OFF
2020-07-08 14:41:37,924 - DEBUG - mycodo.controllers.controller_output - output_on_off(4bca794a-4dfe-41f7-8ae8-6a712a72639f, off, 0.0, 0.0, 0.0, False)
2020-07-08 14:41:37,924 - INFO - mycodo.controllers.controller_output - Setting Output edcb66f5 startup state to OFF
2020-07-08 14:41:37,924 - DEBUG - mycodo.controllers.controller_output - output_on_off(edcb66f5-ef1b-4898-9005-1edf424eb8cd, off, 0.0, 0.0, 0.0, False)
2020-07-08 14:41:37,924 - DEBUG - mycodo.controllers.controller_output - Outputs Initialized
2020-07-08 14:41:37,924 - INFO - mycodo.controllers.controller_output - Activated in 85.0 ms
2020-07-08 14:41:38,425 - DEBUG - mycodo.daemon - Output Controller fully started
2020-07-08 14:41:38,426 - DEBUG - mycodo.daemon - Starting all activated Conditional controllers
2020-07-08 14:41:38,426 - INFO - mycodo.daemon - All activated Conditional controllers started
2020-07-08 14:41:38,427 - DEBUG - mycodo.daemon - Starting all activated Trigger controllers
2020-07-08 14:41:38,427 - INFO - mycodo.daemon - All activated Trigger controllers started
2020-07-08 14:41:38,427 - DEBUG - mycodo.daemon - Starting all activated Input controllers
2020-07-08 14:41:38,779 - DEBUG - mycodo.pyro_monitor - Pyro5 daemon monitor: daemon_status() response: 'alive'
2020-07-08 14:41:38,835 - INFO - mycodo.controllers.controller_input_b8e204a5 - Activated in 236.5 ms
2020-07-08 14:42:19,308 - ERROR - mycodo.controllers.controller_input_b8e204a5 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-07-08 14:42:44,671 - ERROR - mycodo.inputs.dht22_b8e204a5 - InputModule raised an exception when taking a reading: unpack requires a buffer of 16 bytes
2020-07-08 14:42:46,439 - INFO - mycodo.daemon - Received command to terminate daemon
2020-07-08 14:42:53,784 - ERROR - mycodo.inputs.dht22_b8e204a5 - Could not initialize sensor. Check if it's connected properly and pigpiod is running. Error: [Errno 32] Broken pipe
2020-07-08 14:42:55,787 - ERROR - mycodo.inputs.dht22_b8e204a5 - Could not initialize sensor. Check if it's connected properly and pigpiod is running. Error: [Errno 32] Broken pipe
2020-07-08 14:42:57,789 - ERROR - mycodo.inputs.dht22_b8e204a5 - Could not initialize sensor. Check if it's connected properly and pigpiod is running. Error: [Errno 32] Broken pipe
2020-07-08 14:42:59,791 - ERROR - mycodo.inputs.dht22_b8e204a5 - Could not initialize sensor. Check if it's connected properly and pigpiod is running. Error: [Errno 32] Broken pipe
2020-07-08 14:43:01,794 - ERROR - mycodo.controllers.controller_input_b8e204a5 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-07-08 14:43:08,803 - ERROR - mycodo.inputs.dht22_b8e204a5 - Could not initialize sensor. Check if it's connected properly and pigpiod is running. Error: [Errno 32] Broken pipe
2020-07-08 14:43:10,805 - ERROR - mycodo.inputs.dht22_b8e204a5 - Could not initialize sensor. Check if it's connected properly and pigpiod is running. Error: [Errno 32] Broken pipe
2020-07-08 14:43:12,808 - ERROR - mycodo.inputs.dht22_b8e204a5 - Could not initialize sensor. Check if it's connected properly and pigpiod is running. Error: [Errno 32] Broken pipe

_This is the log from when I booted up cold:_
2020-07-08 01:55:27,140 - INFO - mycodo.daemon - Mycodo daemon v8.5.8 starting
[... cut ...]
2020-07-08 14:17:26,184 - ERROR - mycodo.controllers.controller_input_b8e204a5 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-07-08 14:18:11,262 - ERROR - mycodo.controllers.controller_input_b8e204a5 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-07-08 14:18:15,279 - INFO - mycodo.daemon - Received command to terminate daemon

The logs seem to populate much slower on the backend restart, the cold boot logs displayed all the errors almost immediately.
Relay controls seem to work ok, at least.

kizniche, 2020-07-08:
I don't see any of the previous errors you posted. After these particular startups, do those errors surface or do they no longer appear?

kizniche, 2020-07-08:
Before we get into Mycodo, can you verify your DS18B20 sensors are detected by the operating system. Also, are your Atlas Scientific sensors in UART or I2C mode? If in I2C mode, verify they're detected by the operating system.

kizniche, 2020-07-08:
I read again your Atlas Scientific sensors are connected via UART. How specifically are these connected to your Pi?

trezhunter, 2020-07-08:
These errors are persisting right now:
2020-07-08 15:44:19,173 - ERROR - mycodo.controllers.controller_input_b8e204a5 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-07-08 15:45:04,152 - ERROR - mycodo.controllers.controller_input_b8e204a5 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-07-08 15:45:49,134 - ERROR - mycodo.controllers.controller_input_b8e204a5 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-07-08 15:46:34,106 - ERROR - mycodo.controllers.controller_input_b8e204a5 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-07-08 15:47:19,180 - ERROR - mycodo.controllers.controller_input_b8e204a5 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-07-08 15:48:04,158 - ERROR - mycodo.controllers.controller_input_b8e204a5 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-07-08 15:48:49,137 - ERROR - mycodo.controllers.controller_input_b8e204a5 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-07-08 15:49:34,108 - ERROR - mycodo.controllers.controller_input_b8e204a5 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-07-08 15:50:19,176 - ERROR - mycodo.controllers.controller_input_b8e204a5 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-07-08 15:51:04,159 - ERROR - mycodo.controllers.controller_input_b8e204a5 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-07-08 15:51:49,138 - ERROR - mycodo.controllers.controller_input_b8e204a5 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-07-08 15:52:34,113 - ERROR - mycodo.controllers.controller_input_b8e204a5 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-07-08 15:53:19,194 - ERROR - mycodo.controllers.controller_input_b8e204a5 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-07-08 15:54:04,170 - ERROR - mycodo.controllers.controller_input_b8e204a5 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-07-08 15:54:49,147 - ERROR - mycodo.controllers.controller_input_b8e204a5 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-07-08 15:55:34,127 - ERROR - mycodo.controllers.controller_input_b8e204a5 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-07-08 15:56:19,103 - ERROR - mycodo.controllers.controller_input_b8e204a5 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-07-08 15:57:04,183 - ERROR - mycodo.controllers.controller_input_b8e204a5 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-07-08 15:57:49,162 - ERROR - mycodo.controllers.controller_input_b8e204a5 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-07-08 15:58:34,138 - ERROR - mycodo.controllers.controller_input_b8e204a5 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-07-08 15:59:19,116 - ERROR - mycodo.controllers.controller_input_b8e204a5 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-07-08 16:00:04,190 - ERROR - mycodo.controllers.controller_input_b8e204a5 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2020-07-08 16:00:49,151 - ERROR - mycodo.controllers.controller_input_b8e204a5 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.

I only get the original errors when I try to de-activate/activate:
2020-07-08 16:04:09,209 - WARNING - mycodo.daemon - Input controller with ID 2fe2f0ae-30e8-4929-bd95-7a604f301d44 not found
2020-07-08 16:04:14,662 - INFO - mycodo.controllers.controller_input_2fe2f0ae - Activated in 150.5 ms
2020-07-08 16:04:19,103 - ERROR - mycodo.inputs.dht11_2fe2f0ae - Could not acquire a measurement

Hmm, for some reason the dropdown for the 1-wire DS18b20 (GPIO pin4, dtoverlay=w1-gpio) is "None Detected". There were addresses when I set them before.. odd. The LS on /sys/bus/w1/devices only shows w1_bus_master1. Doesn't look like they are working any more. I will spin up another pi and breadboard them in to check them.

Atlas sensors are in UART mode (previously UART1 and UART5, moved to UART2 and UART5) confirmed with minicom -b 9600 -D /dev/ttyAMA1 (PH-storage solution: 8.274,. find=*OK) and ttyAMA2 (EC-dry: 0.00, find=*OK)

kizniche, 2020-07-08:
> I read again your Atlas Scientific sensors are connected via UART. How specifically are these connected to your Pi?

Connected to the Pi, as in hardware.

trezhunter, 2020-07-08:
I have the stamps mounted on the Atlas electrically-isolated EZO boards and connected to the pi from there.
**PH**
TX > Pin 28 RXD2
RX > Pin 27 TXD2
VCC > Pin 17 3.3v
GND > Pin20 Ground

**EC**
TX > Pin 33 RXD5
RX > Pin 32 TXD5
VCC > VCC 3.3v thru-pin on relay board
GND > Pin 34 Ground

[image]
[image]

kizniche, 2020-07-09:
I just tried to get UART working on my Pi 4 with the Atlas EC sensor without luck. I couldn't even get it working with their UART code at https://github.com/AtlasScientific/Raspberry-Pi-sample-code Have you been able to get it working with their code?

I'd recommend using I2C, as this is what I've been using for the past several months with the Atlas Scientific flow, EC, pH, and 4 peristaltic pumps, without issue.

kizniche, 2020-07-09:
Could you please provide a detailed step-by-step protocol for how you were able to successfully get commands to and data from your sensor using UART? Once I have this working, I can begin to start diagnosing the Mycodo code.

trezhunter, 2020-07-09:
I was considering switching to I2c mode, but the i22c mode requires continuous requests, where UART (default) just polls regularly, seemed simpler on the surface.

I'm going to take a look at that sample code out on my other pi, it looks like  a large chunk of that configuration can be done via the raspi-config (disabling serial console).

Green blinking is standby mode, Blue blinking is reading. In default mode, the EZO stamps will continuously poll at a 1 second interval, so they will just blink forever. Believe in I2c mode, it will idle blue.

Here is what I did to enable them and test:
_NOTE: I use minicom from an SSH session, since sometimes it has issue bringing up the menu to exit out on my local console. Additionally, when I first updated the firmware on Pi4, I had to hit enter after starting the minicom session to start seeing values the first time, I'm not sure why to be honest, but it was only the first time i ran it and never again_

sudo raspi-config
interfaces
disable serial console

sudo nano /boot/config.txt
enable_uart=1
dtoverlay=UART1
dtoverlay=UART2
dtoverlay=UART5

_(NOTE: the first time I tried the above config.txt settings I did dtoverlay=UART1 , dtoverlay-UART2, and dtoverlay=disable-bt, since BT maps to UART0 and I was able to use TTYAMA0 and get data, but after I reimaged, after troubleshooting the DHT prior, and reloaded Mycodo and set things up fresh, it doesn't work. Not sure if a dependency grabbed those pins perhaps, so I added one more dtoverlay , moved to UART2 which maps to TTYAMA1, while UART 5 now maps to TTYAMA2 as shown above)_

sudo reboot

sudo apt-get install minicom

ls -l /dev/ttyAMA*
_(to confirm UART overlays are active as /dev/TTYAMA1 and /dev/TTYAMA2)_

minicom -b 9600 -D /dev/ttyAMA1
_confirm reading-value, will automatically update each second_
find _<hit enter>_
_sensor transmits *OK_
r _<hit enter>_
_sensor manually reads current value *OK, followed by value_
ctrl-A then X to exit minicom

minicom -b 9600 -D /dev/ttyAMA2
_confirm reading-value, will automatically update each second_
find _<hit enter>_
_sensor transmits *OK_
r _<hit enter>_
_sensor manually reads current value *OK, followed by value_
ctrl-A then X to exit minicom

trezhunter, 2020-07-09:
New Pi firmware seems buggy.. rolled it back. Don't recommend it.

kizniche, 2020-07-09:
Well, that's basically what I did and it wasn't working last night. However, after enabling UART4 again today it's now working with minicom. So, now that I've confirmed it's working, I can start diagnosing the Mycodo code. I'll report back with my findings soon.

kizniche, 2020-07-09:
I can confirm the Atlas Scientific example code does not work. Since the Mycodo code is heavily based on it, when I fix one I'll likely fix the other. Looks like I'll soon have a pull request or issue for Atlas Scientific.

kizniche, 2020-07-09:
I'm successfully retrieving data with the AS code, however it's not properly interpreting the stop character and therefore continues infinitely in the background. Should be a fairly easy fix.

kizniche, 2020-07-09:
I think I fixed the issue. It was an error in the Mycodo code, not the AS code. I'll provide something you can test in a moment once I've tested it.

kizniche, 2020-07-10:
Well, it was a lot more involved to fix the issue than I previously thought. I have it working and successfully storing measurements for Mycodo, however I had to edit 3 different files, and will need to edit every Atlas Scientific Input module. I still need to plug in my very old version temperature EZO board to make sure it didn't break the ability to read older boards.

kizniche, 2020-07-10:
I just confirmed the issue comes down to the Atlas Scientific code not being python 3 compatible, even though the [last commit for uart.py](https://github.com/AtlasScientific/Raspberry-Pi-sample-code/commit/0ff1a4f228ce17ec2bf94f78dc0258f7a1a3bc94) had the commit message "Updated sample code for both python 2 and 3 compatibility". This was from 2018, so there's likely some changes to the serial library since then that might be the cause of the issue.

I think I found the simplest solution, and just pushed a fix for only the Electrical Conductivity Input (for now). If you'd like to test the changes prior to release, you'll need to [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch).

trezhunter, 2020-07-10:
Awesome! I will set the upgrade to master option up and test it out shortly.

FYI, I also fixed the DHT22 and the DS18b20 issue. Seems to have been related to DC power. 
I switched my project board over to dedicated PSU, instead being powered by the Pi 5v, but that was not returning the right levels to the GPIO thru my voltage dividers at all when I tested with the voltmeter. Way too much power back into GPIO (5v returned instead of 3.3v) Now the numbers are up.

[image]

Will do the upgrade now..

trezhunter, 2020-07-10:
Looks like the fix was effective.
EC data is now happily populating!

[image]

kizniche, 2020-07-10:
I'm going to test my older board and verify the fix still works with it, and if successful, I'll modify the remaining AS Input modules with the fix. Also, I would check out https://github.com/cpetrich/counterfeit_DS18B20 to test your DS18B20s for authenticity.

trezhunter, 2020-07-10:
I can't seem to find any of my Arduinos off hand, but at a glance it doesn't look like they pass the address test:

28-xx-xx-xx-xx-00-00-xx
mine are:
28-00-00-06-08-39-41
28-00-00-06-08-6d-21
28-00-00-06-08-c7-3c

The two 00 octets are transposed to the 2nd and 3rd Octet.

Have an Arduino being overnighted, may just get some new DS18B20s for good measure.

kizniche, 2020-07-11:
Just a quick update... I've made considerable progress with improving the Atlas Scientific code, for all devices (sensors and pump). I've successfully tested the changes for I2C and UART with both old and new EZO board versions. Unfortunately I wasn't able to get UART working with the old board version, but this shouldn't be too much of an issue as I2C still works and there aren't many old board versions in use as it is. I'm going to try to get FTDI tested this weekend, and if all goes well, I'll push all the changes. You can then upgrade to master and have all your Atlas devices work with UART.

kizniche, 2020-07-12:
I just pushed all the Atlas Scientific changes. If you upgrade to master to test, let me know if you experience any issues.

trezhunter, 2020-07-12:
Looks like it didn't work, back to No Data:
[image]

LOG:
```
2020-07-12 14:25:19,837 - INFO - mycodo.daemon - Mycodo daemon v8.5.8 starting
2020-07-12 14:25:19,839 - DEBUG - mycodo.daemon - Refreshing camera settings
2020-07-12 14:25:19,865 - DEBUG - mycodo.daemon - Refreshing misc settings
2020-07-12 14:25:19,880 - DEBUG - mycodo.daemon - Anonymous statistics enabled
2020-07-12 14:25:19,899 - INFO - mycodo.pyro_daemon - Starting Pyro5 daemon
2020-07-12 14:25:19,902 - INFO - mycodo.pyro_monitor - Starting Pyro5 daemon monitor (30 min timer)
2020-07-12 14:25:20,100 - DEBUG - mycodo.daemon - Starting Output Controller
2020-07-12 14:25:20,169 - DEBUG - mycodo.controllers.controller_output - Initializing Outputs
2020-07-12 14:25:20,293 - INFO - mycodo.outputs.wired_e3352217 - Output setup on pin 23 and turned OFF (OFF=LOW)
2020-07-12 14:25:20,294 - INFO - mycodo.outputs.wired_e3352217 - Initialized in 14.5 ms
2020-07-12 14:25:20,294 - DEBUG - mycodo.controllers.controller_output - e3352217 (pH Down) Initialized
2020-07-12 14:25:20,297 - INFO - mycodo.outputs.wired_4bca794a - Output setup on pin 24 and turned OFF (OFF=LOW)
2020-07-12 14:25:20,297 - INFO - mycodo.outputs.wired_4bca794a - Initialized in 0.8 ms
2020-07-12 14:25:20,297 - DEBUG - mycodo.controllers.controller_output - 4bca794a (pH Up) Initialized
2020-07-12 14:25:20,300 - INFO - mycodo.outputs.wired_edcb66f5 - Output setup on pin 25 and turned OFF (OFF=LOW)
2020-07-12 14:25:20,300 - INFO - mycodo.outputs.wired_edcb66f5 - Initialized in 0.9 ms
2020-07-12 14:25:20,300 - DEBUG - mycodo.controllers.controller_output - edcb66f5 (Solution) Initialized
2020-07-12 14:25:20,301 - INFO - mycodo.controllers.controller_output - Setting Output e3352217 startup state to ON
2020-07-12 14:25:20,301 - DEBUG - mycodo.controllers.controller_output - output_on_off(e3352217-12af-4e7b-9e20-bfeae3f36ef9, on, 0.0, 0.0, 0.0, True)
2020-07-12 14:25:20,373 - DEBUG - mycodo.controllers.controller_output - Output e3352217-12af-4e7b-9e20-bfeae3f36ef9 (pH Down) ON at 2020-07-12 14:25:20.301770.
2020-07-12 14:25:20,447 - INFO - mycodo.controllers.controller_output - Setting Output 4bca794a startup state to ON
2020-07-12 14:25:20,448 - DEBUG - mycodo.controllers.controller_output - output_on_off(4bca794a-4dfe-41f7-8ae8-6a712a72639f, on, 0.0, 0.0, 0.0, True)
2020-07-12 14:25:20,517 - DEBUG - mycodo.controllers.controller_output - Output 4bca794a-4dfe-41f7-8ae8-6a712a72639f (pH Up) ON at 2020-07-12 14:25:20.448387.
2020-07-12 14:25:20,589 - INFO - mycodo.controllers.controller_output - Setting Output edcb66f5 startup state to ON
2020-07-12 14:25:20,589 - DEBUG - mycodo.controllers.controller_output - output_on_off(edcb66f5-ef1b-4898-9005-1edf424eb8cd, on, 0.0, 0.0, 0.0, True)
2020-07-12 14:25:20,659 - DEBUG - mycodo.controllers.controller_output - Output edcb66f5-ef1b-4898-9005-1edf424eb8cd (Solution) ON at 2020-07-12 14:25:20.589867.
2020-07-12 14:25:20,733 - DEBUG - mycodo.controllers.controller_output - Outputs Initialized
2020-07-12 14:25:20,733 - INFO - mycodo.controllers.controller_output - Activated in 632.1 ms
2020-07-12 14:25:20,925 - DEBUG - mycodo.pyro_monitor - Pyro5 daemon monitor: daemon_status() response: 'alive'
2020-07-12 14:25:21,234 - DEBUG - mycodo.daemon - Output Controller fully started
2020-07-12 14:25:21,235 - DEBUG - mycodo.daemon - Starting all activated Conditional controllers
2020-07-12 14:25:21,235 - INFO - mycodo.daemon - All activated Conditional controllers started
2020-07-12 14:25:21,236 - DEBUG - mycodo.daemon - Starting all activated Trigger controllers
2020-07-12 14:25:21,236 - INFO - mycodo.daemon - All activated Trigger controllers started
2020-07-12 14:25:21,236 - DEBUG - mycodo.daemon - Starting all activated Input controllers
2020-07-12 14:25:22,154 - INFO - mycodo.controllers.controller_input_b8e204a5 - Activated in 403.5 ms
2020-07-12 14:25:22,830 - INFO - mycodo.controllers.controller_input_c013febf - Activated in 384.8 ms
2020-07-12 14:25:23,529 - DEBUG - mycodo.controllers.controller_input_c013febf - Adding measurements to InfluxDB with ID c013febf-1588-4ea5-9ac4-76bac24f6eb3: {0: {'measurement': None, 'unit': 'F', 'value': 76.82, 'timestamp_utc': None}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 57.1, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 15.821499924437587, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 1350.6032023176695, 'timestamp_utc': None}}
2020-07-12 14:25:29,654 - INFO - mycodo.devices.atlas_scientific_uart_dev_ttyAMA2 - Atlas Scientific Board: EC, Rev: 2, Firmware: 2.14
2020-07-12 14:25:35,998 - INFO - mycodo.devices.atlas_scientific_uart_dev_ttyAMA2 - Atlas Scientific Board: EC, Rev: 2, Firmware: 2.14
2020-07-12 14:25:35,998 - ERROR - mycodo.controllers.controller_input_343acaf0 - initialize_variables() Exception: 'NoneType' object has no attribute 'setup'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/base_controller.py", line 72, in run
    self.initialize_variables()
  File "/var/mycodo-root/mycodo/controllers/controller_input.py", line 350, in initialize_variables
    self.measure_input = input_loaded.InputModule(self.input_dev)
  File "/home/pi/Mycodo/mycodo/inputs/atlas_ec.py", line 123, in __init__
    self.get_measurement()
  File "/home/pi/Mycodo/mycodo/inputs/atlas_ec.py", line 143, in get_measurement
    if not self.atlas_device.setup:
[... cut ...] }, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 57.7, 'timestamp_utc': None}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 15.798295957530911, 'timestamp_utc': None}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 1315.9196014863994, 'timestamp_utc': None}}
2020-07-12 14:31:08,372 - ERROR - mycodo.controllers.controller_input_343acaf0 - Error while attempting to read input: 'NoneType' object has no attribute 'next'
Traceback (most recent call last):
```

kizniche, 2020-07-12:
Whoops. sorry about that. A slight oversight in variable names. I'll push another commit shortly.

kizniche, 2020-07-12:
Though I just checked and the pH Input should have worked. The variable issue was only present in the other Inputs.

trezhunter, 2020-07-12:
It managed to change the pH sensor from the regular polling (blinking green/blue) over to solid green, Ready state. But no data.

kizniche, 2020-07-12:
I found all the issues. About to push the code changes.

trezhunter, 2020-07-12:
sounds good. Standing by

kizniche, 2020-07-12:
All ready to test

trezhunter, 2020-07-12:
Working!:
[image]

EC didn't come up at first, but after a force Acquire, seems to work just fie.

trezhunter, 2020-07-12:
Thanks a lot for your help. Really appreciate your dedication. Hopefully everything works on the older rev Pis and you can push out the next release without worry.

kizniche, 2020-07-14:
Glad we got it working. I've been meaning to overhaul the Atlas Scientific code and test all the interfacing options.

closed 2020-07-23 by kizniche

snickers2k, 2022-05-10:
i got the exact same problem, trying to use the atlas sensors with uart

```
2022-05-11 01:11:45,427 - ERROR - mycodo.inputs.atlas_ph_52990ea8 - Error 101: Device not set up. See https://kizniche.github.io/Mycodo/Error-Codes#error-101 for more info.
2022-05-11 01:12:00,351 - ERROR - mycodo.inputs.atlas_ph_52990ea8 - Error 101: Device not set up. See https://kizniche.github.io/Mycodo/Error-Codes#error-101 for more info.
2022-05-11 01:12:15,377 - ERROR - mycodo.inputs.atlas_ph_52990ea8 - Error 101: Device not set up. See https://kizniche.github.io/Mycodo/Error-Codes#error-101 for more info.
2022-05-11 01:12:15,378 - ERROR - mycodo.controllers.controller_input_52990ea8 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2022-05-11 01:12:30,403 - ERROR - mycodo.inputs.atlas_ph_52990ea8 - Error 101: Device not set up. See https://kizniche.github.io/Mycodo/Error-Codes#error-101 for more info.
2022-05-11 01:12:45,430 - ERROR - mycodo.inputs.atlas_ph_52990ea8 - Error 101: Device not set up. See https://kizniche.github.io/Mycodo/Error-Codes#error-101 for more info.
2022-05-11 01:13:00,357 - ERROR - mycodo.inputs.atlas_ph_52990ea8 - Error 101: Device not set up. See https://kizniche.github.io/Mycodo/Error-Codes#error-101 for more info.
2022-05-11 01:13:00,357 - ERROR - mycodo.controllers.controller_input_52990ea8 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2022-05-11 01:13:15,381 - ERROR - mycodo.inputs.atlas_ph_52990ea8 - Error 101: Device not set up. See https://kizniche.github.io/Mycodo/Error-Codes#error-101 for more info.
2022-05-11 01:13:30,407 - ERROR - mycodo.inputs.atlas_ph_52990ea8 - Error 101: Device not set up. See https://kizniche.github.io/Mycodo/Error-Codes#error-101 for more info.
2022-05-11 01:13:45,434 - ERROR - mycodo.inputs.atlas_ph_52990ea8 - Error 101: Device not set up. See https://kizniche.github.io/Mycodo/Error-Codes#error-101 for more info.
2022-05-11 01:13:45,434 - ERROR - mycodo.controllers.controller_input_52990ea8 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2022-05-11 01:14:00,359 - ERROR - mycodo.inputs.atlas_ph_52990ea8 - Error 101: Device not set up. See https://kizniche.github.io/Mycodo/Error-Codes#error-101 for more info.
2022-05-11 01:14:15,388 - ERROR - mycodo.inputs.atlas_ph_52990ea8 - Error 101: Device not set up. See https://kizniche.github.io/Mycodo/Error-Codes#error-101 for more info.

```

kizniche, 2022-05-10:
> i got the exact same problem

It is not the same problem. This issue was resolved and your error log doesn't show the relevant information to diagnose the issue. Open a new issue but include the error that occurs when the input is activated, This is all mentioned in the link provided by the error:

>You will often need to review the Daemon Log ([Gear Icon] -> Mycodo Logs) for any relevant errors that occurred when the Controller was initially activated to determine the issue setting up the device. Try deactivating, then activating the device, to see the initialization error again. Enabling Log Level: Debug in the Controller setting can also be useful by providing debugging log lines (when available) in addition to the info and error log lines.